jsk-ros-pkg/jsk_robot #1726: [spot] 散歩中にstrelkaga

a-ichikura, 2022-11-14:


k-okada, 2022-11-14:
何時ころだかわかるかな？

mqcmd196, 2022-11-14:
11:30頃だったと記憶しています

a-ichikura, 2022-11-14:
- [ ] head_lead_demoでSpotの腕を引っ張っていると疲れてくる(特に坂道)

- [ ] head_lead_demoの時もう少し速く動いてほしい

- [ ] 外に出るとsshする方法がないのでデバッグできなくて怖い

- [ ] head_lead_demoをしながらjoyで動かすと頭（腕）のかたちが変になる

- [ ] Boston Dynamicsの文字が目立ちすぎる

a-ichikura, 2022-11-14:
- [ ] head_lead_demoでSpotの腕を引っ張っていると疲れてくる(特に坂道)

```
(send *ri* :set-impedance-param :linear-stiffness #f(250 50 50) :rotational-stiffness #f(10 10 30) :linear-damping #f(1.5 1.5 1.5) :rotational-damping #f(0.5 0.5 0.5))  
```
- [ ] head_lead_demoの時もう少し速く動いてほしい

`jsk_spot_startup/config/head_teleop_twist_joy.yaml`
```
scale_linear:
  x: 2.0
  y: 2.0
```
を変更

```
<param name="~deadzone" value="4" />
 <param name="~maxrange" value="10" />
```
を変えると
`rostopic echo /joy_head/joy_raw`が変わる
- [ ] head_lead_demoをしながらjoyで動かすと頭（腕）のかたちが変になる
Right Stick 長押し：腕が初期状態に戻る

- [ ] 外に出るとsshする方法がないのでデバッグできなくて怖い
有線でつなげる（ethernetをspotに設定, ssh 192.168.150.x)

k-okada, 2022-11-24:
Spot本体のWIFIにつなげるか，お尻のネットワークにつなげて
```
 ssh -p 20022 10.0.0.3 -l spot
```
でCOREにログイン出来る気がしました．

a-ichikura, 2023-03-02:
岡田先生 @k-okada 

今日、spotで学内を散歩しようとしたら、工学部広場くらいで画像の保存がストップしてしまいました。
コードとしては,rostopicで`/spot/camera/hand_color/image`を10秒ごとに取得しようとしていただけで、google chatやdialogflowは使っていません。

ネットワークを必要としないコードだと思い,
そのまま散歩に出たのですが、133系が途切れるところらへんでtopicが消えて画像の保存ができていませんでした。

@sktometometo と一緒に調べたところ,
spotの`/usr/sbin/jsk_spot_startup-start/`と`/usr/sbin/app_manager-start`の
```export ROS_HOSTNAME=$(hostname)```
になっていると,
133につなげた状態でsupervisorを立ち上げると133系に繋がっていないところros nodeの通信が切れてしまうことが問題であると発見しました。

【解決策】
1.
```export ROS_HOSTNAME=localhost```
にすると, 133に繋がっていなくてもnode間の通信は可能。
しかし、この方法だと,133経由のrosの通信ができない。

2.
そのままの状態で, 133に繋ぐドングルを抜いて, supervisorを立ち上げる。
この方法だと,ロボットが133に接続したり切断したりしても, ロボット内のnodeの通信は切れない。
しかし, ドングルの抜き差しと, supervisorの立ち上げ直しが少し面倒。

これがわかったことで、
今までのコントローラが切れる問題もこの設定の影響っぽいです。

とりあえず今は2.の方法で散歩に出ようと思います。

ちなみに,修論の散歩の時にこの問題に気が付かなかったのは,
google chatとかdialogflowを使うためにポケットwifiを繋いだ状態で散歩に出ていて, そうするとインターネットにポケットwifi経由で繋がるからであると予想されます。

knorth55, 2023-03-02:
Belkaの`/etc/hosts`に`Belka`がいないのが原因ではとおもっている．
現状の`/etc/hosts`は以下になっています．

```
127.0.0.1       localhost
::1             localhost ip6-localhost ip6-loopback
ff02::1         ip6-allnodes
ff02::2         ip6-allrouters
```

sktometometo, 2023-03-02:
> Belkaの`/etc/hosts`に`Belka`がいないのが原因ではとおもっている．
> 現状の`/etc/hosts`は以下になっています．
> 
> ```
> 127.0.0.1       localhost
> ::1             localhost ip6-localhost ip6-loopback
> ff02::1         ip6-allnodes
> ff02::2         ip6-allrouters
> ```

BelKaを追加したら解決しました

closed 2023-06-27 by sktometometo