jsk-ros-pkg/jsk_robot #1733: [Fetch] 音声を使用する実験中に音声で喋るノードが喋ると実験を邪魔する

sktometometo, 2022-11-18:
音声を利用する実験をする際にロボットが勝手に喋らないように、自動で話している以下のノードの状態を切り替えたい

- /tweet
- /audible_warning

他にあれば追記お願いします。

sktometometo, 2022-11-18:
デバッグで使用する`sound_play`ノードとデモで喋らせる`sound_play`ノードを分けるべき?

sktometometo, 2022-11-18:
`/tweet`ノードは`respawn=true`のため、`rosnode kill /tweet`しても復活してしまう.

k-okada, 2022-11-18:
ナルホド。tweet -> sound_play の間にmux 入れてON/OFF切り替えられないかな。
--
◉ Kei Okada

sktometometo, 2022-11-18:
/robotsound/goal に mux入れる感じですかね

sktometometo, 2022-11-18:
アイディア的には

- ON/OFFできる sound_play のトピック/アクション -> /tweet や /audible_warning などデバッグ用
- ダイレクトに使用できる sound_play トピック/アクション -> 実験ノード用

みたいなのを用意するという意味合いですよね, mux だとアクションの切り替えが大変になるのでノードをそもそも分けてしまった方が使用時にするべき作業がシンプルで使いやすいと思うのですがいかがでしょうか?

knorth55, 2022-11-18:
tweetはしゃべっていなくて
tweet_warningとかです

knorth55, 2022-11-18:
で，ここらへんの話はどうだろう？
https://github.com/jsk-ros-pkg/jsk_robot/issues/1724

dynamic_reconfigureでvolumeを0にしようとおもっているんだけど，そもそもactionすらたたかないほうがいいのかな．
`audible_warning`では`enable`という`dynamic_reconfigure`があって，それを`false`にすると黙ります．
`tweet_warning`は`respawn`していないので，`rosnode kill`で黙ります．
ただ`tweet_warning`にも`enable`という`dynamic_reconfigure`を与えてもいい気がします．

sktometometo, 2022-11-18:
> tweetはしゃべっていなくて tweet_warningとかです

`tweet_warning` の実体ってどこにありますか？

```bash
fetch@fetch1075:~/ros/melodic/src$ grep -ir tweet_warning
fetch@fetch1075:~/ros/melodic/src$ 
```

knorth55, 2022-11-18:
`jsk_robot_startup/lifelog`かな

https://github.com/jsk-ros-pkg/jsk_robot/pull/1725

sktometometo, 2022-11-18:
> で，ここらへんの話はどうだろう？ #1724
> 
> dynamic_reconfigureでvolumeを0にしようとおもっているんだけど，そもそもactionすらたたかないほうがいいのかな． `audible_warning`では`enable`という`dynamic_reconfigure`があって，それを`false`にすると黙ります． `tweet_warning`は`respawn`していないので，`rosnode kill`で黙ります． ただ`tweet_warning`にも`enable`という`dynamic_reconfigure`を与えてもいい気がします．

まったく喋りたくないわけではなくて、実験ノードは喋らせたいのでvolume変えるのは違いかと思います。

knorth55, 2022-11-18:
`volume`は`sound_play`ではなくて，各ノードが送る`action`に記述されています．
なので，`audible_warning`とか`tweet_client_warning`だけが黙ります．

https://github.com/ros-drivers/audio_common/blob/ea45179ca7e1cad76f096318166c291a13378a67/sound_play/msg/SoundRequest.msg#L27

knorth55, 2022-11-18:
ただ`volume`が`0`でも`action`がたたかれるので，それによって上書きされてしまいます．
なので正しいのは`audible_warning`と同じで，`enable`みたいな`dynamic_reconfigure`を作るほうが，より良い気がします．
詳しくはこちら

https://github.com/jsk-ros-pkg/jsk_robot/pull/1725

sktometometo, 2022-11-18:
そちらのvolumeでしたか、各スクリプトに喋らなくするインターフェースを作る作戦はスケールしづらいと思うのですが.

sktometometo, 2022-11-18:
あたらしく何か喋るものを作るたびに喋らなくするインターフェースを作る必要が出てきます

knorth55, 2022-11-18:
それは増えてから考えたらいいのでは，と思うが．．．
muxも結局分類していって，増える未来しか見えないし，とりあえず手っ取り早いほうでいいのでは．
とりあえず`rosnode kill /tweet_client_warning`と`dynamic_reconfigure`で臨むことはできるはず

sktometometo, 2022-11-18:
よければついでに伺っておきたいのですが、 `/tweet_client_hogehoge` ってそれぞれどういう用途のものですか?

https://github.com/jsk-ros-pkg/jsk_robot/blob/ee8662bafc16cc0d1af38b2994ff51990367d052/jsk_robot_common/jsk_robot_startup/lifelog/tweet.launch#L60-L79

knorth55, 2022-11-18:
`mux`でもいいが，もともと僕らがやりたかったことは`volume`を下げることがなので，やりたいことが違うね．そもそも．
黙らせたいわけではなく，静かにさせたいなので．
で，黙らせるには`mux`がいいかと思います．

knorth55, 2022-11-18:
`tweet_client_warning`がdiagnosticsに基づいてtweetしているやつ．
これがだいたい定期的に写真撮っているノードです．写真撮るときにしゃべります．

`tweet_client_uptime`は起動している時間に応じてTweetしているやつです．
Fetchでこれが正しく動いていない気もしますが，たまにtweetして，そのときにしゃべります．

`tweet_client_worktime`はユーザがログインしている時間に応じてTweetしているやつです．
Fetchでこれが正しく動いていない気もしますが，たまにtweetして，そのときにしゃべります．

`tweet_client_tablet`は使われていませんが，`tablet`からTweetするようになっていたものです．
いまは何もしていません．

北川がやっている限りだと，`tweet_client_warning`を止めるだけで黙った気がしますが，上3つは止めてもいい気がします．

sktometometo, 2022-11-18:
なるほど、ありがとうございます。あと`develop/fetch`ブランチだと `jsk_3rdparty/rostwitter` の `tweet_image_server.py` も起動していてこいつも喋りそうなコードあるんですけど使っていますか? 

https://github.com/jsk-ros-pkg/jsk_robot/blob/699948413720871c8537c29a8438fd5d5e2261d9/jsk_robot_common/jsk_robot_startup/lifelog/tweet.launch#L30-L36

knorth55, 2022-11-18:
`rostwitter`だけでは写真を撮るのが大変だったので，Actionを叩くと`tweet`してくれるようにしたのが`tweet_image_server.py`です．
これは`tweet_client.l`の一部をPython化しているもので，このActionを叩くように全部していけば，ここのノードさえ黙ってくれれば，１つを黙らせるだけでいいことになりますね．