jsk-ros-pkg/jsk_robot #1757: [Spot] サッカーボールで遊ぶデモ

mqcmd196, 2022-12-06:
Cc: @a-ichikura , @k-okada 

## How to start?
In spot@Belka
1. 認識系を立ち上げる
```bash
roslaunch jsk_spot_startup perception.launch ml_confidence:=0.60 start_ncb_provider:=true start_aws_auto_checkin_app:=true start_aws_detect_faces:=true image_sources:="[frontleft_fisheye_image, frontright_fisheye_image, left_fisheye_image, right_fisheye_image]"
```
2. lifelog(mongo系)をlaunchする
```bash
roslaunch jsk_spot_startup lifelog.launch
```
3. interaction(google_chatなど)をlaunchする
```bash
roslaunch jsk_spot_startup interaction.launch
```
4. ボール遊びサンプルプログラム
```bash
~/spot_driver_ws/src/jsk-ros-pkg/jsk_robot/jsk_spot_robot/jsk_spot_startup/apps/head_lead_demo/fuga.l
```

## TODO
@mqcmd196 
- [ ]  Mongoに`/spot/pick_object_in_image` actionのGoalとResultをdumpする
- [ ] `fuga.l`をlead_teleopにマージする
- [ ] スピーカを取り付ける
@a-ichikura 
- [ ] 実験をする

mqcmd196, 2022-12-06:
@tkmtnt7000 BelkaにStrelkaと同じスピーカーを取り付けたいんだけど，あまりってありますか？

tkmtnt7000, 2022-12-06:
73A4かキッチン向かい側のUnitree道具置き場になければ、余りはないかもしれません。。。

mqcmd196, 2022-12-06:
@tkmtnt7000 ありがとう，発見しました！