jsk-ros-pkg/jsk_robot #1758: panda_eus build failure with `xacro`

knorth55, 2022-12-06:
```
___________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << panda_eus:make /home/fetch/ros/melodic/logs/panda_eus/build.make.001.log                                                                                                                     
xacro: in-order processing became default in ROS Melodic. You can drop the option.
name 'xacro' is not defined 
when evaluating expression 'xacro.load_yaml('$(find franka_description)/robots/panda/joint_limits.yaml')'
when processing file: dual_panda.urdf.xacro
make[2]: *** [/home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_robot/jsk_panda_robot/panda_eus/models/dual_panda.urdf] Error 2
make[2]: *** Deleting file '/home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_robot/jsk_panda_robot/panda_eus/models/dual_panda.urdf'
make[1]: *** [CMakeFiles/generate_panda_lisp.dir/all] Error 2
make: *** [all] Error 2
cd /home/fetch/ros/melodic/build/panda_eus; catkin build --get-env panda_eus | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................................................................................
Failed     << panda_eus:make                    [ Exited with code 2 ]                                                                                                                                     
Failed    <<< panda_eus                         [ 1.5 seconds ]    
```

knorth55, 2022-12-06:
this error happens `xacro` `1.13.11`.
I update `xacro` to `1.13.18`, it works.

knorth55, 2022-12-06:
we need this PR to use `xacro.load_yaml`.
`xacro>=1.13.14` is required to build `panda_eus`.

https://github.com/ros/xacro/pull/283

pazeshun, 2022-12-06:
そうです、xacroの問題があったので、melodic未満ではdual_panda.lを生成しないように書きました。
https://github.com/jsk-ros-pkg/jsk_robot/blob/4d43301a00f30a985ffb9f65597519ab8d40669b/jsk_panda_robot/panda_eus/CMakeLists.txt#L17-L19
この時は、franka_descriptionのバージョンを書かなかったように、パッケージをaptの最新にしておくのはユーザーの責任という思想でしたが、franka_descriptionのバージョンをCMakeLists.txtに書くなら、xacroのバージョンもCMakeLists.txtに書いた方がいい気もしてきたので、やっておきます。

ただ、xacroはROSの基本パッケージな気がして、panda_eusのCMakeLists.txtが出すwarningを読んでxacroだけアップデートすると、xacroに依存している古いままのパッケージが動かなくなったりしそうなので、やっぱりaptの最新にしておくのはユーザーがやっておいてほしい…とは正直思いますが…

pazeshun, 2022-12-06:
#1751 でxacroのバージョンを明示的にチェックするようにしました

knorth55, 2022-12-07:
#1751 確認しました．OKです．
xacroを最新にし忘れていたときのエラーがわかりにくいので，最新版にすべきという話と，エラーがわかりにくい話は別だと思います．

pazeshun, 2022-12-07:
それなら、「エラーがわかりにくい」と明示的に書いてほしかったです。
今回はたまたま直せてましたが

knorth55, 2022-12-07:
あ，個人的には最新版にすべきであるというのは，全員そうすべきであるとは思わないです．
実際xacroが先に入っていたらrosdepしても，upgradeされないので，自分でapt upgradeしないといけないのは，なかなか難しい気がします．
ただ長谷川の意見を聞き入れたとしても，エラーがわかりにくい話は残るなと思っています．

pazeshun, 2022-12-13:
Closed via #1751

closed 2022-12-13 by pazeshun