jsk-ros-pkg/jsk_robot #1765: [Spot] 81C1でsanshiroとSpotとの通信が遅い

knorth55, 2022-12-15:
SSIDは同じだが，どうにも`sanshiro`が不安定？

### 遅い時

```
spot@StrelKa:~/Downloads$ nmcli -c yes -f SSID,BSSID,CHAN,FREQ,SIGNAL,BARS dev wifi 
SSID                         BSSID              CHAN  FREQ      SIGNAL  BARS 
JSK300M                      90:9A:4A:95:6F:06  4     2427 MHz  34      ▂▄__ 
--                           96:9A:4A:95:6F:06  4     2427 MHz  34      ▂▄__ 
spot-BD-00060012             00:30:1A:4F:43:5F  11    2462 MHz  34      ▂▄__ 
sanshiro                     74:88:BB:02:C0:D6  6     2437 MHz  30      ▂___ 
Echinometridae_2.4GHz@robot  68:FF:7B:23:7A:2F  3     2422 MHz  27      ▂___ 
eng.ut-1x                    74:88:BB:02:C0:D5  6     2437 MHz  24      ▂___ 
bld2-guest                   74:88:BB:02:C0:D4  6     2437 MHz  24      ▂___ 
eng.ut                       74:88:BB:02:C0:D0  6     2437 MHz  17      ▂___ 
eduroam                      74:88:BB:02:C0:DB  6     2437 MHz  17      ▂___ 
ist_members                  74:88:BB:02:C0:D7  6     2437 MHz  17      ▂___ 
UTokyo-WiFi                  74:88:BB:02:C0:DD  6     2437 MHz  17      ▂___ 
UTokyo-Guest                 74:88:BB:02:C0:DC  6     2437 MHz  17      ▂___ 
Buffalo-G-8890               90:96:F3:BB:88:92  10    2457 MHz  17      ▂___ 
Buffalo-G-8890-WPA3          90:96:F3:BB:88:93  10    2457 MHz  14      ▂___ 
ne-lab-81b                   D8:0D:17:90:FC:B4  8     2447 MHz  10      ▂___ 
ist_members                  C8:84:A1:F2:C1:87  1     2412 MHz  7       ▂___ 
YNL_2G                       B0:6E:BF:9E:C6:38  6     2437 MHz  4       ____ 
BUFFALO-F7E85F               10:6F:3F:F7:E8:5F  6     2437 MHz  4       ____ 
--                           9C:65:F9:18:32:D8  6     2437 MHz  4       ____ 
Echinometridae_2.4GHz@D3     68:FF:7B:23:7A:43  11    2462 MHz  4       ____ 
--                           B0:6E:BF:9E:7E:E8  6     2437 MHz  0       ____
```

### 早い時

```
spot@StrelKa:~/Downloads$ nmcli -c yes -f SSID,BSSID,CHAN,FREQ,SIGNAL,BARS dev wifi list 
SSID                         BSSID              CHAN  FREQ      SIGNAL  BARS 
sanshiro                     74:88:BB:02:C0:D6  6     2437 MHz  100     ▂▄▆█ 
JSK300M                      90:9A:4A:95:6F:06  4     2427 MHz  34      ▂▄__ 
--                           96:9A:4A:95:6F:06  4     2427 MHz  34      ▂▄__ 
spot-BD-00060012             00:30:1A:4F:43:5F  11    2462 MHz  34      ▂▄__ 
Echinometridae_2.4GHz@robot  68:FF:7B:23:7A:2F  3     2422 MHz  27      ▂___ 
UTokyo-WiFi                  74:88:BB:02:C0:DD  6     2437 MHz  24      ▂___ 
eng.ut-1x                    74:88:BB:02:C0:D5  6     2437 MHz  17      ▂___ 
eng.ut                       74:88:BB:02:C0:D0  6     2437 MHz  17      ▂___ 
bld2-guest                   74:88:BB:02:C0:D4  6     2437 MHz  17      ▂___ 
eduroam                      74:88:BB:02:C0:DB  6     2437 MHz  17      ▂___ 
ist_members                  74:88:BB:02:C0:D7  6     2437 MHz  17      ▂___ 
UTokyo-Guest                 74:88:BB:02:C0:DC  6     2437 MHz  17      ▂___ 
ne-lab-81b                   D8:0D:17:90:FC:B4  8     2447 MHz  14      ▂___ 
Buffalo-G-8890-WPA3          90:96:F3:BB:88:93  10    2457 MHz  10      ▂___ 
Buffalo-G-8890               90:96:F3:BB:88:92  10    2457 MHz  10      ▂___ 
ist_members                  C8:84:A1:F2:C1:87  1     2412 MHz  7       ▂___ 
Echinometridae_2.4GHz@D3     68:FF:7B:23:7A:43  11    2462 MHz  7       ▂___ 
BUFFALO-F7E85F               10:6F:3F:F7:E8:5F  6     2437 MHz  4       ____ 
--                           9C:65:F9:18:32:D8  6     2437 MHz  4       ____ 
yoshilab525-g                00:24:A5:E9:A8:0E  6     2437 MHz  4       ____ 
Buffalo-G-A9B0               C4:3C:EA:F6:A9:B1  10    2457 MHz  4       ____ 
YNL_2G                       B0:6E:BF:9E:C6:38  6     2437 MHz  0       ____ 
--                           B0:6E:BF:9E:7E:E8  6     2437 MHz  0       ____ 
```

knorth55, 2022-12-15:
スクリプトを書いて，sanshiroの強度をみてみました．
強度があがったり下がったりしています．
100から30に落ちて，100に上がってを繰り返していて，電波強度がそこまで不安定になることがあるのでしょうか．．．

```
2022-12-15 14:02:19 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:02:20 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:02:21 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:02:22 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:02:24 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:25 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:26 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:27 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:28 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:29 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:30 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:32 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:33 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:34 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:35 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:36 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:37 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:38 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:40 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:41 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:42 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 34 ▂▄__ 
2022-12-15 14:02:43 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 34 ▂▄__ 
2022-12-15 14:02:44 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 34 ▂▄__ 
2022-12-15 14:02:45 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 34 ▂▄__ 
2022-12-15 14:02:46 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 34 ▂▄__ 
2022-12-15 14:02:48 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:49 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:50 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:51 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:52 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:53 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:54 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:56 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:57 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:58 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:02:59 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:03:00 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:03:01 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:03:03 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:03:04 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:03:05 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:03:06 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 34 ▂▄__ 
2022-12-15 14:03:07 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 34 ▂▄__ 
2022-12-15 14:03:08 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 34 ▂▄__ 
2022-12-15 14:03:10 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 34 ▂▄__ 
2022-12-15 14:03:11 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 34 ▂▄__ 
2022-12-15 14:03:12 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 34 ▂▄__ 
2022-12-15 14:03:13 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 34 ▂▄__ 
2022-12-15 14:03:14 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 34 ▂▄__ 
2022-12-15 14:03:15 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 34 ▂▄__ 
2022-12-15 14:03:16 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:03:18 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:03:19 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:03:20 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:03:21 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:03:22 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:03:23 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:03:25 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:03:26 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:03:27 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:03:28 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:03:29 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:03:30 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:03:31 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:03:33 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
[... 52 lines cut ...]
2022-12-15 14:04:34 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:04:35 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:04:36 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:04:37 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:04:38 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:04:39 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:04:41 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:04:42 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:04:43 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:04:44 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:04:45 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:04:46 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 100 ▂▄▆█ 
2022-12-15 14:04:48 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 58 ▂▄▆_ 
2022-12-15 14:04:49 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 58 ▂▄▆_ 
[... cut ...]
2022-12-15 14:04:52 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 58 ▂▄▆_ 
2022-12-15 14:04:53 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:04:54 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:04:56 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:04:57 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:04:58 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:04:59 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___ 
2022-12-15 14:05:00 sanshiro 74:88:BB:02:C0:D6 6 2437 MHz 30 ▂___
```

k-okada, 2022-12-15:
１） 5Ghz のドングルを使っています．（以下のREADME.md）https://www.amazon.co.jp/gp/product/B07MXHJ6KB/ref=ppx_yo_dt_b_asin_title_o01_s01?ie=UTF8&psc=1　だったかな．Belkaのものを確認してください

２）Wifiの強度が下がったら，自動でAPを探しに行く設定が以下の，`echo 'install 88x2bu ... `の部分になります．これが動いて5Ghz動くものが１）になります．
https://github.com/k-okada/jsk_robot/blob/spot_arm/jsk_spot_robot/SetupInternalPCAndSpotUser.md#setup-wifi-interfaces

３）5Gに優先的（？５Gだけつながるように？）いかの`band=a`を手動で書いています
```
spot@BelKa:/etc/NetworkManager/system-connections$ sudo cat sanshiro 

[wifi]
...
band=a
```

knorth55, 2022-12-15:
73b2にいるbelkaで試してみましたが，電波強度は`68`と低いですが，同じ値で安定しています．
81c1の環境は，電波強度が乱高下するのが特異な気がします．

```
2022-12-15 14:09:01 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:02 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:03 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:04 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:05 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:06 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:07 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:08 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:09 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:10 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:11 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:12 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:13 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:14 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:15 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:16 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:17 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:18 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:19 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:20 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:21 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:22 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:23 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:24 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:25 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:26 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:27 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:28 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:29 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:30 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:31 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:32 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:33 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:35 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:36 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:37 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:38 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:39 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:40 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:41 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:42 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:43 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:44 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:45 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:46 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:47 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:09:48 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 14:09:49 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 14:09:50 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 14:09:51 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 14:09:52 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 14:09:53 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 14:09:54 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 14:09:55 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 14:09:56 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 14:09:57 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 14:09:58 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 14:09:59 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 14:10:00 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 14:10:01 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 14:10:02 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 14:10:03 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 14:10:04 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 14:10:05 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 14:10:06 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 14:10:07 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 67 ▂▄▆_ 
[... 37 lines cut ...]
2022-12-15 14:10:46 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:10:47 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:10:48 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:10:49 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:10:50 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:10:51 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:10:52 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:10:53 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:10:54 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:10:56 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:10:57 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:10:58 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
[... cut ...]
2022-12-15 14:11:03 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:11:04 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:11:05 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:11:06 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:11:07 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:11:08 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:11:09 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_ 
2022-12-15 14:11:10 sanshiro 78:72:5D:B6:CC:A9 136 5680 MHz 68 ▂▄▆_
```

knorth55, 2022-12-15:
今はstrelkaは2.4GHzにつながっているようです．

```
spot@StrelKa:~$ nmcli -c yes -f IN-USE,SSID,BSSID,CHAN,FREQ,SIGNAL,BARS dev wifi list
IN-USE  SSID                         BSSID              CHAN  FREQ      SIGNAL  BAR
        JSK300M                      90:9A:4A:95:6F:06  4     2427 MHz  34      ▂▄_
        --                           96:9A:4A:95:6F:06  4     2427 MHz  34      ▂▄_
        spot-BD-00060012             00:30:1A:4F:43:5F  11    2462 MHz  30      ▂__
        Echinometridae_2.4GHz@robot  68:FF:7B:23:7A:2F  3     2422 MHz  24      ▂__
        eng.ut                       74:88:BB:02:C0:D0  6     2437 MHz  20      ▂__
        eng.ut-1x                    74:88:BB:02:C0:D5  6     2437 MHz  17      ▂__
        bld2-guest                   74:88:BB:02:C0:D4  6     2437 MHz  17      ▂__
        eduroam                      74:88:BB:02:C0:DB  6     2437 MHz  17      ▂__
        ist_members                  74:88:BB:02:C0:D7  6     2437 MHz  17      ▂__
        UTokyo-WiFi                  74:88:BB:02:C0:DD  6     2437 MHz  17      ▂__
        UTokyo-Guest                 74:88:BB:02:C0:DC  6     2437 MHz  17      ▂__
*       sanshiro                     74:88:BB:02:C0:D6  6     2437 MHz  17      ▂__
        Buffalo-G-8890-WPA3          90:96:F3:BB:88:93  10    2457 MHz  14      ▂__
        ne-lab-81b                   D8:0D:17:90:FC:B4  8     2447 MHz  10      ▂__
        Buffalo-G-8890               90:96:F3:BB:88:92  10    2457 MHz  10      ▂__
        Echinometridae_2.4GHz@D3     68:FF:7B:23:7A:43  11    2462 MHz  10      ▂__
        eng.ut                       C8:84:A1:F2:C1:80  1     2412 MHz  7       ▂__
        eng.ut-1x                    C8:84:A1:F2:C1:85  1     2412 MHz  7       ▂__
        Nakaolab_G                   DC:FB:02:68:C8:40  3     2422 MHz  7       ▂__
        YNL_2G                       B0:6E:BF:9E:C6:38  6     2437 MHz  7       ▂__
        --                           9C:65:F9:18:32:D8  6     2437 MHz  7       ▂__
        eng.ut-1x                    C8:84:A1:EE:F4:45  1     2412 MHz  4       ___
        106F3FF38E50                 10:6F:3F:F3:8E:50  1     2412 MHz  4       ___
        --                           B0:6E:BF:9E:7E:E8  6     2437 MHz  4       ___
        BUFFALO-F7E85F               10:6F:3F:F7:E8:5F  6     2437 MHz  4       ___
        Buffalo-G-D628               C4:3C:EA:EC:D6:28  11    2462 MHz  4       ___
```

knorth55, 2022-12-15:
belkaとstrelkaでwifiドングル違うものですね．．．

knorth55, 2022-12-15:
strelkaも同じドングルにして同じくらいのSIGNALで安定するようになりました．

```
2022-12-15 15:41:04 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:05 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:06 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:07 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:08 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:09 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:10 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:11 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:12 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:13 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:14 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:15 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:16 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:17 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:18 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:19 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:20 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:21 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:22 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:23 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:24 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:25 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:26 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:27 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:28 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:29 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:30 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:31 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:32 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:33 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:34 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:35 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:36 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:37 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:39 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:40 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:41 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:42 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:43 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:44 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:45 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:46 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:47 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:48 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:49 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:50 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:51 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:52 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:53 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:54 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:55 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:56 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:57 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:58 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:41:59 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:42:00 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:42:01 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:42:02 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:42:03 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:42:04 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:42:05 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:42:06 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:42:07 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:42:08 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:42:09 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:42:10 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:42:11 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
[... 86 lines cut ...]
2022-12-15 15:43:41 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:43:42 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:43:43 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:43:44 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:43:45 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:43:46 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:43:47 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:43:48 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:43:49 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:43:50 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:43:51 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
[... cut ...]
2022-12-15 15:43:56 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 15:43:57 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 67 ▂▄▆_ 
2022-12-15 15:43:58 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:43:59 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:44:00 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:44:01 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:44:03 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 
2022-12-15 15:44:04 sanshiro 74:88:BB:02:C0:D9 136 5680 MHz 63 ▂▄▆_ 

```

knorth55, 2022-12-15:
BelkaもStrelkaも現在：

ドングル： Archer T2U Nano
カーネルモジュール: 88XXau
https://github.com/aircrack-ng/rtl8812au

をつかっていて，signalがdb単位ではなく，かつbgscanによる自動切り替えが走っていません．

73B2に Archer T3U Nanoがあったので，こちらは88X2buで動くはずなので，こちらを試してます．

https://tech.nosuz.jp/post/2021-09/ubuntu_wifi/#rtl8812bu-and-rtl8822bu

knorth55, 2022-12-15:
Archer T3U Nanoについて．
https://github.com/cilynx/rtl88x2bu では動かない．

下の2つでは動く．ただしsignalが`dB`単位ではでない．
https://github.com/RinCat/RTL88x2BU-Linux-Driver
https://github.com/morrownr/88x2bu-20210702

knorth55, 2022-12-15:
StrelkaもBelkaと同じ`TP-LINK Archer T2U Nano`にしてネットワークを改善しました．
デバイスが変わったのでDHCPの静的割当も変更しました．
Strelkaで名前解決できない場合はDNSのキャッシュを消してみてください．

closed 2022-12-15 by knorth55