jsk-ros-pkg/jsk_robot #1769: [jsk_naoqi_robot/cross] Building ROS system on Dockerでのエラー: `zope.interface-4.3.2: No such file or directory`

kochigami, 2022-12-21:
パソコンにUbuntu18.04を入れ直し、ROS melodic、jsk_naoqi_robot環境 (https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_naoqi_robot#setup-environment) をインストールしました。

```
leus@leus-ThinkStation-P320-Tiny:~/catkin_ws/src/jsk_robot/jsk_naoqi_robot/cross$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.6 LTS
Release:	18.04
Codename:	bionic
```
```
leus@leus-ThinkStation-P320-Tiny:~/catkin_ws/src$ wstool info
workspace: /home/leus/catkin_ws/src

 Localname         S SCM Version (Spec)    UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------         - --- --------------    -----------  ---------------------------
 pepper_robot        git master  (-)       efad3979b374 github.com/ros-naoqi/pepper_robot
 naoqi_driver        git kochigami-develop 98c0b678286a github.com/kochigami/naoqi_driver
 naoqi_dashboard     git kochigami-develop 7f32005e08e0 github.com/kochigami/naoqi_dashboard
 naoqi_bridge_msgs   git kochigami-develop d7417613690e github.com/kochigami/naoqi_bridge_msgs
 naoqi_bridge        git kochigami-develop c28b727e1e9b github.com/kochigami/naoqi_bridge
 nao_robot           git master  (-)       67476469a137 github.com/ros-naoqi/nao_robot
 nao_interaction     git master  (-)       f97ad12f3896 github.com/ros-naoqi/nao_interaction
 jsk_robot           git master            d551865511c3 github.com/jsk-ros-pkg/jsk_robot.git
```

クロスコンパイル環境のセットアップの所 (https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_naoqi_robot/cross#setup-pepper-robot)で `make system`する所で、`make system`を打つ代わりに、
中身のコマンドを1つずつ打っているのですが (`./prepare_requirements_ros1.sh`, `./build_ros1_dependencies.sh`)、`./build_ros1_dependencies.sh`の所で以下のエラーが出ています。

i386_build_ros1_dependemcies.logの抜粋
        

```
+ for script_file in '$(ls /home/nao/ros1_dependencies_build_scripts/|sort)'
+ /home/nao/ros1_dependencies_build_scripts/1012-python-zope.interface
+ DEBIAN_DIR=/home/nao/ros1_dependencies_sources/src/python-zope.interface/debian/debian
+ SOURCE_DIR=/home/nao/ros1_dependencies_sources/src/python-zope.interface/zope.interface-4.3.2
+ cd /home/nao/ros1_dependencies_sources/src/python-zope.interface/zope.interface-4.3.2
/home/nao/ros1_dependencies_build_scripts/1012-python-zope.interface: line 15: cd: /home/nao/ros1_dependencies_sources/src/python-zope.interface/zope.interface-4.3.2: No such file or directory
+ exit 1
```

今の所、2回試して2回とも同じエラーが出ています。
どう直せばよろしいでしょうか。

参考：ログファイル全体  [i386_build_ros1_dependencies.log](https://github.com/jsk-ros-pkg/jsk_robot/files/10274041/i386_build_ros1_dependencies.log)

kochigami, 2023-02-17:
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/cross/build_ros1_dependencies.sh#L46
で`i386_ws_ros1_dependencies_sources/src`以下に
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/cross/repos/ros1_dependencies.repos
で書かれたパッケージがインストールされる。

`zope.interface-4.3.2`に関しては、
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/cross/repos/ros1_dependencies.repos#L140
のURLにアクセスする際に、
```
File Transfer Blocked

Transfer of the file you were trying to download or upload has been blocked in accordance with company policy. Please contact your system administrator if you believe this is in error.

File name: build.cmd
```
というようにブロックされている。（2台の異なるパソコンで確認（ダウンロードリンクをブラウザで直接打ってみた））

多分、インストールができていない。
ブロックされているのはこれだけみたいで、違うバージョンであればインストールできそうだが、
18.04では4.3.2を入れることになっている。

`i386_ws_ros1_dependencies_sources`のバックアップがあれば、この部分のディレクトリだけ追加して何とかなりそう・・・？

kochigami, 2023-02-21:
`make system`に関しては、新しいパッケージを追加する、ということでなければ、
とりあえず`make system`して途中で止まっても、バックアップの`i386_System`を取ってくれば大丈夫そう。
一旦閉じます。

closed 2023-02-21 by kochigami

k-okada, 2023-05-02:
@kochigami 
めちゃめちゃ遅くなりました．
> File Transfer Blocked
については，今はなおっているかな？　http://archive.ubuntu.com/ubuntu/pool/main/z/zope.interface/ にアクセスできれば問題ないはずですね．ここが無くなることはないので，ネットワークのつなぎ先を変えてみる（自宅→職場，JSKなら工学部→情報理工）で再確認できます．

これ以外に，現時点で実行すると，
`pixel-ring` をインストールするところでエラーがでるはずで，これは

https://github.com/jsk-ros-pkg/jsk_robot/pull/1797

で直るはずです．僕の手元では動いているので，確認してみてください．

kochigami, 2023-05-08:
ありがとうございます。
`zope...` のFile Transfer Blockedについては、直っていることを確認しました。
#1797 で`make system`がエラーなく実行できることを確認しました。

kochigami, 2023-05-08:
雑な試し方ですみません・・・。
明日か、今週中に試します。

> ところで，make system 動いていますでしょうか？こっちではエラーの気もしています．

`make system`一つのコマンドでは試していなくて、中のコマンド3つを1つずつ試していました。

> もう一つは，これで作ったバージョンで実機は動いていますか？

コンパイルできて、満足していました・・・（実機コピーはしていません。）
試します。

k-okada, 2023-05-08:
ごめん，make system ではなく make user でした．
--
◉ Kei Okada

k-okada, 2023-05-10:
@kochigami ありががとうございます．
https://github.com/jsk-ros-pkg/jsk_robot/issues/1775#issuecomment-1434444806
ですでに報告があったのね．見逃していました．
https://github.com/jsk-ros-pkg/jsk_robot/pull/1797
でlibjpeg-turbo をインストールするようにしてみました．

> （バージョンをどこかに指定するのが正しい直し方だと思います。）

は余りいい方法がなくて，https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/cross/repos/roseus_no_window.repos に書けば良いんですが，個々に書いている特別なブランチは将来的にはmasterにmerge されるハズなので，あまり追加してくないという状況です．

https://github.com/jsk-ros-pkg/jsk_robot/pull/1797
でも以下のように怪しいことをしているので，上手く動かなかったらlibjpeg-turboは諦めて，1.9.5をインストールするようにするのが良いと思います．

https://github.com/jsk-ros-pkg/jsk_robot/pull/1797/files#diff-b84bd801d27ce2b8d004b399067663a39da81f0df9fa6ee2622c63d5104c2c66R23-R30