jsk-ros-pkg/jsk_robot #1771: [jsk_naoqi_robot/cross/README] バックアップのros1-pepper.tarを使う場面がよく分かりません

kochigami, 2022-12-22:
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/cross/README.md#development
にある、
「You can restore from saved docker container. For JSK users, you can find latest data at xxx.」
としてバックアップされている`ros1-pepper.tar`はどのような時に使えば良いのでしょうか。

例えば、`make system`が途中で失敗して、バックアップの`i386_System`フォルダをダウンロードした後、
`make user`する時に本当は使わないといけない、などありますでしょうか。

#### そのように思った理由
#1769 で`make system`が途中で失敗したため（`./prepare_requirements_ros1.sh`, `./build_ros1_dependencies.sh`, `./build_ros1.sh`とあるうちの、`./build_ros1_dependencies.sh`の途中でエラー）、
バックアップの`i386_System`フォルダをダウンロードし、`cross`ディレクトリの下に置きました。
これで、`make system`は実質完了したと考えております。

そして、`prepare_requirements_ros1.sh`を実行した時にros1-pepper イメージは作られています（ただし、バックアップされているものと比べ、色々ビルドできていないもの）。
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/cross/prepare_requirements_ros1.sh#L23

その後、`make user`を実行し、
作られたプログラムをPepperにコピーし、起動することを確認しました。

しかし、make userの中身である`build_user.sh`を見ると、
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/cross/build_user.sh#L71
のように、Dockerのros1-pepper イメージを使用していて、
`make system`と`make user`は処理が独立しているのかな、と心配になりました。
（`make user`でdockerが何をしているのかよく分かっていないので、もう少し勉強します。）

k-okada, 2023-05-02:
@kochigami 

	./prepare_requirements_ros1.sh

→　docker image (ros1-pepper:i386) (ros1-pepper.tarはそのイメージを書き出したもの）を作成し，
Pythonをコンパイル，（以降では /home/nao/System/Pythno-2.7.17　でマウントされる）
  狙いはPepperの実機の中のPCと同じ環境をDocker上に再現すること
なので，これ以降の作業は全て ros1-pepper のdocker 上でコンパイルして，体内のPC（ATOM?)で動くバイナリを生成する．つまりクロスコンパイルしている．

	./build_ros1_dependencies.sh
→ ROSを使うのに必要なライブラリ（普通のUbuntuだとapt get で入るもの）をインストール，インストールするライブラリの一覧はrepos/ros1_dependencies.repos これを ros1_dependencies_build_scripts のスクリプトでコンパイル＆インストール
普通のPCだったら，apt-get で入るはずだが，pepper はsudo が出来ないので（だっけ？），自前でインストールする．あるいは，そもそも既存の環境を壊したくないので，ユーザのホームディレクトリ内に全てのライブラリを持っている．

	./build_ros1.sh

rosinstall_generator でROSのソース一式を持ってきてコンパイル http://wiki.ros.org/rosinstall_generator

        ./build_user.sh
基本はbuild_ros1.sh と同じことをしている．が，イメージとしてはbuid_ros1 は一回やって
System ディレクトリ以下を完成させる．自分が開発する時は，build_ros1 でコンパイルされるものはいじる可能性が低いので，Systemディレクトリは１回作成されれば後はコピーでよい．

一方，User以下は何度も繰り返しデバッグ修正するので，最悪 rm -fr i386_User して消しながらコンパイルする．

kochigami, 2023-08-30:
#1846 
`./prepare_requirements_ros1.sh`が失敗する時に、
ros1-pepper.tarを使ってイメージを復元し、`./build_ros1_dependencies.sh`などの続きの処理を実行できる。

closed 2023-08-30 by kochigami