jsk-ros-pkg/jsk_robot #1776: [PR1012] PR1012のC1が突然おちることがある

knorth55, 2023-02-17:
PR1012のC1が突然おちることがある．
電源ランプはついたままだが，sshや通信などができなくなる．
今日もcatkin build中に発生していて，`/var/log/syslog`は以下のようになっている．
メモリのハードウェアエラーのよう？

cc. @nakane11 @iory 

```
Feb 17 11:45:01 pr1012 CRON[16036]: (root) CMD (command -v debian-sa1 > /dev/null && debian-sa1 1 1)
Feb 17 11:45:36 pr1012 kernel: [149396.111599] {48}[Hardware Error]: Hardware error from APEI Generic Hardware Error Source: 1
Feb 17 11:45:36 pr1012 kernel: [149396.111603] {48}[Hardware Error]: It has been corrected by h/w and requires no further action
Feb 17 11:45:36 pr1012 kernel: [149396.111618] {48}[Hardware Error]: event severity: corrected
Feb 17 11:45:36 pr1012 kernel: [149396.111620] {48}[Hardware Error]:  Error 0, type: corrected
Feb 17 11:45:36 pr1012 kernel: [149396.111621] {48}[Hardware Error]:  fru_text: CorrectedErr
Feb 17 11:45:36 pr1012 kernel: [149396.111622] {48}[Hardware Error]:   section_type: memory error
Feb 17 11:45:36 pr1012 kernel: [149396.111624] {48}[Hardware Error]:   node: 1 device: 0
Feb 17 11:45:36 pr1012 kernel: [149396.111625] {48}[Hardware Error]:   error_type: 2, single-bit ECC
^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@
^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@
^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@
^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@
^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@
```

nakane11, 2023-02-28:
今日も同じエラーがおきました

一回目
```
Feb 28 21:33:51 pr1012 kernel: [107328.679342] IPMI message handler: BMC returned incorrect response, expected netfn 2d cmd 0, got netfn 2d cmd 1
Feb 28 21:34:47 pr1012 kernel: [107385.111954] {20}[Hardware Error]: Hardware error from APEI Generic Hardware Error Source: 1
Feb 28 21:34:47 pr1012 kernel: [107385.111957] {20}[Hardware Error]: It has been corrected by h/w and requires no further action
Feb 28 21:34:47 pr1012 kernel: [107385.111959] {20}[Hardware Error]: event severity: corrected
Feb 28 21:34:47 pr1012 kernel: [107385.111961] {20}[Hardware Error]:  Error 0, type: corrected
Feb 28 21:34:47 pr1012 kernel: [107385.111962] {20}[Hardware Error]:  fru_text: CorrectedErr
Feb 28 21:34:47 pr1012 kernel: [107385.111964] {20}[Hardware Error]:   section_type: memory error
Feb 28 21:34:47 pr1012 kernel: [107385.111965] {20}[Hardware Error]:   node: 1 device: 0 
Feb 28 21:34:47 pr1012 kernel: [107385.111966] {20}[Hardware Error]:   error_type: 2, single-bit ECC
Feb 28 21:35:01 pr1012 CRON[21138]: (root) CMD (command -v debian-sa1 > /dev/null && debian-sa1 1 1)
Feb 28 21:35:05 pr1012 kernel: [107402.577163] IPMI message handler: BMC returned incorrect response, expected netfn 5 cmd 2d, got netfn b cmd 23
Feb 28 21:35:13 pr1012 kernel: [107410.839751] IPMI message handler: BMC returned incorrect response, expected netfn b cmd 23, got netfn 20 cmd 7
Feb 28 21:35:13 pr1012 kernel: [107410.954524] IPMI message handler: BMC returned incorrect response, expected netfn b cmd 23, got netfn b cmd 0
Feb 28 21:35:19 pr1012 kernel: [107416.788211] IPMI message handler: BMC returned incorrect response, expected netfn b cmd 23, got netfn 23 cmd fc
Feb 28 21:35:34 pr1012 kernel: [107431.405958] IPMI message handler: BMC returned incorrect response, expected netfn 5 cmd 2d, got netfn b cmd 23
Feb 28 21:35:47 pr1012 kernel: [107445.388922] {21}[Hardware Error]: Hardware error from APEI Generic Hardware Error Source: 1
Feb 28 21:35:47 pr1012 kernel: [107445.388926] {21}[Hardware Error]: It has been corrected by h/w and requires no further action
Feb 28 21:35:47 pr1012 kernel: [107445.388927] {21}[Hardware Error]: event severity: corrected
Feb 28 21:35:47 pr1012 kernel: [107445.388928] {21}[Hardware Error]:  Error 0, type: corrected
Feb 28 21:35:47 pr1012 kernel: [107445.388929] {21}[Hardware Error]:  fru_text: CorrectedErr
Feb 28 21:35:47 pr1012 kernel: [107445.388930] {21}[Hardware Error]:   section_type: memory error
Feb 28 21:35:47 pr1012 kernel: [107445.388931] {21}[Hardware Error]:   node: 1 device: 0 
Feb 28 21:35:47 pr1012 kernel: [107445.388932] {21}[Hardware Error]:   error_type: 2, single-bit ECC
Feb 28 21:36:48 pr1012 kernel: [107506.435380] {22}[Hardware Error]: Hardware error from APEI Generic Hardware Error Source: 1
Feb 28 21:36:48 pr1012 kernel: [107506.435383] {22}[Hardware Error]: It has been corrected by h/w and requires no further action
Feb 28 21:36:48 pr1012 kernel: [107506.435385] {22}[Hardware Error]: event severity: corrected
Feb 28 21:36:48 pr1012 kernel: [107506.435386] {22}[Hardware Error]:  Error 0, type: corrected
Feb 28 21:36:48 pr1012 kernel: [107506.435388] {22}[Hardware Error]:  fru_text: CorrectedErr
Feb 28 21:36:48 pr1012 kernel: [107506.435389] {22}[Hardware Error]:   section_type: memory error
Feb 28 21:36:48 pr1012 kernel: [107506.435390] {22}[Hardware Error]:   node: 1 device: 0 
Feb 28 21:36:48 pr1012 kernel: [107506.435391] {22}[Hardware Error]:   error_type: 2, single-bit ECC
Feb 28 21:37:48 pr1012 kernel: [107566.455847] {23}[Hardware Error]: Hardware error from APEI Generic Hardware Error Source: 1
Feb 28 21:37:48 pr1012 kernel: [107566.455863] {23}[Hardware Error]: It has been corrected by h/w and requires no further action
Feb 28 21:37:48 pr1012 kernel: [107566.455865] {23}[Hardware Error]: event severity: corrected
Feb 28 21:37:48 pr1012 kernel: [107566.455866] {23}[Hardware Error]:  Error 0, type: corrected
Feb 28 21:37:48 pr1012 kernel: [107566.455867] {23}[Hardware Error]:  fru_text: CorrectedErr
Feb 28 21:37:48 pr1012 kernel: [107566.455869] {23}[Hardware Error]:   section_type: memory error
Feb 28 21:37:48 pr1012 kernel: [107566.455870] {23}[Hardware Error]:   node: 1 device: 0 
Feb 28 21:37:48 pr1012 kernel: [107566.455871] {23}[Hardware Error]:   error_type: 2, single-bit ECC
Feb 28 21:38:04 pr1012 kernel: [107581.710983] IPMI message handler: BMC returned incorrect response, expected netfn 5 cmd 2d, got netfn b cmd 23
Feb 28 21:38:43 pr1012 kernel: [107621.600387] IPMI message handler: BMC returned incorrect response, expected netfn b cmd 23, got netfn b cmd 2d
Feb 28 21:38:49 pr1012 kernel: [107626.878056] {24}[Hardware Error]: Hardware error from APEI Generic Hardware Error Source: 1
Feb 28 21:38:49 pr1012 kernel: [107626.878059] {24}[Hardware Error]: It has been corrected by h/w and requires no further action
Feb 28 21:38:49 pr1012 kernel: [107626.878060] {24}[Hardware Error]: event severity: corrected
Feb 28 21:38:49 pr1012 kernel: [107626.878062] {24}[Hardware Error]:  Error 0, type: corrected
Feb 28 21:38:49 pr1012 kernel: [107626.878063] {24}[Hardware Error]:  fru_text: CorrectedErr
Feb 28 21:38:49 pr1012 kernel: [107626.878064] {24}[Hardware Error]:   section_type: memory error
Feb 28 21:38:49 pr1012 kernel: [107626.878065] {24}[Hardware Error]:   node: 1 device: 0 
Feb 28 21:38:49 pr1012 kernel: [107626.878066] {24}[Hardware Error]:   error_type: 2, single-bit ECC
[... cut ...]
Feb 28 21:59:55 pr1012 kernel: [    0.000000] BIOS-e820: [mem 0x0000000080aa0000-0x0000000080aa0fff] ACPI NVS
Feb 28 21:59:55 pr1012 kernel: [    0.000000] BIOS-e820: [mem 0x0000000080aa1000-0x0000000080aeafff] reserved
Feb 28 21:59:55 pr1012 kernel: [    0.000000] BIOS-e820: [mem 0x0000000080aeb000-0x00000000855fffff] usable
Feb 28 21:59:55 pr1012 kernel: [    0.000000] BIOS-e820: [mem 0x0000000085600000-0x000000008593dfff] reserved
Feb 28 21:59:55 pr1012 kernel: [    0.000000] BIOS-e820: [mem 0x000000008593e000-0x0000000085afdfff] usable

```

knorth55, 2023-02-28:
1回目は同じようにメモリーエラーがでているけど，２回目はちょっと違いそうですね．
ただメモリを新しいものに変えて試してみるというのはありかもしれません．

knorth55, 2023-03-01:
PR2のメモリの規格：
DDR4-2133 ECC Unbuffered DIMM 16GB×2

現在使っているメモリ：
CENTURY MICRO CK16GX2-D4UE2133

@nakane11 @iory この規格にあうメモリを買って試してみるのがいいと思います．

https://docs.google.com/presentation/d/1b-UQShSEY_pswifYKupBrZt8wmQEFET3_E3nsp-V9x8/edit#slide=id.ge9022d6b43_0_10

nakane11, 2023-03-01:
@knorth55 ありがとうございます。

昨日からの作業ログ
- PR1012のSSDを予備にコピー
- 交換した後に電源を付けるとc1の起動が進まない
- 元のSSDを使用しても改善しない

knorth55, 2023-03-01:
ディスプレイをつないでBIOSにはいって，Bootの優先順位があっているかを確認するといいと思います．
PR2のマザボは一回SSDをはずすと，SSDの優先順位が一番下になって，UEFI shellやHDDで起動しようとしてHangすることがあります．