jsk-ros-pkg/jsk_robot #1784: 結局ロボットのROS_HOSTNAMEは何が正解でしょうか?

sktometometo, 2023-03-24:
条件1: ロボットと同じネットワークに繋がったWindowsマシン内WSLからRVizでロボットの状態を確認したい
条件2: ロボットがポケットWi-Fiに繋がった状態でロボット体内のシステムが正常に動いてほしい (cf. https://github.com/jsk-ros-pkg/jsk_robot/issues/1726#issuecomment-1451373384)

例えば、ロボットのIPとHost名は 

```bash
~ $ echo $ROS_HOSTNAME

~ $ echo $ROS_IP

~ $ hostname
BelKa
~ $ ifconfig wlxc006c31b198c
wlxc006c31b198c: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 2312
        inet 133.11.216.165  netmask 255.255.255.0  broadcast 133.11.216.255
        ...
~ $ cat /etc/hosts
127.0.0.1       localhost
127.0.0.1       BelKa

::1             localhost ip6-localhost ip6-loopback
ff02::1         ip6-allnodes
ff02::2         ip6-allrouters
```

となっていて、WSL側は例えば

```bash
~ $ ifconfig eth0
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 172.24.182.158  netmask 255.255.240.0  broadcast 172.24.191.255
```

となっていてネットワークが違う.

```bash
~ $ rossetmaster belka.jsk.imi.i.u-tokyo.ac.jp
set ROS_MASTER_URI to http://belka.jsk.imi.i.u-tokyo.ac.jp:11311
~ $ rostopic info /tf
Type: tf2_msgs/TFMessage

Publishers:
 * /robot_state_publisher (http://BelKa:38367/)
 * /spot/spot_ros (http://BelKa:40767/)

Subscribers: None
```

したがって、ROS_MASTER_URI を FQDN にしても、`BelKa:38367` などは解決できずに通信できないことがある。
by @k-okada

sktometometo, 2023-03-24:
答えとしては、

1. ロボット体内の `ROS_HOSTNAME` をFQDNにする( `export ROS_HOSTNAME=BelKa.jsk.imi.i.u-tokyo.ac.jp` を行う )
2. ロボット体内の `/etc/hosts` に FQDNを加える (以下のように書き加える)
```
127.0.0.1       BelKa BelKa.jsk.imi.i.u-tokyo.ac.jp
```

FQDNでなくても、`BelKa.local` (mDNS) を使って同様の名前解決をすることができる。

@kindsenior @Naoki-Hiraoka はどうしていますか?

sktometometo, 2023-03-24:
https://github.com/clearpathrobotics/robot_upstart/blob/987376f5c49864e9e250e5bb5b88ae13e0ed6973/templates/job-start.em#L62
の部分でhostnameではなく、FQDNなり、`hoge.local` とするか、`.bashrc` に書いておく必要がある。

k-okada, 2023-03-27:
@sktometometo `.local` が動く条件はどうなっているのでしょうか？
以下のようにbelkaは見つかるけど，strelka は見つからない．．

```
k-okada@73A2-SURFACE:~$ nslookup belka.local
Server:         192.168.64.1
Address:        192.168.64.1#53

Non-authoritative answer:
Name:   BelKa.local
Address: 133.11.216.165
Name:   BelKa.local
Address: fe80::f5d9:8862:dd51:d015

k-okada@73A2-SURFACE:~$ nslookup strelka.local
Server:         192.168.64.1
Address:        192.168.64.1#53

** server can't find strelka.local: NXDOMAIN

k-okada@73A2-SURFACE:~$ cat /etc/hosts
# This file was automatically generated by WSL. To stop automatic generation of this file, add the following entry to /etc/wsl.conf:
# [network]
# generateHosts = false
127.0.0.1       localhost
127.0.1.1       73A2-SURFACE.   73A2-SURFACE
```

sktometometo, 2023-03-27:
@k-okada strelkaが起動してないとstrelka.localの名前解決はできないですが起動してますか？

k-okada, 2023-03-27:
動いています
.local 動く条件はなんなんでしょう？

tkmtnt7000, 2023-03-28:
関係ない気もしますが，belkaとstrelkaではホスト名の命名方法が違う気がします．
strelkaはcore-ioになっています．
```bash
# strelka's terminal
spot@core-io:~$ ping strelka
PING StrelKa.jsk.imi.i.u-tokyo.ac.jp (133.11.216.166) 56(84) bytes of data.
64 bytes from core-io.jsk.imi.i.u-tokyo.ac.jp (133.11.216.166): icmp_seq=1 ttl=64 time=0.093 ms
64 bytes from core-io.jsk.imi.i.u-tokyo.ac.jp (133.11.216.166): icmp_seq=2 ttl=64 time=0.088 ms
64 bytes from core-io.jsk.imi.i.u-tokyo.ac.jp (133.11.216.166): icmp_seq=3 ttl=64 time=0.118 ms
64 bytes from core-io.jsk.imi.i.u-tokyo.ac.jp (133.11.216.166): icmp_seq=4 ttl=64 time=0.093 ms
```
```bash
# belka's terminal
spot@BelKa:~$ ping belka
PING BelKa (127.0.0.1) 56(84) bytes of data.
64 bytes from localhost (127.0.0.1): icmp_seq=1 ttl=64 time=0.031 ms
64 bytes from localhost (127.0.0.1): icmp_seq=2 ttl=64 time=0.033 ms
64 bytes from localhost (127.0.0.1): icmp_seq=3 ttl=64 time=0.029 ms
64 bytes from localhost (127.0.0.1): icmp_seq=4 ttl=64 time=0.046 ms

```

k-okada, 2023-03-28:
なるほど，以下で名前解決しました
```
$ nslookup core-io.local
Server:         192.168.64.1
Address:        192.168.64.1#53

Non-authoritative answer:
Name:   core-io.local
Address: 133.11.216.166
Name:   core-io.local
Address: fe80::169d:a1ea:5370:917d
```