jsk-ros-pkg/jsk_robot #1821: develop/fetch での実験・開発について

sktometometo, 2023-06-21:
`develop/fetch` 環境についての説明
( https://github.com/jsk-ros-pkg/jsk_robot/pull/1593 のブランチを基準としたワークスペースやロボット構成 )

## How to contribute (素案)

1. 必ずPull Requestを通じて変更する。
2. Pull Requestは基本的に少なくとも一人以上のレビューを必要とする。レビューしないでマージする場合はPull Requestに理由を記載する。(e.g. 実験のため等)
3. Merge は Rebase (or Squash) で行う。
4. Merge はPRを出した人がする。
5. master にマージするべき機能はmasterにもPull Requestを出す。
6. master にも Pull Request を出した場合は https://github.com/jsk-ros-pkg/jsk_robot/pull/1593 の関連PRリストに追加する。
7. develop/fetch を master にリベースする際は Fetch chat スペースで周知してから行う。実験等で develop/fetch がリベースすると困る場合は見送る。

sktometometo, 2023-06-21:
@k-okada @mqcmd196 @tkmtnt7000 @nakane11

mqcmd196, 2023-06-21:
あらかた同意です．で，今 `/home/fetch/ros/melodic/src` 以下にブランチ切ってコミットすると `git branch -D` されてgit reset --hardされるようで，その仕様までは知りませんでした．なのでこれは普通に `git checkout develop/fetch` にするだけに変更したらいいと思います．

sktometometo, 2023-06-21:
これはもともと `develop/fetch` が diverged した際に remote の `develop/fetch` を確実にcheckoutするための仕様なのだけど、diverged したときのみ branch -D して checkout したい

mqcmd196, 2023-06-21:
diverged したときに `branch -D` する理由って何でしょうか？そのまま `git checkout develop/fetch` したらいいだけだと思いますが

sktometometo, 2023-06-21:
local に develop/fetch あったらリモートの `develop/fetch` checkout 出来ないよね

sktometometo, 2023-06-21:
`update_workspace.sh` の仕様と毎日どういう機能を回すかはちょっと話変わるから https://github.com/jsk-ros-pkg/jsk_robot/pull/1808 かあたらしい issue で話そう

mqcmd196, 2023-06-21:
すいません，

> local に diverge 前の develop/fetch あったらリモートの develop/fetch checkout 出来ないよね

は正しくて，言いたかったことは，今起きている状況として

1. `/home/fetch/ros/melodic/src` で自分でなにかブランチを切って作業してコミットする
2. するとcronで自分が作ったブランチがすべて消去され，resetされてしまう
という状況が起きていて，別にブランチは残ったままで，ただ `git checkout develop/fetch` したらいいだけじゃないか，ということでした．間違えて `develop/fetch` にコミットしてしまった，というものについてはstashしたらいいんじゃないかと思います