jsk-ros-pkg/jsk_robot #1831: Smach_image_publisher shows error during kitchen demo.

sktometometo, 2023-06-27:
`smach_image_publisher` shows error during kitchen demo.

```
[ERROR] [1687845042.887504] [/smach_image_publisher:rosout]: bad callback: <bound method SmachImagePublisher._status_msg_update of <__main__.SmachImagePublisher object at 0x7f05a4f9a790>>
Traceback (most recent call last):                                                   
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback                
    cb(msg)                                                                               
  File "/home/fetch/ros/melodic/src/ros-visualization/executive_smach_visualization/smach_viewer/src/smach_viewer/smach_viewer_base.py", line 489, in _status_msg_update
    if container.update_status(msg):                                                                                                     
  File "/home/fetch/ros/melodic/src/ros-visualization/executive_smach_visualization/smach_viewer/src/smach_viewer/smach_viewer_base.py", line 101, in update_status
    self._local_data._data = self._load_local_data(msg)                                                             
  File "/home/fetch/ros/melodic/src/ros-visualization/executive_smach_visualization/smach_viewer/src/smach_viewer/smach_viewer_base.py", line 75, in _load_local_data
    local_data = pickle.loads(msg.local_data.encode('utf-8'))                        
UnicodeDecodeError: 'ascii' codec can't decode byte 0x88 in position 11: ordinal not in range(128)    
```

tkmtnt7000, 2023-06-27:
I have fixed at https://github.com/ros-visualization/executive_smach_visualization/pull/54, but not merged yet...

tkmtnt7000, 2023-06-27:
I'll switch fetch's robot internal branch to bugfix branch.

closed 2023-06-27 by tkmtnt7000

k-okada, 2023-06-28:
merged
https://github.com/ros-visualization/executive_smach_visualization/pull/54
--
◉ Kei Okada

PR #1832 (2023-06-27): [Fetch] Use bugfix branch of executive_smach_visualization
This PR closes #1831.

Until https://github.com/ros-visualization/executive_smach_visualization/pull/54 is merged.