jsk-ros-pkg/jsk_robot #1838: mongo_record.pyの中でrospy.Rateで指定されたHzでサブスクライブされていない

tkmtnt7000, 2023-07-03:
edgetpuの認識結果が10Hzも入っているのはやっぱり何かおかしいと思って調べていくと

https://github.com/jsk-ros-pkg/jsk_robot/blob/5093a5acd7293f3bc96c5014ed6d60c7dc0d6d5b/jsk_robot_common/jsk_robot_startup/src/jsk_robot_startup/lifelog/mongo_record.py#L84-L88

でrate指定でspinonceされているように見せかけておいて実はされていない，ということに気づいた．（おそらく裏でスレッドか何かが走っているので）

毎回サブスクライバをunregisterしてやると直りそうだが，不完全な気もする．

sktometometo, 2023-07-03:
rospy の サブスクライバは roscpp や euslisp と違って、サブスクライバやサーバーのハンドラごとにスレッドが立っていて、msgがくるとmainスレッドでスリープしているかどうかにかかわらず callback が呼ばれるので、mongo_record.py のこの書き方は subscribe のレートを指定する意味での書き方ではないけど、そもそも何が問題なのか聞いてもいい?

もし subscribe して insert する部分を一定のHzにしたいのであれば throttle するか record ノード側で throttle する機構を入れるかする必要があるはず。

tkmtnt7000, 2023-07-03:
self.update_rateで設定されている値がサブスクライブのレートを指定していると勘違いしました．
よく読むとupdate_rateで管理しようとしているのはspinonceの前のcheck_topicのほうですね．．．

closed 2023-07-04 by sktometometo