jsk-ros-pkg/jsk_robot #1846: [pepper/cross] ros1-pepper.tarを使って`./build_ros1_dependencies.sh`から実行することはできますでしょうか

kochigami, 2023-08-03:
`./prepare_requirements_ros1.sh`が以下の行で失敗するのですが、
この状況で`./build_ros1_dependencies.sh`を実行することは可能でしょうか。
そのために、[バックアップファイル](https://drive.google.com/drive/u/0/folders/10rINVGt1iDM2WNofmf0sZBX_iTnpXya6) として用意されている `ros1-pepper.tar`を使うことはできますでしょうか。

### `./prepare_requirements_ros1.sh`が以下の行で失敗 (Ubuntu 18.04, 20.04の両方で確認)

https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/cross/docker/Dockerfile_ros1#L18C5-L18C71

```
#12 [ 9/23] RUN apt-key adv --keyserver keyserver.ubuntu.com --recv-key 0xCBF125EA
#12 0.243 Executing: /tmp/tmp.vin0xo5Yc7/gpg.1.sh --keyserver
#12 0.243 keyserver.ubuntu.com
#12 0.243 --recv-key
#12 0.243 0xCBF125EA
#12 0.244 gpg: requesting key CBF125EA from hkp server keyserver.ubuntu.com
#12 120.2 gpg: keyserver timed out
#12 120.2 gpg: keyserver receive failed: keyserver error
#12 ERROR: process "/bin/sh -c apt-key adv --keyserver keyserver.ubuntu.com --recv-key 0xCBF125EA" did not complete successfully: exit code: 2
------
 > [ 9/23] RUN apt-key adv --keyserver keyserver.ubuntu.com --recv-key 0xCBF125EA:
#12 0.243 Executing: /tmp/tmp.vin0xo5Yc7/gpg.1.sh --keyserver
#12 0.243 keyserver.ubuntu.com
#12 0.243 --recv-key
#12 0.243 0xCBF125EA
#12 0.244 gpg: requesting key CBF125EA from hkp server keyserver.ubuntu.com
#12 120.2 gpg: keyserver timed out
#12 120.2 gpg: keyserver receive failed: keyserver error
------
Dockerfile_ros1:18
--------------------
  16 |     #
  17 |     RUN echo "deb http://snapshots.ros.org/kinetic/final/ubuntu `lsb_release -cs` main" > /etc/apt/sources.list.d/ros-latest.list
  18 | >>> RUN apt-key adv --keyserver keyserver.ubuntu.com --recv-key 0xCBF125EA
  19 |     RUN apt-get update
  20 |     # RUN apt-get install -y gcc g++ gcc-4.9 g++-4.9 cpp-4.9 make  # build-essential
--------------------
ERROR: failed to solve: process "/bin/sh -c apt-key adv --keyserver keyserver.ubuntu.com --recv-key 0xCBF125EA" did not complete successfully: exit code: 2
```

### この状況で `./build_ros1_dependencies.sh` を実行すると、

前のコマンドが失敗しているので、
`Unable to find image 'ros1-pepper:i386' locally` となる。

k-okada, 2023-08-07:
.tar ファイルがあると
> docker load < ros1-pepper.tar
として、イメージを復元できると思います。
``
ros1-pepper                  i386                          6b910da52032        3 months ago        503MB
```
（ https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_naoqi_robot/cross#development )

> apt-key adv --keyserver keyserver.ubuntu.com --recv-key 0xCBF125EA

については、docker とか os とか寒けなく、基本はネットワークの問題（自分の場合も、keyserver.ubuntu.comの場合も）なので、そちらを確認するのがよいです。

kochigami, 2023-08-07:
ありがとうございます。dockerに不慣れなだけでした。
`ros1-pepper.tar`を読み込ませることができました。

ros1-pepper.tarを使う場面がよく分かっていなかったため、
https://github.com/jsk-ros-pkg/jsk_robot/pull/1767 に追記しようと思います。

```
kochigami@kochigami-pc:~/catkin_ws/src/jsk_robot/jsk_naoqi_robot/cross$ docker image ls
REPOSITORY                   TAG       IMAGE ID       CREATED         SIZE
...
ros1-pepper                  i386      e3f06b009d8b   11 months ago   503MB
…
```
```
kochigami@kochigami-pc:~/catkin_ws/src/jsk_robot/jsk_naoqi_robot/cross$ docker image rm ros1-pepper:i386
Untagged: ros1-pepper:i386
Deleted: sha256:e3f06b009d8bd5469bf94215363b33c8c28c388ded8e21d40da23038fb7cc334
Deleted: sha256:2fc8dcd0511dbfc0184c7e3675d5a3af60fd9ab946ecb234e058ad3dbabcf336
Deleted: sha256:77050a82111f8798cc73b03e50b5bf860f015b5d477c37c3b7fbbdbc27791ffa
Deleted: sha256:136146960c05c7892e1f87c09b3d340f41a188a0ec27723cfd8528a73def0daf
Deleted: sha256:a32b8dde57cbd3dd9d13276b17bdfaa9f5009d2c0d49f1ad4bfd7b2aaf3d2c71
Deleted: sha256:859cf16a15b530738c48f5d256f5bcc22ce43275e49ecd74031627b8b31863c3
Deleted: sha256:efc05090e27c1a0a7bca74934c0d370884c86157b160df4634d259e8ff9a3f91
Deleted: sha256:9a1e29c37fb26c8f76f429ed8ed53e9325636e2afa6ce0de29c8ce2f99a0d801
Deleted: sha256:0e7fd718479ea9739d0e9d89b3e3eb5a142e4096c0c03968349d2a143f697a07
Deleted: sha256:937611f6b0b1e016f65a8d6d67732cc0daa62580d2aea4c9b39f626734f32602
Deleted: sha256:09e4433ed28c34ef84c70a0c780793317e83a420ef5eac1a1b96505170bbfe90
Deleted: sha256:df559cb2fce30ffaec242026d4cf7ea03658d8ac6c721f62b791a09f09a94445
Deleted: sha256:391213d250eb14189ca203131ce665db702057e265ccfb914c74d018f0e84529
Deleted: sha256:4950909238a2dedce1f38f1f12b4bcd2efd0e17709482c0762173d2ee426b0d7
Deleted: sha256:0362ada4fa493f31d136a6aadf3d20688f3dfe75b7fe33c9bbe288c09d27cac6
Deleted: sha256:73cf71ffeebd09a9a521db5d250253b9fabd3603e1a5238249220053ed44ff39
Deleted: sha256:4a129a58935fcec9d98f8e72baf17b0300a53d10aa305ca04eba0349cd3018ba
Deleted: sha256:9494d125612991cc6d86da4f59289449ea6fb7a697cad913e1143ebdf91e564f
Deleted: sha256:37fdee1c16a23c9402d8b7ae2ae802f506c80e8b3b909211c1acea8ac66f29f2
Deleted: sha256:84973ef8caa466c25cd5a4a7eaddb7984b8d57e5902883621ab2abd9c80a78c3
Deleted: sha256:2bde1f65bdff9bf6e41124f93dfdfa5f7b47e21a0c4dc53fb3509bbb81d8de20
Deleted: sha256:b9bf2140d251e4692fb1d8f7cd37ddb183c60c96ac6f75896c4f6b5c94341022
Deleted: sha256:f301b8ab1228ac7367d993ae1b250dc58dbd7176fcaf5e7c52dc12e2d0b9bcf2
```
```
kochigami@kochigami-pc:~/catkin_ws/src/jsk_robot/jsk_naoqi_robot/cross$ docker image ls
REPOSITORY                   TAG       IMAGE ID       CREATED        SIZE
(ros1-pepperはない)
```

ros1-pepperがない状態で
```
kochigami@kochigami-pc:~/catkin_ws/src/jsk_robot/jsk_naoqi_robot/cross$ docker load < ros1-pepper.tar
4a74e195e076: Loading layer  27.96MB/27.96MB
b3fd581f5e7a: Loading layer  14.19MB/14.19MB
7610ac36276d: Loading layer  1.979MB/1.979MB
2735446335d6: Loading layer  3.072kB/3.072kB
85df60a2921a: Loading layer  71.13MB/71.13MB
abbea770ee6f: Loading layer  612.4kB/612.4kB
839fbe824f98: Loading layer   16.9kB/16.9kB
8b4e79c891f9: Loading layer   34.3kB/34.3kB
0bd99eb36b64: Loading layer  1.931MB/1.931MB
b120e95cdc83: Loading layer  218.5MB/218.5MB
aece80ff8054: Loading layer  42.78MB/42.78MB
93f4f4c823b9: Loading layer  1.102MB/1.102MB
b5fad9cefdde: Loading layer  11.26kB/11.26kB
187a26bb3257: Loading layer  1.869MB/1.869MB
1dbc91300548: Loading layer  1.733MB/1.733MB
3fda8661770d: Loading layer  840.2kB/840.2kB
eef395cd0f47: Loading layer  1.901MB/1.901MB
adc16646c5db: Loading layer  825.3kB/825.3kB
d5a37f1aa252: Loading layer  343.6kB/343.6kB
5f70bf18a086: Loading layer  1.024kB/1.024kB
48f82157d065: Loading layer  3.072kB/3.072kB
Loaded image: ros1-pepper:i386
```
```
kochigami@kochigami-pc:~/catkin_ws/src/jsk_robot/jsk_naoqi_robot/cross$ docker image ls
REPOSITORY                   TAG       IMAGE ID       CREATED         SIZE
...
ros1-pepper                  i386      e3f06b009d8b   11 months ago   503MB
...
```
最後まで確認していないが、`./build_ros1_dependencies.sh`も実行可能になった。

closed 2023-08-07 by kochigami