jsk-ros-pkg/jsk_robot #1849: [naoqi/cross] jsk_naoqi_robot/cross 環境ではどこで Python NAOqi SDK を読み込んでいるのでしょうか

kochigami, 2023-08-30:
jsk_pepper_startup (またはpepper_bringup) では、
ロボットの関節角を動かすために `naoqi_pose/nodes/pose_controller.py` が必要で、
以下で `from naoqi_driver.naoqi_node import NaoqiNode` しています。

https://github.com/ros-naoqi/naoqi_bridge/blob/master/naoqi_pose/nodes/pose_controller.py#L49

これは、`naoqi_driver_py/src/naoqi_driver/naoqi_node.py` のことで、
その中では、
```
# import Aldebaran API (must be in PYTHONPATH):
try:
    from naoqi import ALProxy
```
しています。
https://github.com/ros-naoqi/naoqi_bridge/blob/master/naoqi_driver_py/src/naoqi_driver/naoqi_node.py#L33-L35

そのため、`pose_controller.py` を使うためにはPython NAOqi SDK (https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/README.md?plain=1#L73-L74)  が必要なはずなのに、
`jsk_naoqi_robot/cross` のソースコードにはそれを読み込んでいる箇所が見つからず、不思議に思って書かせていただきました。

k-okada, 2023-09-01:
Pepper の中に最初からはいっているnaoqiを使っているようです．
```
Pepper [0] ~/User $ source user_setup.bash 
Pepper [0] ~/User $ python
Python 2.7.17 (default, Sep  1 2022, 10:28:56) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import naoqi
>>> naoqi
<module 'naoqi' from '/opt/aldebaran/lib/python2.7/site-packages/naoqi.py'>
```
一方Cのライブラリの方はcrossでコピーしているのかな．
```
Pepper [0] ~/User/devel/lib/naoqi_driver $ ldd naoqi_driver_node| grep libqi.so
        libqi.so => /home/nao/User/devel/lib/libqi.so (0xb55f7000)
Pepper [0] ~/User/devel/lib/naoqi_driver $ ls -al /opt/aldebaran/lib/libqi.so 
-rw-r--r-- 1 root root 4060508 Nov 28  2016 /opt/aldebaran/lib/libqi.so
```

上の情報はバジルの状態で，これはcross環境自体が開発中のものだったから最新のは違うかも．
@kochigami の環境でも確認してもらえると嬉しいです．

kochigami, 2023-09-01:
ありがとうございます。
NAOqiのバージョン：2.5.11.13   
ハードウェアのバージョン：1.8a
で調べました。

- naoqi.py: 同様に、Pepperの中に最初から入っている naoqi が使われていました。
```
Pepper [0] ~ $ cd User/
Pepper [0] ~/User $ python                 
Python 2.7.6 (default, Jan 28 2017, 00:13:02) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import naoqi
>>> naoqi
<module 'naoqi' from '/opt/aldebaran/lib/python2.7/site-packages/naoqi.py'>
```

- C++ の naoqi に関しては、こちらは `/opt`以下にあるものを使っていました。
```
Pepper [0] ~/User/devel/lib/naoqi_driver $ ldd naoqi_driver_node| grep libqi.so
	libqi.so => /opt/aldebaran/lib/libqi.so (0xb738e000)
Pepper [0] ~/User/devel/lib/naoqi_driver $ ls -al /opt/aldebaran/lib/libqi.so
-rw-r--r-- 1 root root 4060508 Sep  6  2018 /opt/aldebaran/lib/libqi.so
```

- `libqi.so`自体は、`~/User/devel/lib`以下にコピーはされていました。
```
Pepper [0] ~/User/devel/lib $ ls
joy                               libnaoqi_driver_module.so  naoqi_driver
jsk_robot_startup                 libqi.so                   pkgconfig
libcompressed_image_transport.so  libqicore.so               python2.7
libnaoqi_driver.so                libteleop_twist_joy.so     teleop_twist_joy
```
まだ完全に理解できた訳ではないのですが、最初の疑問は解決したためクローズします。
ありがとうございました。

closed 2023-09-01 by kochigami