jsk-ros-pkg/jsk_robot #1850: [jsk_naoqi_robot/cross] jsk_naoqi_robot/cross 環境はNAOでも使えますでしょうか

kochigami, 2023-08-30:
今後、jsk_naoqi_robot/cross を NAOでも使えるといいなと思い、
現状のプログラムでどこを直したら良さそうか書いてみたのですが、
コンパイルして実機にコピーし、試す前に気をつけたら良さそうなことがございましたらご教示ください。

直したら良さそうな部分：
1. ロボット体内で立ち上げるプログラム (jsk_nao_startup.launchに直す必要がある) 
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/cross/startup_scripts/start.sh#L7

2. 引数 TARGET_ROBOT の変更
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/cross/build_user.sh#L10-L12

3. pepper_meshes をコピーしている部分（nao_meshes もコピーする必要がある）
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/cross/build_user.sh#L44-L53

4. app_manager で立ち上がるPepper用に作ったアプリの画像コピー（多分そのままで大丈夫）
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/cross/install.sh#L87

確認が必要な部分：
- ペッパーとナオのパソコンのバージョンを確認する

k-okada, 2023-09-01:
NaoとPepperの体内のPCが同じだったら基本は問題ないと思うけど、それぞれで
> uname -a
> lsb_release -a
> cat /proc/cpuinfo 
みたいなのをそれぞれ試してみてください。

もしおなじだったら、NaoもPepperも同じcross で動くと良いので、、、
１）nao関連のパッケージ(meshes)などもコピーする
２）nao/pepper を自動で判別する方法を見つける /etc の下とかにJuliette みたいな文字列のあるファイルがないかな？あれば、start.sh や install.sh は切り替える
３）TARGET_ROBOT はビルド時に手動で指定するのが良い中（つまりできたUser はNao用・Pepper用になってしまう。あるいは2つのロボットに対応できるようにTARGET_ROBOTをリストを受け取るようにする）

kochigami, 2023-09-01:
ありがとうございます。

Pepper: NAOqi 2.5.11.13
NAO V6: NAOqi 2.8
で調べました。

> uname -a
> lsb_release -a
> cat /proc/cpuinfo

そのまま貼ってよい情報なのか分からないため、一旦ファイルとして共有いたします。
https://docs.google.com/document/d/1rYese8Or2nFNKyvMKNXaQYr7Y7e0CoUi/edit?usp=sharing&ouid=102048303683139089959&rtpof=true&sd=true

結論としては、体内のPCが同じなのでNAO (V6) もjsk_naoqi_robot/cross 環境が使えそうです。
http://doc.aldebaran.com/2-8/family/nao_technical/motherboard_naov6.html
http://doc.aldebaran.com/2-5/family/pepper_technical/motherboard_pep.html

> １）nao関連のパッケージ(meshes)などもコピーする
こちら、承知いたしました。

> ２）nao/pepper を自動で判別する方法を見つける /etc の下とかにJuliette みたいな文字列のあるファイルがないかな？あれば、start.sh や install.sh は切り替える

探し方が悪いのかもしれないのですが、見つかりませんでした。
もう少し調べます。

１つ思いついたのは、
ロボットの体内で
`qicli call ALSystem.systemVersion` すると、NAOqi OSのバージョンが取ってこれそうです。
http://doc.aldebaran.com/2-5/naoqi/core/alsystem-api.html#ALSystemProxy::systemVersion

NAO V4, V5のNAOqi OSバージョンを忘れてしまったのですが（多分 2.1 だったはずです・・・）、
NAO V6: 2.8
Pepper for Education: 2.5
のため、見分けられそうです。

一様hostnameで見分けられるのかな、とも思ったのですが、
```
nao [0] /etc $ cat hostname 
nao
```
```
Pepper [0] /etc $ cat hostname 
Pepper
```

以下のような「ロボットの名前を変える」メソッドがあり、これを実行すると、hostnameが変更されると思うので、駄目そうです。
http://doc.aldebaran.com/2-5/naoqi/core/alsystem-api.html#ALSystemProxy::setRobotName__ssCR

３）TARGET_ROBOT はビルド時に手動で指定するのが良い中（つまりできたUser はNao用・Pepper用になってしまう。あるいは2つのロボットに対応できるようにTARGET_ROBOTをリストを受け取るようにする）

２つのロボットに対応していた方がビルドし直しの手間が無くせるのかな、という気がします。
まずは、それができるか考えて、難しそうだったらNAO用、Pepper用をその都度つくるようにしようと思います。

9月中旬にNAOを使う用事があるため、それが一段落してから取り組もうと思います。

k-okada, 2023-09-06:
なるほど、体内のCPUがハードウェアとしては同じなんだけど、
> Pepper: Linux Pepper 4.0.4-rt1-aldebaran-rt-00002-g2edd9f0 #1 SMP PREEMPT RT Mon Oct 30 18:18:58 Local time zone must be se i686 Intel(R) Atom(TM) CPU E3845 @ 1.91GHz GenuineIntel GNU/Linux
> Nao: Linux nao 4.4.86-rt99-aldebaran #1 SMP PREEMPT RT Fri Mar 9 17:19:39 CET 2018 x86_64 x86_64 x86_64 GNU/Linux

Pepper は i686  (32bit)で、Naoは x86_64 (64bit)　になっているのが不思議。Pepperはi686 だからこそ、
 `ctc-linux64-atom-2.5.10.7/bin/i686-aldebaran-linux-gnu-gcc`
を使ってクロスコンパイルしたので、そう考えると本当は
`ctc-linux64-atom-2.8.4.2/bin/x86_64-aldebalan-linux-gnu-gcc`
みたいな環境が必要、という理屈になります。

とはいえ、ふつうはx86_64の64bit CPUは32bitのコードは動かしてくれる(*)ので、まずは pepper/cross の環境でcross した環境を、Nao体内へコピーして動くか。Pythonのコードは動くと思うけど、Cのコードが動くか？例えば、なんかのライブラリがありませんとか、セグフォするとか、そういう事が無いか確認しましょう。

*: 以下のように自分のPCでも確認できます
```
$ sudo  apt install gcc-7-i686-linux-gnu
$ i686-linux-gnu-gcc -o hello hello.c
$ ./hello
hello world
```