jsk-ros-pkg/jsk_robot #1851: [aibo_selenium_ros] seleniumを使うとログインできない

a-ichikura, 2023-09-07:
https://github.com/jsk-ros-pkg/jsk_robot/pull/1798#issue-1698908494

seleniumを利用してchrome webdriverを自動起動後、手動ログインしようとするとサインインに失敗する。
メールアドレス、パスワードを入力後、bot対策のクイズに答えたあとにログインできず入力したパスワードが消えてしまう。

ブラウザ(firefox, chrome)を手動で立ち上げ、同じように[myaibo](https://myaibo.aibo.sony.jp/#/)にログインすることは可能。

少なくとも8月20日頃まではサインインできていて、
9月7日現在サインインできなくなった。
→SONYのセキュリティーが変わったのかも（bot対策用クイズが8月の頃と変わっていたので…)

@sktometometo

a-ichikura, 2023-09-07:
追加情報

webdriverをfirefoxにしてみましたが、
同じようにサインインしようとすると
"サーバーとの接続を確立できませんでした。"
と表示されました。

２回試して同じエラーメッセージが出たので、my aiboのサイトでブロックが働いていそうです。

mqcmd196, 2023-09-07:
とりあえず別のPCで試してみるのはありそうです

a-ichikura, 2023-09-09:
とりあえず、自分の新しいデスクトップで試して見た結果

・google chromeでやろうとしたら、chromeのversionが**116.0.5845.179**で、有効なchromedriverのversionが**116.0.5845.96**であっていないためかうまくいきませんでした [参考](https://googlechromelabs.github.io/chrome-for-testing/)

・firefoxのdriverで試してみたらmy aiboのサイトを開くことはできましたがaibo用pcと同じように「サーバーとの接続を確立できませんでした」となりました

ちなみにデスクトップでも、seleniumを経由せず検索から手動でログインする場合は正常にログインすることができます。

sktometometo, 2023-09-09:
`navigator.webdriver` をはじめとする, botであることを示す項目が隠蔽されていないので検知されているのかと考えています。

オリジナルのWebDriverクラスではなく、 https://github.com/ultrafunkamsterdam/undetected-chromedriver を使うと動作しました。 `PR/add-jsk-aibo-robot` の最新版をpullして使ってみてください。

@a-ichikura @sawada10

sktometometo, 2023-09-09:
あと、現在の実装ではchromeではなく `rosdep install` で入る chromium と chromium-chromedriver を使用しているので、_webdriver オプションで指定せずに立ち上げて動作するはずなのでそちらも試してみてください。(README.mdを更新しました)

a-ichikura, 2023-09-15:
デバックしていただき動くようになりました。

ただし注意点：
chrome-driverとchrome-browserのversionを合わせる必要があり、aiboのPCはどちらも115なので
[undetected-chrome-driver](https://github.com/ultrafunkamsterdam/undetected-chromedriver) もバージョンを合わせる必要がありました。
それに関するissue：https://github.com/ultrafunkamsterdam/undetected-chromedriver/issues/1564#issuecomment-1720271517
を参考に、`__init__.py`の中に`version_main=115`を追加しました。

closed 2023-10-16 by sktometometo