jsk-ros-pkg/jsk_robot #1857: [Unitree] Failed to run `make system` under `cross`

sktometometo, 2023-10-05:
I have run `make system` under `cross` to run cross compile for development and got an error during it.

```bash
~/ros/ws_jsk_unitree/src/jsk_robot/jsk_unitree_robot/cross $ make system
./prepare_requirements_ros1.sh
+ docker buildx
+ '[' -e /proc/sys/fs/binfmt_misc/qemu-aarch64 ']'
++ docker images -q multiarch/qemu-user-static
+ '[' '!' 3539aaa87393 ']'
+ tee arm64v8_prepare_requirements_ros1.log
++ id -u sktometometo
+ docker buildx build --progress plain -t ros1-unitree:arm64v8 --build-arg TARGET_MACHINE=arm64v8 --build-arg UID=1000 -f docker/Dockerfile_ros1 docker/
#0 building with "default" instance using docker driver

#1 [internal] load build definition from Dockerfile_ros1
#1 transferring dockerfile: 2.07kB done
#1 DONE 0.0s

#2 [internal] load .dockerignore
#2 transferring context: 2B done
#2 DONE 0.0s

#3 [internal] load metadata for docker.io/arm64v8/ubuntu:bionic
#3 DONE 1.9s

#4 [ 1/26] FROM docker.io/arm64v8/ubuntu:bionic@sha256:f97a5103cca28097326814718e711c9c41b54853c26959d73495e40b1dd608f2
#4 DONE 0.0s

#5 [internal] load build context
#5 transferring context: 75B done
#5 DONE 0.0s

#6 [14/26] COPY ros-packages.txt .
#6 CACHED

#7 [ 5/26] RUN update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
#7 CACHED

#8 [ 7/26] RUN apt-get install -y --no-install-recommends apt-transport-https
#8 CACHED

#9 [12/26] RUN echo 'debconf debconf/frontend select Noninteractive' | debconf-set-selections
#9 CACHED

#10 [10/26] RUN curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | apt-key add -
#10 CACHED

#11 [22/26] RUN useradd --uid 1000 -ms /bin/bash -G sudo user
#11 CACHED

#12 [ 4/26] RUN locale-gen en_US en_US.UTF-8
#12 CACHED

#13 [ 2/26] RUN apt-get update
#13 CACHED

#14 [ 9/26] RUN apt-get install -y --no-install-recommends curl gnupg2
#14 CACHED

#15 [21/26] RUN apt-get install -y sudo
#15 CACHED

#16 [24/26] RUN echo user:user | chpasswd
#16 CACHED

#17 [16/26] RUN apt install -y --no-install-recommends $(cat ros-packages.txt | grep -v ^#)
#17 CACHED

#18 [ 8/26] RUN echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" | tee /etc/apt/sources.list.d/ros-latest.list
#18 CACHED
[... 208 lines cut ...]
Downloaded tarball from 'http://archive.ubuntu.com/ubuntu/pool/universe/s/speech-tools/speech-tools_2.5.0.orig.tar.gz' and unpacked it
=== src/speech_tools/debian (tar) ===
Downloaded tarball from 'http://archive.ubuntu.com/ubuntu/pool/universe/s/speech-tools/speech-tools_2.5.0-4.debian.tar.xz' and unpacked it
=== src/tcl (tar) ===
Downloaded tarball from 'http://archive.ubuntu.com/ubuntu/pool/universe/t/tcl8.5/tcl8.5_8.5.19.orig.tar.gz' and unpacked it
=== src/tcl/debian (tar) ===
Downloaded tarball from 'http://archive.ubuntu.com/ubuntu/pool/universe/t/tcl8.5/tcl8.5_8.5.19-4.debian.tar.xz' and unpacked it
=== src/tinyxml (tar) ===
Downloaded tarball from 'http://archive.ubuntu.com/ubuntu/pool/universe/t/tinyxml/tinyxml_2.6.2.orig.tar.gz' and unpacked it
=== src/tinyxml/debian (tar) ===
Downloaded tarball from 'http://archive.ubuntu.com/ubuntu/pool/universe/t/tinyxml/tinyxml_2.6.2-4.debian.tar.xz' and unpacked it
=== src/urdfdom (tar) ===
Downloaded tarball from 'http://archive.ubuntu.com/ubuntu/pool/universe/u/urdfdom/urdfdom_1.0.0.orig.tar.gz' and unpacked it
=== src/urdfdom-headers (tar) ===
Downloaded tarball from 'http://archive.ubuntu.com/ubuntu/pool/universe/u/urdfdom-headers/urdfdom-headers_1.0.0.orig.tar.gz' and unpacked it
=== src/urdfdom-headers/debian (tar) ===
Downloaded tarball from 'http://archive.ubuntu.com/ubuntu/pool/universe/u/urdfdom-headers/urdfdom-headers_1.0.0-1.debian.tar.xz' and unpacked it
=== src/urdfdom/debian (tar) ===
Downloaded tarball from 'http://archive.ubuntu.com/ubuntu/pool/universe/u/urdfdom/urdfdom_1.0.0-2build2.debian.tar.xz' and unpacked it
=== src/yaml-cpp (tar) ===
Downloaded tarball from 'http://archive.ubuntu.com/ubuntu/pool/universe/y/yaml-cpp/yaml-cpp_0.5.2.orig.tar.gz' and unpacked it
=== src/yaml-cpp/debian (tar) ===
Downloaded tarball from 'http://archive.ubuntu.com/ubuntu/pool/universe/y/yaml-cpp/yaml-cpp_0.5.2-4ubuntu1.debian.tar.xz' and unpacked it
make: *** [Makefile:14: system] エラー 1
```

tkmtnt7000, 2023-10-06:
I faced this problem yesterday, but retrying make clean and make system have made it solved in my case.

sktometometo, 2023-10-06:
That does not work for my case.

k-okada, 2023-10-06:
you may try increasing retry number and run `make system` many times without `make clean
` https://github.com/jsk-ros-pkg/jsk_robot/blob/9a1da3dccba70a646dc42c81f911ef2dd2a5475c/jsk_unitree_robot/cross/build_ros1_dependencies.sh#L41

But, even if you do not have arm64 native machine, we *strongly recommend* to use arm64v8_System.tgz

sktometometo, 2023-10-06:
> But, if you do not have arm64 native machine, we strongly recommend to use arm64v8_System.tgz, sometimes it takes 24 hour to build system.

Thanks, I will do that.