jsk-ros-pkg/jsk_robot #1858: [Unitree] duplicated ROS packages in environment

sktometometo, 2023-10-06:
I am trying to create a ROS workspace by following this tutorial https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_unitree_robot#setup-environment

But I have got error below during `rosdep install -y -r --from-paths . --ignore-src
`

```bash
~/ros/ws_jsk_unitree/src $ rosdep install -y -r --from-paths . --ignore-src

WARNING: Package name "aliengoZ1_description" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.

ERROR: Rosdep experienced an error: Multiple packages found with the same name "unitree_legged_msgs":
- unitree_ros/unitree_ros_to_real/unitree_legged_msgs
- unitree_ros_to_real/unitree_legged_msgs
Multiple packages found with the same name "unitree_legged_real":
- unitree_ros/unitree_ros_to_real/unitree_legged_real
- unitree_ros_to_real/unitree_legged_real
Multiple packages found with the same name "unitree_legged_sdk":
- unitree_legged_sdk
- unitree_ros/unitree_ros_to_real/unitree_legged_sdk
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.22.2

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 146, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 446, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python3/dist-packages/rosdep2/main.py", line 501, in _package_args_handler
    pkgs = find_catkin_packages_in(path, options.verbose)
  File "/usr/lib/python3/dist-packages/rosdep2/catkin_packages.py", line 35, in find_catkin_packages_in
    packages = find_packages(path)
  File "/usr/lib/python3/dist-packages/catkin_pkg/packages.py", line 103, in find_packages
    raise RuntimeError('\n'.join(duplicates))
RuntimeError: Multiple packages found with the same name "unitree_legged_msgs":
- unitree_ros/unitree_ros_to_real/unitree_legged_msgs
- unitree_ros_to_real/unitree_legged_msgs
Multiple packages found with the same name "unitree_legged_real":
- unitree_ros/unitree_ros_to_real/unitree_legged_real
- unitree_ros_to_real/unitree_legged_real
Multiple packages found with the same name "unitree_legged_sdk":
- unitree_legged_sdk
- unitree_ros/unitree_ros_to_real/unitree_legged_sdk
```

There are duplicated ROS packages.

sktometometo, 2023-10-06:
This is because [unitree_ros](https://github.com/unitreerobotics/unitree_ros) has [unitree_ros_to_real](https://github.com/unitreerobotics/unitree_ros_to_real) as a submodules at [this commit](https://github.com/unitreerobotics/unitree_ros/commit/2e566b78bcdabd4d99be1a0ab7c29cf67e96673b)

k-okada, 2023-10-06:
please use `50ff982dfbd4` of unitree_ros

```
src/unitree_ros                    git <detached>  (-)     50ff982dfbd4 git://github.com/unitreerobotics/unitree_ros.git
```

```
k-okada@p51s:~/catkin_ws/ws_unitree/src/unitree_ros$ git log --oneline
50ff982 (HEAD, k-okada/noetic) support GAZEBO_MAJOUR_VERSION >= 10 (noetic)
49e936b reconfirm inertia params
7325aa5 increase simulation frequency to 5000Hz
```

closed 2023-10-06 by k-okada