jsk-ros-pkg/jsk_robot #1860: [Unitree] Failed to run `make user`

sktometometo, 2023-10-06:
I am trying to do cross-compile following [this instruction](https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_unitree_robot#program-go1-robot) and when running `make user`, I have got following error.

```bash
~/ros/ws_jsk_unitree/src/jsk-ros-pkg/jsk_robot/jsk_unitree_robot/cross $ make user
./build_user.sh
+ mkdir -p arm64v8_User/src/jsk_robot
+ rsync -avzh --delete --exclude 'jsk_unitree_robot/cross*' --exclude go1.l --exclude go1-simple.l 

sent 314.65M bytes  received 30.60K bytes  29.97M bytes/sec
total size is 399.28M  speedup is 1.27
++ find arm64v8_User/src/jsk_robot -maxdepth 1 -mindepth 1 -type d
+ for dir in $(find ${SOURCE_ROOT}/src/jsk_robot -maxdepth 1 -mindepth 1 -type d)
+ [[ ! arm64v8_User/src/jsk_robot/doc =~ jsk_unitree_robot|jsk_robot_common ]]
+ touch arm64v8_User/src/jsk_robot/doc/CATKIN_IGNORE
+ for dir in $(find ${SOURCE_ROOT}/src/jsk_robot -maxdepth 1 -mindepth 1 -type d)
+ [[ ! arm64v8_User/src/jsk_robot/jsk_naoqi_robot =~ jsk_unitree_robot|jsk_robot_common ]]
+ touch arm64v8_User/src/jsk_robot/jsk_naoqi_robot/CATKIN_IGNORE
+ for dir in $(find ${SOURCE_ROOT}/src/jsk_robot -maxdepth 1 -mindepth 1 -type d)
+ [[ ! arm64v8_User/src/jsk_robot/jsk_magni_robot =~ jsk_unitree_robot|jsk_robot_common ]]
+ touch arm64v8_User/src/jsk_robot/jsk_magni_robot/CATKIN_IGNORE
+ for dir in $(find ${SOURCE_ROOT}/src/jsk_robot -maxdepth 1 -mindepth 1 -type d)
+ [[ ! arm64v8_User/src/jsk_robot/jsk_fetch_robot =~ jsk_unitree_robot|jsk_robot_common ]]
+ touch arm64v8_User/src/jsk_robot/jsk_fetch_robot/CATKIN_IGNORE
+ for dir in $(find ${SOURCE_ROOT}/src/jsk_robot -maxdepth 1 -mindepth 1 -type d)
+ [[ ! arm64v8_User/src/jsk_robot/jsk_hand =~ jsk_unitree_robot|jsk_robot_common ]]
+ touch arm64v8_User/src/jsk_robot/jsk_hand/CATKIN_IGNORE
+ for dir in $(find ${SOURCE_ROOT}/src/jsk_robot -maxdepth 1 -mindepth 1 -type d)
+ [[ ! arm64v8_User/src/jsk_robot/.travis =~ jsk_unitree_robot|jsk_robot_common ]]
+ touch arm64v8_User/src/jsk_robot/.travis/CATKIN_IGNORE
+ for dir in $(find ${SOURCE_ROOT}/src/jsk_robot -maxdepth 1 -mindepth 1 -type d)
+ [[ ! arm64v8_User/src/jsk_robot/jsk_denso_robot =~ jsk_unitree_robot|jsk_robot_common ]]
+ touch arm64v8_User/src/jsk_robot/jsk_denso_robot/CATKIN_IGNORE
+ for dir in $(find ${SOURCE_ROOT}/src/jsk_robot -maxdepth 1 -mindepth 1 -type d)
+ [[ ! arm64v8_User/src/jsk_robot/jsk_robot_common =~ jsk_unitree_robot|jsk_robot_common ]]
+ for dir in $(find ${SOURCE_ROOT}/src/jsk_robot -maxdepth 1 -mindepth 1 -type d)
+ [[ ! arm64v8_User/src/jsk_robot/.github =~ jsk_unitree_robot|jsk_robot_common ]]
+ touch arm64v8_User/src/jsk_robot/.github/CATKIN_IGNORE
+ for dir in $(find ${SOURCE_ROOT}/src/jsk_robot -maxdepth 1 -mindepth 1 -type d)
+ [[ ! arm64v8_User/src/jsk_robot/jsk_baxter_robot =~ jsk_unitree_robot|jsk_robot_common ]]
+ touch arm64v8_User/src/jsk_robot/jsk_baxter_robot/CATKIN_IGNORE
+ for dir in $(find ${SOURCE_ROOT}/src/jsk_robot -maxdepth 1 -mindepth 1 -type d)
+ [[ ! arm64v8_User/src/jsk_robot/.git =~ jsk_unitree_robot|jsk_robot_common ]]
+ touch arm64v8_User/src/jsk_robot/.git/CATKIN_IGNORE
+ for dir in $(find ${SOURCE_ROOT}/src/jsk_robot -maxdepth 1 -mindepth 1 -type d)
+ [[ ! arm64v8_User/src/jsk_robot/jsk_robot =~ jsk_unitree_robot|jsk_robot_common ]]
+ touch arm64v8_User/src/jsk_robot/jsk_robot/CATKIN_IGNORE
+ for dir in $(find ${SOURCE_ROOT}/src/jsk_robot -maxdepth 1 -mindepth 1 -type d)
+ [[ ! arm64v8_User/src/jsk_robot/jsk_panda_robot =~ jsk_unitree_robot|jsk_robot_common ]]
+ touch arm64v8_User/src/jsk_robot/jsk_panda_robot/CATKIN_IGNORE
+ for dir in $(find ${SOURCE_ROOT}/src/jsk_robot -maxdepth 1 -mindepth 1 -type d)
+ [[ ! arm64v8_User/src/jsk_robot/jsk_aero_robot =~ jsk_unitree_robot|jsk_robot_common ]]
+ touch arm64v8_User/src/jsk_robot/jsk_aero_robot/CATKIN_IGNORE
+ for dir in $(find ${SOURCE_ROOT}/src/jsk_robot -maxdepth 1 -mindepth 1 -type d)
+ [[ ! arm64v8_User/src/jsk_robot/jsk_unitree_robot =~ jsk_unitree_robot|jsk_robot_common ]]
+ for dir in $(find ${SOURCE_ROOT}/src/jsk_robot -maxdepth 1 -mindepth 1 -type d)
+ [[ ! arm64v8_User/src/jsk_robot/jsk_pr2_robot =~ jsk_unitree_robot|jsk_robot_common ]]
+ touch arm64v8_User/src/jsk_robot/jsk_pr2_robot/CATKIN_IGNORE
+ for dir in $(find ${SOURCE_ROOT}/src/jsk_robot -maxdepth 1 -mindepth 1 -type d)
+ [[ ! arm64v8_User/src/jsk_robot/jsk_kinova_robot =~ jsk_unitree_robot|jsk_robot_common ]]
+ touch arm64v8_User/src/jsk_robot/jsk_kinova_robot/CATKIN_IGNORE
+ '[' 1 -eq 0 ']'
+ vcs import arm64v8_User/src
....
=== arm64v8_User/src/app_manager (git) ===
Cloning into '.'...
=== arm64v8_User/src/unitree_legged_sdk (git) ===
Cloning into '.'...
=== arm64v8_User/src/unitree_ros (git) ===
Cloning into '.'...
=== arm64v8_User/src/unitree_ros_to_real (git) ===
Cloning into '.'...
+ docker run -it --rm ros1-unitree:arm64v8 bash -c exit
WARNING: The requested image's platform (linux/arm64/v8) does not match the detected host platform (linux/amd64/v3) and no specific platform was requested
+ tee arm64v8_build_user.log
++ id -u sktometometo
+ docker run -it --rm -u 1000 -e INSTALL_ROOT=System -v /home/sktometometo/ros/ws_jsk_unitree/src/jsk-ros-pkg/jsk_robot/jsk_unitree_robot/cross/arm64v8_System/ros1_dependencies:/opt/jsk/System/ros1_dependencies:ro -v /home/sktometometo/ros/ws_jsk_unitree/src/jsk-ros-pkg/jsk_robot/jsk_unitree_robot/cross/arm64v8_System/Python:/opt/jsk/System/Python:ro -v /home/sktometometo/ros/ws_jsk_unitree/src/jsk-ros-pkg/jsk_robot/jsk_unitree_robot/cross/arm64v8_System/ros1_inst:/opt/jsk/System/ros1_inst:ro -v /home/sktometometo/ros/ws_jsk_unitree/src/jsk-ros-pkg/jsk_robot/jsk_unitree_robot/cross/arm64v8_System/ros1_dependencies_setup.bash:/opt/jsk/System/ros1_dependencies_setup.bash:ro -v /home/sktometometo/ros/ws_jsk_unitree/src/jsk-ros-pkg/jsk_robot/jsk_unitree_robot/cross/arm64v8_System/system_setup.bash:/opt/jsk/System/system_setup.bash:ro -v /home/sktometometo/ros/ws_jsk_unitree/src/jsk-ros-pkg/jsk_robot/jsk_unitree_robot/cross/arm64v8_User:/opt/jsk/User:rw -v /home/sktometometo/ros/ws_jsk_unitree/src/jsk-ros-pkg/jsk_robot/jsk_unitree_robot/cross/rosinstall_generator_unreleased.py:/home/user/rosinstall_generator_unreleased.py:ro ros1-unitree:arm64v8 bash -c '    source /opt/jsk/System/system_setup.bash &&     env &&     set -xeuf -o pipefail &&     cd /opt/jsk/User &&     [ 1 -eq 0 ] || ROS_PACKAGE_PATH=src:${ROS_PACKAGE_PATH} /home/user/rosinstall_generator_unreleased.py jsk_unitree_startup unitreeeus --rosdistro melodic --exclude RPP --exclude mongodb_store | tee user.repos &&     [ 1 -eq 0 -o -z "$(cat user.repos)" ] || PYTHONPATH= vcs import src < user.repos &&     catkin build jsk_unitree_startup unitreeeus -s -vi         --cmake-args -DCATKIN_ENABLE_TESTING=FALSE     '
WARNING: The requested image's platform (linux/arm64/v8) does not match the detected host platform (linux/amd64/v3) and no specific platform was requested
/bin/bash: line 0: source: /opt/jsk/System/system_setup.bash: is a directory
# 'jsk_unitree_startup'
- Check if 'jsk_unitree_startup' exists
- Check if 'jsk_unitree_startup' released
'jsk_unitree_startup' is not releasead, you need add them within your workspace
# 'unitreeeus'
- Check if 'unitreeeus' exists
- Check if 'unitreeeus' released
'unitreeeus' is not releasead, you need add them within your workspace
rosinstall_generator  --rosdistro melodic --deps --exclude ['RPP', 'RPP', 'mongodb_store']
rosinstall said : No packages/stacks left after ignoring unreleased
                  The packages may have been already installed
/bin/bash: catkin: command not found
make: *** [Makefile:18: user] エラー 127

```

sktometometo, 2023-10-06:
My PC is

- Thinkpad P15s
- Ubuntu 20.04
- ROS Noetic

And workspace configuration is

```bash
~/ros/ws_jsk_unitree/src $ wstool info
workspace: /home/sktometometo/ros/ws_jsk_unitree/src

 Localname             S SCM Version (Spec)                                         UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------             - --- --------------                                         -----------  ---------------------------
 unitree_ros_to_real     git develop                                                36a877d87818 github.com/k-okada/unitree_ros_to_real.git
 unitree_ros             git <detached>  (5dee20c5ba1c86dfeb906e24c7870f19cc93d605) 5dee20c5ba1c github.com/unitreerobotics/unitree_ros.git
 unitree_legged_sdk      git develop                                                b4a01bf3068b github.com/k-okada/unitree_legged_sdk.git
 jsk-ros-pkg/jsk_robot   git master  (-)                                            9a1da3dccba7 github.com/jsk-ros-pkg/jsk_robot.git

```

tkmtnt7000, 2023-10-06:
Follow this instruction.
https://github.com/jsk-ros-pkg/jsk_robot/blob/9a1da3dccba70a646dc42c81f911ef2dd2a5475c/jsk_unitree_robot/unitree.rosinstall#L9

Download the newest system build tar files from google drive and unpack them under cross directory.

sktometometo, 2023-10-06:
> download the arm64v8_System.tgz archive file from [here](https://drive.google.com/drive/u/2/folders/1SBA9oAwjfD84yRFEB-jsCH1m5Q8eEGSK) and extract under jsk_unitree_robot/cross/ directory before run make command.

Is the aim of this step to only speed up building process? or is it necessary for some reason?

sktometometo, 2023-10-06:
If it is intended to run `make system` before doing `make user`, instruction should include that.

sktometometo, 2023-10-06:
I have created a pull request about it https://github.com/jsk-ros-pkg/jsk_robot/pull/1861