jsk-ros-pkg/jsk_robot #1863: [Unitree] question about development on Go1

sktometometo, 2023-10-06:
I would like to use extra packages ( e.g. https://github.com/sktometometo/esp_now_ros ) in addition to jsk_robot on Go1.
Is it necessary to modify `build_user.sh` for this?

And also I would like to use another rosserial device with lead teleop. So I would like to modify udev to fix device name. What should I do for this?

sktometometo, 2023-10-06:
@k-okada

sktometometo, 2023-10-06:
I have tried to build `esp_now_ros` by modifying unitree.repos like below.
But I have got errors.

```bash
[esp_now_ros:make] /usr/bin/make -f CMakeFiles/std_msgs_generate_messages_py.dir/build.make CMakeFiles/std_msgs_generate_messages_py.dir/depend
[unitree_legged_real:make] [  0%] Built target sensor_msgs_generate_messages_nodejs
[unitree_legged_real:make] /usr/bin/make -f CMakeFiles/walk_lcm.dir/build.make CMakeFiles/walk_lcm.dir/depend
[esp_now_ros:make] make[2]: Entering directory '/opt/jsk/User/build/esp_now_ros'
[esp_now_ros:make] cd /opt/jsk/User/build/esp_now_ros && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /opt/jsk/User/src/esp_now_ros /opt/jsk/User/src/esp_now_ros /opt/jsk/User/build/esp_now_ros /opt/jsk/User/build/esp_now_ros /opt/jsk/User/build/esp_now_ros/CMakeFiles/std_msgs_generate_messages_nodejs.dir/DependInfo.cmake --color=
[esp_now_ros:make] [  6%] Built target std_msgs_generate_messages_eus
[esp_now_ros:make] /usr/bin/make -f CMakeFiles/std_msgs_generate_messages_cpp.dir/build.make CMakeFiles/std_msgs_generate_messages_cpp.dir/depend
[esp_now_ros:make] make[2]: Entering directory '/opt/jsk/User/build/esp_now_ros'
[esp_now_ros:make] make[2]: Nothing to be done for 'CMakeFiles/std_msgs_generate_messages_lisp.dir/build'.
[esp_now_ros:make] make[2]: Leaving directory '/opt/jsk/User/build/esp_now_ros'
[esp_now_ros:make] make[2]: Entering directory '/opt/jsk/User/build/esp_now_ros'
[esp_now_ros:make] cd /opt/jsk/User/build/esp_now_ros && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /opt/jsk/User/src/esp_now_ros /opt/jsk/User/src/esp_now_ros /opt/jsk/User/build/esp_now_ros /opt/jsk/User/build/esp_now_ros /opt/jsk/User/build/esp_now_ros/CMakeFiles/std_msgs_generate_messages_py.dir/DependInfo.cmake --color=
[esp_now_ros:make] [  6%] Built target _esp_now_ros_generate_messages_check_deps_Packet
[unitree_legged_real:make] make[2]: Entering directory '/opt/jsk/User/build/unitree_legged_real'_now_ros:make (6%) - ...
[unitree_legged_real:make] cd /opt/jsk/User/build/unitree_legged_real && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /opt/jsk/User/src/unitree_ros_to_real/unitree_legged_real /opt/jsk/User/src/unitree_ros_to_real/unitree_legged_real /opt/jsk/User/build/unitree_legged_real /opt/jsk/User/build/unitree_legged_real /opt/jsk/User/build/unitree_legged_real/CMakeFiles/walk_lcm.dir/DependInfo.cmake --color=
[unitree_legged_real:make] /usr/bin/make -f CMakeFiles/torque_lcm.dir/build.make CMakeFiles/torque_lcm.dir/depend
[esp_now_ros:make] make[2]: Entering directory '/opt/jsk/User/build/esp_now_ros'
[esp_now_ros:make] cd /opt/jsk/User/build/esp_now_ros && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /opt/jsk/User/src/esp_now_ros /opt/jsk/User/src/esp_now_ros /opt/jsk/User/build/esp_now_ros /opt/jsk/User/build/esp_now_ros /opt/jsk/User/build/esp_now_ros/CMakeFiles/std_msgs_generate_messages_cpp.dir/DependInfo.cmake --color=
[unitree_legged_real:make] make[2]: Entering directory '/opt/jsk/User/build/unitree_legged_real'_now_ros:make (6%) - ...
[unitree_legged_real:make] cd /opt/jsk/User/build/unitree_legged_real && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /opt/jsk/User/src/unitree_ros_to_real/unitree_legged_real /opt/jsk/User/src/unitree_ros_to_real/unitree_legged_real /opt/jsk/User/build/unitree_legged_real /opt/jsk/User/build/unitree_legged_real /opt/jsk/User/build/unitree_legged_real/CMakeFiles/torque_lcm.dir/DependInfo.cmake --color=
[esp_now_ros:make] [  6%] Built target std_msgs_generate_messages_lispreal:make (0%) - 9.1] [esp_now_ros:make (6%) - ...
[unitree_legged_real:make] /usr/bin/make -f CMakeFiles/ros_server.dir/build.make CMakeFiles/ros_server.dir/depend
[esp_now_ros:make] make[2]: Leaving directory '/opt/jsk/User/build/esp_now_ros' (0%) - 9.2] [esp_now_ros:make (6%) - ...
[esp_now_ros:make] /usr/bin/make -f CMakeFiles/std_msgs_generate_messages_nodejs.dir/build.make CMakeFiles/std_msgs_generate_messages_nodejs.dir/build
[esp_now_ros:make] make[2]: Leaving directory '/opt/jsk/User/build/esp_now_ros'
[esp_now_ros:make] /usr/bin/make -f CMakeFiles/std_msgs_generate_messages_py.dir/build.make CMakeFiles/std_msgs_generate_messages_py.dir/build
[unitree_legged_real:make] make[2]: Entering directory '/opt/jsk/User/build/unitree_legged_real'
[unitree_legged_real:make] cd /opt/jsk/User/build/unitree_legged_real && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /opt/jsk/User/src/unitree_ros_to_real/unitree_legged_real /opt/jsk/User/src/unitree_ros_to_real/unitree_legged_real /opt/jsk/User/build/unitree_legged_real /opt/jsk/User/build/unitree_legged_real /opt/jsk/User/build/unitree_legged_real/CMakeFiles/ros_server.dir/DependInfo.cmake --color=
[esp_now_ros:make] make[2]: Entering directory '/opt/jsk/User/build/esp_now_ros'(0%) - 9.3] [esp_now_ros:make (6%) - ...
[esp_now_ros:make] make[2]: Nothing to be done for 'CMakeFiles/std_msgs_generate_messages_nodejs.dir/build'.
[esp_now_ros:make] make[2]: Leaving directory '/opt/jsk/User/build/esp_now_ros'
[esp_now_ros:make] make[2]: Entering directory '/opt/jsk/User/build/esp_now_ros'
[esp_now_ros:make] make[2]: Nothing to be done for 'CMakeFiles/std_msgs_generate_messages_py.dir/build'.
[esp_now_ros:make] make[2]: Leaving directory '/opt/jsk/User/build/esp_now_ros'
[esp_now_ros:make] make[2]: Leaving directory '/opt/jsk/User/build/esp_now_ros'
[esp_now_ros:make] /usr/bin/make -f CMakeFiles/std_msgs_generate_messages_cpp.dir/build.make CMakeFiles/std_msgs_generate_messages_cpp.dir/build
[... cut ...]
    #   k-okada/unitree_ros_to_real:develop
    version: develop
  app_manager:
    type: git
    url: https://github.com/PR2/app_manager
    version: kinetic-devel
    # enable to run simple apps parallel to other apps
    # https://github.com/PR2/app_manager/pull/59 have been merged,
    # but 1.4.0 have not released yet.
  catkin_virtualenv:
    type: git
    url: https://github.com/iory/catkin_virtualenv
    version: unitree
  esp_now_ros:
    type: git
    url: https://github.com/sktometometo/esp_now_ros.git
    version: v0.3.0
```

sktometometo, 2023-10-06:
This error seems related to https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_unitree_robot/cross/README.md#running-python3 . But there is no descriptions about this. What is `st-000-ros1.bash` and how is `python-futures` used here?

There is no references to them.

```bash
~/ros/ws_jsk_unitree/src/jsk-ros-pkg/jsk_robot/jsk_unitree_robot/cross $ ls
CATKIN_IGNORE   arm64v8_System.tgz      build_ros1_dependencies.sh  nano2_pro_install_scripts     ros1_dependencies_build_scripts
Makefile        arm64v8_User            build_user.sh               nano3_pro_install_scripts     rosinstall_generator_unreleased.py
README.md       arm64v8_build_user.log  docker                      prepare_requirements_ros1.sh  run_user.sh
arm64v8_System  build_ros1.sh           install.sh                  repos                         startup_scripts
~/ros/ws_jsk_unitree/src/jsk-ros-pkg/jsk_robot/jsk_unitree_robot/cross $ grep -ir "st-000-ros1.bash"
README.md:Since `st-000-ros1.bash` set PYTHONPATH and we installed `python-futures` via pip, It breaks python3 execution.
~/ros/ws_jsk_unitree/src/jsk-ros-pkg/jsk_robot/jsk_unitree_robot/cross $ grep -ir python-futures
README.md:Since `st-000-ros1.bash` set PYTHONPATH and we installed `python-futures` via pip, It breaks python3 execution.

```

tkmtnt7000, 2023-10-06:
I used https://github.com/tkmtnt7000/jsk_robot/tree/add-esp-now-ros-build branch (originally https://github.com/iory/jsk_robot/tree/unitree-virtualenv branch) and I managed  to build esp_now_ros successfully.

ちゃんと動くのかどうかは未知数ですが．．．

sktometometo, 2023-10-06:
ありがとう見てみます

tkmtnt7000, 2023-10-06:
> I used [tkmtnt7000/jsk_robot@`add-esp-now-ros-build`](https://github.com/tkmtnt7000/jsk_robot/tree/add-esp-now-ros-build?rgh-link-date=2023-10-06T08%3A11%3A55Z) branch (originally [iory/jsk_robot@`unitree-virtualenv`](https://github.com/iory/jsk_robot/tree/unitree-virtualenv?rgh-link-date=2023-10-06T08%3A11%3A55Z) branch) and I managed to build esp_now_ros successfully.
> 
> ちゃんと動くのかどうかは未知数ですが．．．

生成したものは，https://drive.google.com/drive/folders/17cmkylJD7dE_gWIY9NB1WJOtoPxeJ50x?usp=drive_linkにあげました．systemビルドの方はこちらを展開していただけると良いかもしれません．

k-okada, 2023-10-11:
@sktometometo さっき途中になっていた `./run_user.sh` して `rosrun esp_now_ros python3` してから，これが動かないという `import xxx` を教えてください．

sktometometo, 2023-10-11:
とりあえずpyyamlがないとでていて動きませんでした。Requirementsにpyyaml追加して試している最中です

sktometometo, 2023-10-11:
複数ブランチが分散していてbugがトラックできないので以後masterベースで進めます

sktometometo, 2023-10-11:
> I used https://github.com/tkmtnt7000/jsk_robot/tree/add-esp-now-ros-build branch (originally https://github.com/iory/jsk_robot/tree/unitree-virtualenv branch) and I managed to build esp_now_ros successfully.

In this branch, app_manager.launch cannot run and I cannot make lead_teleop demo (which is required for controlling the robot during demo)