jsk-ros-pkg/jsk_robot #1876: [Fetch] TF Duplicated stamp

sktometometo, 2023-10-15:
Noetic のクライアントでつなぐと TF Duplicated stamp が出てくる件について

`/odom` -> `/base_link` が問題になっているらしく、たどっていくと https://github.com/cra-ros-pkg/robot_localization でタイムスタンプが重複してpublishされている
( `/odom_corrected` では stamp は被りがなく、 `/odom_robot_localization` と `/tf_odom_robot_localization` では被りがある )

mqcmd196, 2023-10-15:
ナイスです！これってなんでmelodicのクライアントだと出ないんでしょうか

sktometometo, 2023-10-15:
https://github.com/ros/geometry2/issues/467
まさにこれってのあったわ

sktometometo, 2023-10-15:
@mqcmd196 
シンプルに melodic-devel には https://github.com/ros/geometry2/commit/04625380bdff3f3e9e860fc0e85f71674ddd1587#diff-c25683741e1f7b4e50eb6c5cdcad9661R275 の変更が入っていないんだと思う

melodic-devel と noetic-devel のdiff
https://github.com/ros/geometry2/compare/melodic-devel...noetic-devel

sktometometo, 2023-10-15:
robot_localization では https://github.com/cra-ros-pkg/robot_localization/issues/574 で議論していて、 https://github.com/cra-ros-pkg/robot_localization/pull/595 でオプションを追加して対応している。

sktometometo, 2023-10-15:
https://github.com/cra-ros-pkg/robot_localization/compare/noetic-devel...melodic-devel 見てみると melodic-devel にもこの変更入っていそうだから、rosparam設定するだけでなくなるんじゃないかな

sktometometo, 2023-10-15:
noetic: https://github.com/cra-ros-pkg/robot_localization/blob/040dd17034d2fae55f99cc6d93598aeeb8b97b47/src/ros_filter.cpp#L870-L871

melodic: https://github.com/cra-ros-pkg/robot_localization/blob/fd961e28b40b31264d2d19bd244cf68943f74bb9/src/ros_filter.cpp#L824-L829

なので noetic-devel 使う必要がありそう