jsk-ros-pkg/jsk_robot #1901: [Spot][BelKa] Spot Battery Issue

sktometometo, 2023-12-07:
There are charging issues for spot batteries.

1. SN: AVD11810011

TODO

2. SN: TODO

TODO

tkmtnt7000, 2023-12-14:
I have checked these batteries more by attaching the batteries to Spot.

1. SN: AVD11810011
It does not seem that the robot recognize the battery.
[image]
[image]

2. SN: AVD11810028
The battery has something damaged in capacity.
[image]
[image]

sktometometo, 2023-12-26:
Support からの返信

Thank you for contacting us. Sorry to hear about the issues with the two batteries AVD11810011 and AVD11810028. I took a look at the zip file you supplied, thank you very much for supplying these details. I'd like to recommend some steps below.

 

AVD11810011

It seems that this battery could be in sleep mode. It may be possible to recover this battery with the following procedure:

Take the battery out of the robot.
Connect the robot to shore power using the charger's Shore Power cable.
Put in the battery.
Take out the battery (very important step).
Put it back in to see if it comes out of Sleep Mode.
 

Let me know if you are able to recover SN AVD11810011 using this procedure.

 

AVD11810028

Checking your screenshots I would say this battery is not recoverable based on the very high Cell Voltage Differential.

 

 

Please try to recover SN AVD11810011 and let me know if it works. Additionally please provide the following information so I can connect you with my colleagues in Services regarding the next steps to either repair or replace the batteries.

 

Battery serial numbers: AVD11810028,
Shipping address
 
Point-of-contact
Name:
Phone number:
Email:

sktometometo, 2023-12-28:
No change for the suggested steps for AVD11810011

mqcmd196, 2024-04-08:
> Battery with AVD20900019 also has problem that it does not blink when
> pressing the button and when it is connectd charger nor applied unsleep
> procedure.
>
> IMG_20240407_143800_1.jpg (view on web)
>
>
> https://github.com/jsk-ros-pkg/jsk_robot/assets/9410362/7f42e6be-1af6-45e7-9fd8-b29755b42b95
>
>
> https://github.com/jsk-ros-pkg/jsk_robot/assets/9410362/12ec7d6c-7257-49da-91ac-edcc8ec1e385

Hmm, the battery breaks per year...

sktometometo, 2024-04-08:
I have posted this to spot sales contacts.

sktometometo, 2024-04-08:
Anyway, I feel like the cause of this problem is that if battery is attached to robot and left uncharged, battery becomes overdischarge even the robot is turned off.
I have found that this problem happens when this battery is inserted to BelKa and it is left uncharged in 81C1.

sktometometo, 2024-04-15:
According to https://support.bostondynamics.com/s/article/Introduction-to-the-Spot-battery-and-charger , 

> Always remove the battery when Spot is not in use unless the robot is connected to shore power or sitting on a powered Spot Dock (even if the robot is powered off). Batteries left in the robot while not in use will continue to discharge, even when the robot is powered off. Batteries left in a robot for more than 24 hours may be damaged beyond repair.

So we have to remove the battery if the robot is not connected to dock or shore charger.

sktometometo, 2024-04-15:
2 of 3 batteries seem to suffer this problem. So we have to take some approach not to cause this again.

1. Add CAUTION to robot so that battery is removed when it is not in use.
2. Prepare battery safety section about battery to introduction document.
3. Prepare automatic checking if the robot is charged when battery is low.

sktometometo, 2024-04-15:
And 1 battery has balancing issue.
https://support.bostondynamics.com/s/article/Introduction-to-the-Spot-battery-and-charger has procedure about battery balancing. So I gonna try this again later.