jsk-ros-pkg/jsk_robot #1905: [jsk_kinova_robot] noeticでKinovaを使いたい

Kanazawanaoaki, 2024-02-04:
### https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_kinova_robot をnoeticで使いたい

[READMEの指示](https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_kinova_robot#how-to-setup-development-environment )にしたがって環境構築を進めていくと，以下の部分で引っかかりポイントがあった．

#### Conanのバージョン問題
Conan Setup で以下のエラーが発生した．
```bash
$ python3 -m pip install --user conan
$ conan config set general.revisions_enabled=1
usage: conan config [-h] [-v [V]] {home,install,list,show} ...
conan config: error: argument subcommand: invalid choice: 'set' (choose from 'home', 'install', 'list', 'show')
ERROR: Exiting with code: 2
```
これはconan 2 にはsetという引数が無いことが原因なので，
https://github.com/conan-io/conan-center-index/issues/18806

以下のようにconan 1をインストールするようにしてエラーを回避した．conan 2を使うようにするべきかは今後要検討か．
```bash
python3 -m pip install --user 'conan<2'
```

#### noeticの環境構築での問題
ROS Environment Setup を進めるなかで，
```
source /opt/ros/melodic/setup.bash
```
はnoeticに読み替える必要あり，もしくは
```bash
source /opt/ros/$ROS_DISTRO/setup.bash
```
に変更するべきか．

また`catkin build jsk_kinova_startup kinovaeus`実行時に
```bash
CMake Error at /opt/ros/noetic/share/collada_urdf/cmake/collada_urdfConfig.cmake:113 (message):
  Project 'collada_urdf' specifies '/usr/../include/include' as an include
  dir, which is not found.  It does neither exist as an absolute directory
  nor in '${{prefix}}//usr/../include/include'.  Check the issue tracker
  'https://github.com/ros/collada_urdf/issues' and consider creating a ticket
  if the problem has not been reported yet.
```
のcollada_urdf 問題が発生したので，
https://github.com/jsk-ros-pkg/jsk_robot/issues/1892 の２）とりあえずの解決策 で対処した．

以上によりbuildを行うことができた．

Kanazawanaoaki, 2024-02-04:
報告しているうちに，現状原因が分かっている問題については，
- noeticの環境構築における問題
- ドキュメントがメインで対象としているGen3 liteと自分が今回使っているGen3の違いにより実行中に起きた問題
の2つに分けられることが分かってきたのでコメントを編集しました．
また，前者については環境構築は完了しているので，https://github.com/jsk-ros-pkg/jsk_robot/pull/1906 にPRを出しました．

tkmtnt7000, 2024-02-05:
>Kinova Gen3の使用における問題(Gen3 liteとGen3の違いにより実行中に起きた問題)

おっしゃる通りで、jsk_kinova_robotのReadmeはGen3 Liteを使いながら書いたものなので、若干の違いから実行できないものがあると思います。きちんと切り分けて書かなかったのが悪いですが、当時B4の僕はそこまであたまが回っていなかったと思います。

>kinova-rest-poseが未定義

これはkinova gen3 liteをspotに載せるために作ったposeだったと思います。用途やアームのハードウェア特性によっていい姿勢は変わると思うので、よさそうな姿勢をみつけたらPRを送ってくくださると助かります。

===
以下は後で対応するPRなどを確認します。

>Moveitを使っているためか，腕が動き始めるまでにラグが結構あるし，動いてもゆっくり([こちら](http://www.nihonbinary.co.jp/Products/Robot/GEN3.html)によると最大動作速度 40 cm/秒 (並進移動)らしいのでそもそもKinovaはそこまで速くない可能性あり)．

この問題はGen3 Liteでも直面していて、一時しのぎとしてangle-vector-rawを一部使っていたと思います。2年前の記憶なのでちゃんとした解決方法は忘れました。後で確認します。

>十分な時間で計画しないと動作が実行されないのか，時間を長く設定しないとsend *ri* :angle-vectorをしても動かないことがある．動作が実行できたかどうかを確認する方法は？

これはjsk_pr2eusに以前プルリクを投げて、follow-joint-trajectory actionの結果(preemptとかrejectとか)がeusのインタプリタ上に流れるようにした記憶がありますが...忘れたのでこれも後で確認します。

tkmtnt7000, 2024-02-05:
>Moveitを使っているためか，腕が動き始めるまでにラグが結構あるし，動いてもゆっくり([こちら](http://www.nihonbinary.co.jp/Products/Robot/GEN3.html)によると最大動作速度 40 cm/秒 (並進移動)らしいのでそもそもKinovaはそこまで速くない可能性あり)

https://github.com/jsk-ros-pkg/jsk_robot/pull/1313#issuecomment-871945537
腕が動き始めるまでのラグの理由はこちらで山口さんが説明されていました
https://github.com/jsk-ros-pkg/jsk_robot/pull/1373
でこちらの問題に関してはいい感じに間引いて解決を図っていそう、というところです。

Kanazawanaoaki, 2024-02-05:
@tkmtnt7000  ありがとう．マージされているし，自分が"結構ラグ"があると感じていたのは，実はすでに修正されてだいぶ短くなっているラグみたいですね．
もっと早くWeb Appから動かすようにするためには補完の部分などを早くする必要があるわけですね．

Kanazawanaoaki, 2024-02-05:
> Gen3のカメラがDepthは使えるがRGBは使えない．
については，Web AppからはRGBのCamra画像が見えているが，launchからは見えていない．  
ログ：https://gist.github.com/Kanazawanaoaki/3fa885db42c0a252fe729f247c1da9e0