jsk-ros-pkg/jsk_robot #1907: [jsk_panda_robot]Touch HIDを使いたい

YUKINA-3252, 2024-04-10:
Touch HIDのドライバのinstallに関して、前モデルのTouch USBとは若干異なる手順、引っかかりポイントがあったのでこのissueにてまとめます。

### Haptic device driver、OpenHapticsのinstallerのバージョンについて
本isuueでの実験環境はHaptic device driverはTouchDriver_2023_11_15、OpenHapticsはopenhaptics_3.4-0-developer-edition-amd64。
以下のドライブに各installerをアーカイブしている。
https://drive.google.com/drive/folders/1SiLa0ZBKllt6mm_c99FPo-aZoehjO0P2
公式HPにはTouchDriver_2023_11_15のプラットフォームはUbuntu22.04と記載があるが、20.04での使用も可能であることが確認済みである。

### コマンド入力時にQtのエラーが発生する
`Touch_Diagnostic`コマンドでデバイスのキャリプレーションを行おうとすると、以下のエラーが出る。
```
undefined_symbol: _ZdlPvm, version Qt_5
```
2024年4月9日現在、`apt-get install qt5-default`でinstallされるQtのバージョンはUbuntu20.04では5.12.8であるが、以下のコマンドで5.12.0をinstallし、`LD_LIBRARY_PATH`でQt5.12.0のlibを参照するようにするとエラーが解消する。
```
$ wget https://download.qt.io/archive/qt/5.12/5.12.0/qt-opensource-linux-x64-5.12.0.run
$ chmod +x qt-opensource-linux-x64-5.12.0.run
$ ./qt-opensource-linux-x64-5.12.0.run
```
このあと、`./install_haptic_driver`で再度ドライバをinstallする必要あり？（再現実験未実施）

公式HPにも記載があるが、一度デバイスをPCに接続すると`~/.3dsystems/config`にconfigファイルが作られてしまうため、Qtのバージョンを入れ直したりした際には予めconfigファイルを消してからデバイスを接続する。

### ttyACMとしてTouchデバイスが認識されない
前モデルのTouch USBではデバイスがttyACM* のかたちで認識されると公式ドキュメントに記載がある。
Touch HIDに関しても `/dev/ttyACM*`の名前ファイルを確認するコマンド(`ListCOMPortHapticDevices`)があるが、ttyACM* のかたちではデバイスは認識されない(`Touch_Diagnostic`コマンドによるキャリプレーション、エンコーダーのリアルタイム表示など使用上は問題がない）。

k-okada, 2024-04-10:
> ttyACMとしてTouchデバイスが認識されない

これって、USB接続？だとして、USB接続したものが /dev/ 以下にどう表示されるかって、多分 USB-Serial のチップだったりドライバ依存の難しい問題じゃないか、と思っています。（で会っている？ @sktometometo @makabe0510 )
なので、とりあえずttyACMが無かったらttyS とか ttyUSBとか疑ってみると良いので、
デバイスをつなげた5秒後ぐらいに、以下のコマンドを打った時の結果が知りたいです。
```
lsusb  ;; 。TouchHID が USB接続の場合
ls -al /dev/tty*
dmesg
```

sktometometo, 2024-04-10:
岡田先生のおっしゃっている通りで、ttyACM*として認識されるのは USB CDC で通信しているデバイス類で、USB CDC デバイスとして認識されている場合は /dev/ttyACM* とでるはずです。

FTDIのUSB-UARTチップのように独自プロトコルでUARTを実現している場合は /dev/ttyUSB* とでるはずで、こちらの場合はLinuxカーネルにドライバが同梱されていなければ自分でインストールする必要があります。

sktometometo, 2024-04-11:
単純にACM（モデム）として見えてるんじゃなくて、HID（キーボードやマウスと同じくHID）として見えてるならttyACMとしては出てこないような

sktometometo, 2024-04-11:
@YUKINA-3252 これって特にttyACMが見えないことで何かが立ち上げられないわけではない？

mqcmd196, 2024-04-11:
横から失礼します．

> これって特にttyACMが見えないことで何かが立ち上げられないわけではない？

これは公式のマニュアル，付属スクリプトが間違っていて，これが見えないとダメだ，とかいてあるのですが，そうでなくてもちゃんと動きます

sktometometo, 2024-04-11:
Touch USB と Touch HID の仕様が違うだけのような気がする。HIDとして見えてるなら /dev/hidraw* あたりになにか出るはず

YUKINA-3252, 2024-04-15:
申し訳ないです、対応遅れました。

`lsusb`で現れるTouchに該当するのは以下のラインでした。
```
Bus 001 Device 008: ID 2988:0304 3D Systems 3D Systems Haptic Device
```
また、`dmesg -T`したときの該当する部分は以下でした。
```
[Mon Apr 15 16:16:48 2024] usb 1-4: new full-speed USB device number 9 using xhci_hcd
[Mon Apr 15 16:16:48 2024] usb 1-4: New USB device found, idVendor=2988, idProduct=0304, bcdDevice= 2.00
[Mon Apr 15 16:16:48 2024] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Mon Apr 15 16:16:48 2024] usb 1-4: Product: 3D Systems Haptic Device
[Mon Apr 15 16:16:48 2024] usb 1-4: Manufacturer: 3D Systems
[Mon Apr 15 16:16:48 2024] usb 1-4: SerialNumber: 4196XXXXX
[Mon Apr 15 16:16:48 2024] hid-generic 0003:2988:0304.0007: hiddev0,hidraw3: USB HID v1.11 Device [3D Systems 3D Systems Haptic Device] on usb-0000:00:14.0-4/input0
```
新城さんのおっしゃる通りで、HIDは`/dev/hidraw3`と`/dev/usb/hiddev0`が該当するかなと思います。
これに関しては私自身もデバッグしている際にttyACM*の形ではなくこの形で出ていることに気づいてはいたのですが、HIDの仕様を理解しておらず、Touch USBの仕様から早とちりしてしまいました。

デバイス自体は動き、キャリブレーションやROSでlaunchして位置姿勢や力の情報をtopicとして見ることも問題なくできたので、引っかかりポイントとしては結局の所Qtのバージョンのみでした。

mqcmd196, 2024-05-23:
closed as solved

closed 2024-05-23 by mqcmd196