jsk-ros-pkg/jsk_robot #1908: [Fetch1075] サーボがONにならない

mqcmd196, 2024-04-10:
Fetch1075を @yoshiki-447 と使おうとしたところ，サーボが入らなかった．

緊急停止をON -> OFF -> ON としてもサーボが入らず，再起動してもサーボが入らなかった．

L515のUSBケーブルを抜いて緊急停止をON -> OFF -> ONとしてもサーボが入らなかった

```shell
cat /var/log/ros/robot.log | tail -n +252643 > /tmp/robot_servo_error.log
```
の実行結果
[robot_servo_error.log](https://github.com/jsk-ros-pkg/jsk_robot/files/14928653/robot_servo_error.log)

mqcmd196, 2024-04-10:
rqt_runtimer_monitor で見ると`head_tilt_mcb` `Joint Encoder Error` とでている

[image]

mqcmd196, 2024-04-10:
Fetch1075のJoy Stickでtorsoを上げ下げしていると，急に使えるようになる．しかししばらくして上のrqt_runtimer_monitorに，`head_tilt_mcb: Joint Encoder Error` が出る

mqcmd196, 2024-04-10:
rostopic echo /robot_state/boards の結果
```
- 
  name: "Mainboard"
  board_type: 15011101
  board_revision: 15
  software_revision: 103
  boot_status: 0
  board_flags: 0
  system_time: 2864396
  voltage: 28.140625
  temperature: 0.0
  packets_sent: 631211
  packets_recv: 654866
  packets_bad: 0
  packets_back: 527846
- 
  name: "Charger"
  board_type: 15011101
  board_revision: 15
  software_revision: 106
  boot_status: 0
  board_flags: 0
  system_time: 2864407
  voltage: 0.0
  temperature: 0.0
  packets_sent: 1055722
  packets_recv: 1085123
  packets_bad: 0
  packets_back: 1055658
- 
  name: "l_wheel_mcb"
  board_type: 15011111
  board_revision: 16
  software_revision: 101
  boot_status: 0
  board_flags: 0
  system_time: 222693
  voltage: 28.390625
  temperature: 32.0
  packets_sent: 494832
  packets_recv: 45234
  packets_bad: 0
  packets_back: 487756
- 
  name: "r_wheel_mcb"
  board_type: 15011112
  board_revision: 16
  software_revision: 101
  boot_status: 0
  board_flags: 0
  system_time: 222718
  voltage: 28.515625
  temperature: 33.0
  packets_sent: 494832
  packets_recv: 45231
  packets_bad: 0
  packets_back: 487755
- 
  name: "torso_lift_mcb"
  board_type: 15011113
  board_revision: 16
  software_revision: 101
  boot_status: 0
  board_flags: 0
  system_time: 222718
  voltage: 28.40625
  temperature: 29.0
  packets_sent: 494832
  packets_recv: 45231
  packets_bad: 0
  packets_back: 487752
- 
  name: "head_pan_mcb"
  board_type: 15011131
  board_revision: 4
  software_revision: 101
  boot_status: 0
  board_flags: 0
  system_time: 222709
  voltage: 28.328125
  temperature: 30.0
  packets_sent: 494832
  packets_recv: 45229
  packets_bad: 0
  packets_back: 487751
- 
  name: "head_tilt_mcb"
  board_type: 15011132
  board_revision: 4
  software_revision: 101
  boot_status: 0
  board_flags: 4
  system_time: 222719
  voltage: 28.46875
  temperature: 33.0
  packets_sent: 494832
  packets_recv: 45932
  packets_bad: 0
  packets_back: 487751
- 
  name: "shoulder_pan_mcb"
  board_type: 15011121
  board_revision: 5
  software_revision: 101
  boot_status: 0
  board_flags: 0
  system_time: 222720
  voltage: 28.46875
  temperature: 28.0
  packets_sent: 494831
  packets_recv: 45234
  packets_bad: 0
  packets_back: 494819
- 
  name: "shoulder_lift_mcb"
  board_type: 15011122
  board_revision: 5
  software_revision: 101
  boot_status: 0
  board_flags: 0
  system_time: 222695
  voltage: 28.53125
  temperature: 27.0
  packets_sent: 494831
  packets_recv: 45237
  packets_bad: 0
  packets_back: 494819
- 
  name: "upperarm_roll_mcb"
  board_type: 15011123
  board_revision: 5
  software_revision: 101
  boot_status: 0
  board_flags: 0
  system_time: 222730
  voltage: 28.171875
  temperature: 27.0
  packets_sent: 494831
  packets_recv: 45236
  packets_bad: 0
  packets_back: 494819
- 
  name: "elbow_flex_mcb"
  board_type: 15011124
  board_revision: 5
  software_revision: 101
  boot_status: 0
  board_flags: 0
  system_time: 222700
  voltage: 28.421875
  temperature: 27.0
  packets_sent: 494832
  packets_recv: 45230
  packets_bad: 0
  packets_back: 494820
- 
  name: "forearm_roll_mcb"
  board_type: 15011125
  board_revision: 5
  software_revision: 101
  boot_status: 0
  board_flags: 0
  system_time: 222716
  voltage: 28.421875
  temperature: 27.0
  packets_sent: 494832
  packets_recv: 45233
  packets_bad: 0
  packets_back: 494820
- 
  name: "wrist_flex_mcb"
  board_type: 15011126
  board_revision: 5
  software_revision: 101
  boot_status: 0
  board_flags: 0
  system_time: 222725
  voltage: 28.171875
  temperature: 28.0
  packets_sent: 494833
  packets_recv: 45235
  packets_bad: 0
  packets_back: 494821
- 
  name: "wrist_roll_mcb"
  board_type: 15011127
  board_revision: 5
  software_revision: 101
  boot_status: 0
  board_flags: 0
  system_time: 222715
  voltage: 28.46875
  temperature: 29.0
  packets_sent: 494833
  packets_recv: 45233
  packets_bad: 0
  packets_back: 494821
---

```

mqcmd196, 2024-04-10:
```
 name: "head_tilt_mcb"
  board_type: 15011132
  board_revision: 4
  software_revision: 101
  boot_status: 0
  board_flags: 4
  system_time: 222719
  voltage: 28.46875
  temperature: 33.0
  packets_sent: 494832
  packets_recv: 45932
  packets_bad: 0
  packets_back: 487751
```
からわかるようにboard_flagsが4になっている

mqcmd196, 2024-04-10:
正常に動いているとき
```
Every 1.0s: rosrun fetch_drivers read_board head_tilt         fetch1075: Wed Apr 10 18:17:[0/104]

Name            head_tilt
ID              0x15
Type:           15011132
Board Serial:   353635313436510d00360053
Board Rev:      4
Software Rev:   101
  Date:         2019_08_21_21_25_08
  Hash:         a02703ce1279fb3a7a8b7020953432d603c36a21
  Verify ID:    e741fcc21ee314aea5f28be6f53017a3fe79dd99785ed38a
  Flags:        0x80 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
Boot Status:    0
Temperature:    34
Voltage:        28.4375 V
Board Flags:    0
Recv Packets:   16396
Bad Packets:    0
System Time:    75080 ms
Motor
  State:        STOPPED
  Error:        NO ERROR
  Temp:         nan
  Angle Offset: -0.361154
  Ratio:        -57.6096
Joint
  Position:     -0.00606501
  Velocity:     0.00115299
  Effort:       -0
  Offset:       1.42768
  Ratio:        1
```
ジョイスティックでtorsoを上下させたり，head_tiltを動かしたりするとサーボが落ちる．サーボが落ちているとき
```
Every 1.0s: rosrun fetch_drivers read_board head_tilt         fetch1075: Wed Apr 10 18:20:0[0/80]

Name            head_tilt
ID              0x15
Type:           15011132
Board Serial:   353635313436510d00360053
Board Rev:      4
Software Rev:   101
  Date:         2019_08_21_21_25_08
  Hash:         a02703ce1279fb3a7a8b7020953432d603c36a21
  Verify ID:    e741fcc21ee314aea5f28be6f53017a3fe79dd99785ed38a
  Flags:        0x80 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
Boot Status:    0
Temperature:    35
Voltage:        28.4219 V
Board Flags:    4 : JOINT_ENCODER_ERROR
Recv Packets:   50049
Bad Packets:    0
System Time:    226530 ms
Motor
  State:        STOPPED
  Error:        NO ERROR
  Temp:         nan
  Angle Offset: -0.361154
  Ratio:        -57.6096
Joint
  Position:     -0.710162
  Velocity:     -3.83258e-05
  Effort:       -0
  Offset:       1.42768
  Ratio:        1

```

mqcmd196, 2024-04-10:
現時点での結論ですが，MCU基盤とエンコーダをつなぐケーブルを，中のUSBハブたちが圧迫している線が濃厚です．修理ログをドライブにアップロードしています．

@sktometometo 中のUSBハブを一旦外に全部避けています．こちら明日，MCUのケーブルを圧迫しないよう中に埋め込んでいただけると助かります．私も一緒に作業します．

また再発したらReopenします

closed 2024-04-10 by mqcmd196

sktometometo, 2024-04-10:
内部に空間無いので外に置くのも考えましょう。頭部の話だよね？

mqcmd196, 2024-04-10:
はい、そうです。外部でもいいかもですねYoshiki Obinata

sktometometo, 2024-04-10:
一度断線してるなら断線箇所交換したほうがいいので断線箇所の特定とケーブルコネクタなら交換、基板ならハンダもり直すくらいもやろうか

sktometometo, 2024-04-10:
もし外に取り付けるなら、マウント作成なりでCADと3Dプリンタ触るいい機会なきがするので、もし時間あったらWindowsにSolidworksインストールして見ておいてください
@yoshiki-447

sktometometo, 2024-04-10:
wiliki の Solidworks -> Solidworksライセンス にインストール手順書いてあるので、周りの人に聞きながらやってみてください

yoshiki-447, 2024-04-17:
頭の土台のCADに、飛び出たハブを入れられる箱を付け加える感じでいいですか？
@sktometometo

sktometometo, 2024-04-17:
詳細はお任せするけど要件をはっきりさせておくと良くて

- 何を固定する必要があるのか
  - ケーブルか基板か
- 何を保護する必要があるのか
  - 外部からなにかぶつかったときの保護か
  - 力が加わったときにコネクタやケーブルを保護するのか
- 何か置いたりする必要があるのか

あたりを作りながら考えてみるといいと思う。