jsk-ros-pkg/jsk_robot #1915: pepper 2.9での動作確認

k-okada, 2024-06-04:
From: @kochigami 

Pepperなのですが、Naoqiのバージョンが2.5だったのが今年から全員機体交換になり、2.9 (android) になりました。
関節角度列を送る方法が変わり、
https://github.com/ros-naoqi/naoqi_driver?tab=readme-ov-file#check-that-the-node-is-running-correctly
にあるように、/joint_angles (naoqi_bridge_msgs/JointAnglesWithSpeed) を使わないといけなくなりました。
 
そのため、jsk_robotのパッケージやpeppereusが現状使えなくなっている状態です。
(cross だけでなく、jsk_pepper_startup.launchも使えなくなりました。安全装置も解除できなくなり、手引き移動もできなくなってしまいました。)

ひとまず、そのまま 関節角度列はtopic を送って使うようにしようと思うのですが、
時間がある時に jsk_robotパッケージ（特にpeppereus）が使えるように直していけたらいいなと思います。

k-okada, 2024-06-04:
> まずは現状どうなっているか確認ですが、peppereus を使うときって pose_manager.launchを必ず立ち上げる、でよかったっけ？

はい，そうです。

pepper-interface.lの中にもnaoqi-interface.lの中にも該当部分が見つけられていなくて、
もしかしたら継承している pr2-interface.lの中にあるのかなと思ったのですが、

:angle-vectorを送ると/pepper_robot/pose/joint_trajectory/goal (naoqi_bridge_msgs/JointTrajectoryActionGoal) に指令が送られるようです。
これは pose_manager.launch に含まれています。（後述の確認１）

Naoqi 2.9のバージョンだと，
Ubuntu 18.04のパソコン(Python2) で試しても，
pepper_bringup/launch/pepper_full.launchの中身である：
- roslaunch naoqi_driver naoqi_driver.launch network_interface:=wlp0s20f3 nao_port:=9503 password:=nao # これは立ち上がる
- roslaunch naoqi_pose pose_manager.launch # これがエラーになる．nao_port:=9503 に変えても駄目
pose_manager.launch が上がらず，/pepper_robot/pose/joint_trajectory/goalが使えない状況です．

pose_manager.launch の中身はPythonプログラムなので，
いずれにせよ，naoqiのpythonプログラムは，Python2の NAOqi SDK（pynaoqi-python2.7-2.5.5.5-linux64）を必要としているため，20.04以降は動かなくなる現状があります。

なので，/joint_angles (naoqi_bridge_msgs/JointAnglesWithSpeed)をpeppereusで使えるようにするか、naoqi_driverのプログラムの中で、/pepper_robot/pose/joint_trajectory アクションを使えるようにする、という方向がありそうだと思いました。

（実は，もともとnaoqi_driver.launchを上げた時に，/joint_angles (naoqi_bridge_msgs/JointAnglesWithSpeed) は前から使えるようになっていたと知りました。）

後者のほうが正しそうなので、できそうか考えてみます。
(/joint_angles をpeppereusで送れるようになっても、複数時間ごとの角度列が送れないので、ちょっと不便かもしれないと思っております。)

k-okada, 2024-06-04:
@a-ichikura こういう状況です。

@kochigami 勝手に引用してすいません。この件、放置しちゃっていてごめんなさい。ちょうど @a-ichikura が https://www.rosielab.ca/ に訪問中で、先生はロボット動かしてほしいけど、学生はデータセットにした興味ない、みたいな状況なので、よいついでなので、３研究室のロボットが動いたら良いなと思って、Issueにしました。

kochigami, 2024-06-04:
いえいえ、皆さんにも関係する話になったみたいで良かったです。

割り込み仕事がなければ、今週ちょうど取り組む予定でした（ですが、とりあえず既存のデモが動くように関節角度列をPythonで送るプログラムを書いて試すだけになりそうです・・・）。
また情報共有いたします。

a-ichikura, 2024-06-04:
ありがとうございます。こちら(rosie lab)のPepperについても確認中です。

>Pepperなのですが、Naoqiのバージョンが2.5だったのが今年から全員機体交換になり、2.9 (android) になりました。
関節角度列を送る方法が変わり、
https://github.com/ros-naoqi/naoqi_driver?tab=readme-ov-file#check-that-the-node-is-running-correctly
にあるように、/joint_angles (naoqi_bridge_msgs/JointAnglesWithSpeed) を使わないといけなくなりました。

というのは、機体そのものが交換になったという解釈で正しいでしょうか？
JSKのPepperは機体自体は変わっていないはず（そしてアップデートなどもしていないはず）なので、
特に変わっていないかと思うのですがいかがでしょうか? ==> @k-okada

kochigami, 2024-06-04:
> 機体そのものが交換になったという解釈で正しいでしょうか？

はい、そうです！
rosie labのPepperのNAOqiが2.5だったら、jsk_robotのパッケージそのまま使えると思います。

a-ichikura, 2024-06-04:
東風上さん

ご返信ありがとうございます。
2.9も2.5も両方あるそうなのでjsk_robotのパッケージをそのまま使う場合は2.5の方を使おうと思います。

k-okada, 2024-06-20:
@kochigami ようやくデバッグ環境ができました．．
https://github.com/jsk-ros-pkg/jsk_robot/pull/1918/files
で，`(setq *ri* (instance pepper-interface :init :type :naoqi-controller-disabled)) ` として `*ri*` を作ると動くとおもいます．

でも，`(/joint_angles をpeppereusで送れるようになっても、複数時間ごとの角度列が送れないので、ちょっと不便かもしれないと思っております。)` が正しくて，さらに`enable-life`とか`pose_manager` 経由で動かしていそうなので，pythonから動くようにする，というのが良い気がします．

で，@a-ichikura が 2.9 ＋Python で動かしていて，2.5で動くコードとの差分が以下のようになっていて，最初の認証周りで何か更新がありそうなので，この辺を頑張れば良さそうな気がしています．
```
    def __init__(self):
        self.app = qi.Application(sys.argv, url="tcp://xx.xx.xx.xx:9559") ;; 2.5で確認
        #
        # naoqi 2.5 have following erros
        #
        # AttributeError: 'Object' object has no attribute 'setClientAuthenticatorFactory'
        #self.app = qi.Application(sys.argv, url="tcps://xx.xx.xx.xx:9503") ;; 2.9で動いているもの
        #logins = ("xxx", "xxx")
        #factory = AuthenticatorFactory(*logins)
        #self.app.session.setClientAuthenticatorFactory(factory)
        #
        self.app.start()
        self.autonomous_life = self.app.session.service("ALAutonomousLife")
        self.motion_service = self.app.session.service("ALMotion")
        self.posture_service = self.app.session.service("ALRobotPosture")

```
で，そのまえにいくつか確認したいんだけど，，，，，
- `naoqi_driver`, `naoqi_bridge`　等は，どのリポジトリを使っている？ `ros-naoqi/master` ? `kochigami/kochigami-develop` ??
- pythonのlibqi? qi? みたいなのが必要だと思うけど，どうやって入れている？ pynaoqi-python2.7-2.5.5.5-linux64 みたいなやつ？それとも pip install qi ???(← @a-ichikura は 2.9 をこれで使っていると言っている）
- 以下のコマンドを打つとどういう値が返ってくる？
```
>>> from qi import version
>>> version
<module 'qi.version' from '/home/k-okada/pynaoqi/pynaoqi-python2.7-2.5.5.5-linux64/lib/python2.7/site-packages/qi/version.pyc'>
```
- `pose_manager.launch` が動かないということだけど，`roslaunch --screen naoqi_pose pose_manager.launch nao_port:=9503` の出力ログを教えて下さい．

k-okada, 2024-06-21:
naoqi_bridge も含めて 多分ですが，Pepper2.9 で動く様になったとおもいます．（実機でもシミュレーションでも確認していない，妄想100%なので，割り引いて読んで下さい）

1 : (僕は20.04を使ったけど，多分18.04でもOK)　https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_naoqi_robot#setup-environment の要領で環境をセットアップする．正確には `3Install ROS packages for NAO and Pepper` の最初の手順（`mkdir`, `cd`, `wstool`から`rosdep`まで）行う．1, 2 は飛ばして良い．ただし，NAO_IP, ROS_IP はセットしておくと楽．

2: naoqi_driverはkochigami-developではなく，https://github.com/ros-naoqi/naoqi_driver/tree/master  を，naoqi_bridge　は https://github.com/k-okada/naoqi_bridge/tree/master に入れ替える．

3: `pip install qi` をする．僕のは
```
>>> import qi
>>> qi.version
>>> qi.__version__
'3.1.5'
```

4: https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_naoqi_robot#setup-environment の要領で環境をセットアップの次ぐ気を行う．正確には `3Install ROS packages for NAO and Pepper` の`sudo apt-get install ros-${ROS_DISTRO}-pepper-meshes` した後，一覧の`catkin build XXXX` と `source devel/setup.bash` を行う

5: https://github.com/jsk-ros-pkg/jsk_robot/issues/1474#issuecomment-1110768907 に従い，`naoqi_driver/share/boot_config.json` を修正

6: `roslaunch pepper_bringup pepper_full.launch nao_ip:=$NAO_IP nao_port:=9503  roscore_ip:=$ROS_IP network_interface:=enpXXXX user:=XXX password:=XXX` で動くと思う．
   2.5 のPepperでは，`roslaunch pepper_bringup pepper_full.launch nao_ip:=$NAO_IP roscore_ip:=$ROS_IP network_interface:=enpXXXX` で動いた

c.f. https://github.com/ros-naoqi/naoqi_bridge/pull/98

kochigami, 2024-06-21:
ありがとうございます。
１つ前のコメントへの返信です。

Ubuntu 20.04 ROS noetic
Pepper NAOqi 2.9

今は、naoqi_driverだけとりあえず使えるようになっています。

jsk_naoqi_robot のREADMEにある環境構築のやり方 (Setup environmentの 3. Install ROS packages for NAO and Pepper) をやって、（pynaoqi-python2.7-2.5.5.5-linux64は入れていないです）
naoqi_libqi, naoqi_libqicore を入れて、
最終的には naoqi_driver: ros-naoqiのmasterブランチに変えています。

- use ros-naoqi/naoqi_driver master branch
```
cd ~/catkin_ws/src
git clone git@github.com:ros-naoqi/libqi.git
git clone git@github.com:ros-naoqi/libqicore.git

cd ~/catkin_ws/src/naoqi_driver
git remote add ros-origin git@github.com:ros-naoqi/naoqi_driver.git
git fetch ros-origin
git checkout -b ros-master ros-origin/master
catkin b
```

Python3 になったので、naoqi_appsなど、pynaoqi-python2.7-2.5.5.5-linux64 を必要とするものは使えないと思っているのですが、そうではないのでしょうか。（めちゃめちゃエラーが出て心が折れた記憶があります。）

> 以下のコマンドを打つとどういう値が返ってくる？
`pip install qi`などしていないので、以下のようになりました。

```
ipython3
Python 3.8.10 (default, Nov 22 2023, 10:22:35) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.13.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from qi import version                                                  
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-1-5f15c41eec7f> in <module>
----> 1 from qi import version

ModuleNotFoundError: No module named 'qi'

In [2]: version                                                                 
---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
<ipython-input-2-605b5d1778ad> in <module>
----> 1 version

NameError: name 'version' is not defined

In [3]:  
```

> `pose_manager.launch` が動かないということだけど，`roslaunch --screen naoqi_pose pose_manager.launch nao_port:=9503` の出力ログを教えて下さい．

NAO_IP, ROS_IPを.bashrcにセットした状態で、
```
roslaunch --screen naoqi_pose pose_manager.launch nao_port:=9503

... logging to /home/kochigami/.ros/log/1101c96a-2fbc-11ef-9d78-bb12a8109141/roslaunch-kochigami-ThinkPad-X13-Gen-3-1482349.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.8.162:45045/

SUMMARY
========

PARAMETERS
 * /pose_manager/poses/init/joint_names: ['Body']
 * /pose_manager/poses/init/positions: [0.0, 0.0, 1.39, ...
 * /pose_manager/poses/init/time_from_start: 1.5
 * /pose_manager/poses/zero/joint_names: ['Body']
 * /pose_manager/poses/zero/positions: [0.0, 0.0, 0.0, 0...
 * /pose_manager/poses/zero/time_from_start: 1.5
 * /pose_manager/xap: /home/kochigami/c...
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    pose_controller (naoqi_pose/pose_controller.py)
    pose_manager (naoqi_pose/pose_manager.py)

auto-starting new master
process[master]: started with pid [1482357]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 1101c96a-2fbc-11ef-9d78-bb12a8109141
process[rosout-1]: started with pid [1482367]
started core service [/rosout]
process[pose_controller-2]: started with pid [1482374]
process[pose_manager-3]: started with pid [1482375]
Traceback (most recent call last):
  File "/home/kochigami/catkin_ws/devel/lib/naoqi_pose/pose_controller.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/kochigami/catkin_ws/src/naoqi_bridge/naoqi_pose/./nodes/pose_controller.py", line 153
    except RuntimeError,e:
                       ^
SyntaxError: invalid syntax
================================================================================REQUIRED process [pose_controller-2] has died!
process has died [pid 1482374, exit code 1, cmd /home/kochigami/catkin_ws/devel/lib/naoqi_pose/pose_controller.py --pip=192.168.8.113 --pport=9503 __name:=pose_controller __log:=/home/kochigami/.ros/log/1101c96a-2fbc-11ef-9d78-bb12a8109141/pose_controller-2.log].
log file: /home/kochigami/.ros/log/1101c96a-2fbc-11ef-9d78-bb12a8109141/pose_controller-2*.log
Initiating shutdown!
================================================================================
[pose_manager-3] killing on exit
[pose_controller-2] killing on exit
Traceback (most recent call last):
  File "/home/kochigami/catkin_ws/devel/lib/naoqi_pose/pose_manager.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/kochigami/catkin_ws/src/naoqi_bridge/naoqi_pose/./nodes/pose_manager.py", line 174, in <module>
    manager = PoseManager()
  File "/home/kochigami/catkin_ws/src/naoqi_bridge/naoqi_pose/./nodes/pose_manager.py", line 55, in __init__
    rospy.init_node('pose_manager')
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/client.py", line 323, in init_node
    raise rospy.exceptions.ROSInitException("init_node interrupted before it could complete")
rospy.exceptions.ROSInitException: init_node interrupted before it could complete
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
出ているエラーは直せそうなのですが、これを直しただけでは（python2.7の記法を3に直しただけでは）解決する問題ではなかったという記憶があります。結局　pynaoqi-python2.7-2.5.5.5-linux64 の中身をいじらないといけなくなって、心が折れた記憶があります。

よろしくお願いいたします。

kochigami, 2024-06-21:
↑ 今更の返信ですみませんでした・・・

https://github.com/jsk-ros-pkg/jsk_robot/issues/1915#issuecomment-2182256494
こちら試します。ありがとうございます。

a-ichikura, 2024-06-22:
@k-okada 

> naoqi_bridge も含めて 多分ですが，Pepper2.9 で動く様になったとおもいます．（実機でもシミュレーションでも確認していない，妄想100%なので，割り引いて読んで下さい）
> 
> 1 : (僕は20.04を使ったけど，多分18.04でもOK)　https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_naoqi_robot#setup-environment の要領で環境をセットアップする．正確には `3. Install ROS packages for NAO and Pepper` の最初の手順（`mkdir`, `cd`, `wstool`から`rosdep`まで）行う．1, 2 は飛ばして良い．ただし，NAO_IP, ROS_IP はセットしておくと楽．
> 
> 2: naoqi_driverはkochigami-developではなく，https://github.com/ros-naoqi/naoqi_driver/tree/master を，naoqi_bridge　は https://github.com/k-okada/naoqi_bridge/tree/master に入れ替える．
> 
> 3: `pip install qi` をする．僕のは
> 
> ```
> >>> import qi
> >>> qi.version
> >>> qi.__version__
> '3.1.5'
> ```
> 
> 4: https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_naoqi_robot#setup-environment の要領で環境をセットアップの続きを行う．正確には `3. Install ROS packages for NAO and Pepper` の`sudo apt-get install ros-${ROS_DISTRO}-pepper-meshes` した後，一覧の`catkin build XXXX` と `source devel/setup.bash` を行う
> 
> 5: [#1474 (comment)](https://github.com/jsk-ros-pkg/jsk_robot/issues/1474#issuecomment-1110768907) に従い，`naoqi_driver/share/boot_config.json` を修正
> 
> 6: `roslaunch pepper_bringup pepper_full.launch nao_ip:=$NAO_IP nao_port:=9503 roscore_ip:=$ROS_IP network_interface:=enpXXXX user:=XXX password:=XXX` で動くと思う． 2.5 のPepperでは，`roslaunch pepper_bringup pepper_full.launch nao_ip:=$NAO_IP roscore_ip:=$ROS_IP network_interface:=enpXXXX` で動いた
> 
> c.f. [ros-naoqi/naoqi_bridge#98](https://github.com/ros-naoqi/naoqi_bridge/pull/98)

こちらに関して、
1.2.4まで行いました。
4のbuildでエラーは出ませんでした。

そして、3.の`pip install qi`なのですが、私のdefaultのpython3が、Python 3.6で
```
$ python3
Python 3.6.9 (default, Mar 10 2023, 16:46:00) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```
qiのドキュメントを読むと3.1.5はpython 3.7からしかインストール出来ないようです。https://pypi.org/project/qi/#description
```
Python with at least version 3.7 and its development libraries.

        On Ubuntu: apt-get install libpython3-dev.

```

私がPythonのみでPepper 2.9を動かしているときはanaconda を使っていて、Python3.8なので
```
$ python
Python 3.8.19 (default, Mar 20 2024, 19:58:24) 
[GCC 11.2.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```
問題なくpip install qiが出来ました。

Python3.7以上を別で入れることも考えたのですが環境を壊しそうで怖かったので一旦やめておきました。したがって5以降はやっていないです。(5はあんまり関係ないですが‥）

k-okada, 2024-06-24:
@kochigami 僕も20.04 なんだけど、pythonは3.8です。Ubuntu 20.04のデフォルトのPythonは3.8ではないでしょうか？
https://packages.ubuntu.com/search?keywords=python3

3.6にしているのは何か理由がありますか？
```
$ which python
/usr/bin/python
$ ls -al /usr/bin/python
lrwxrwxrwx 1 root root 7 Apr 15  2020 /usr/bin/python -> python3
$ ls -al /usr/bin/python3
lrwxrwxrwx 1 root root 9 Mar 13  2020 /usr/bin/python3 -> python3.8
$ dpkg -S /usr/bin/python3.8
python3.8-minimal: /usr/bin/python3.8
```

kochigami, 2024-06-24:
@k-okada => @a-ichikura の間違いですかね。私はPython3.8だと思われます。

a-ichikura, 2024-06-24:
>3.6にしているのは何か理由がありますか？

Python 3.6にしている理由は特にないのですが、Ubuntuは 18.04です。

k-okada, 2024-06-24:
@kochigami そうでした。どこ見てたんだろう？

> Python3 になったので、naoqi_appsなど、pynaoqi-python2.7-2.5.5.5-linux64 を必要とするものは使えないと思っているのですが、そうではないのでしょうか。

これ自体は正しくて、https://github.com/ros-naoqi/naoqi_bridge/pull/98 で、Python3 への対応＋`pip install qi` で入るPythonモジュールで動くように変更してあります。

> pip install qiなどしていないので、以下のようになりました。

`pip install qi` しましょう。多分大丈夫。最悪、この時のログを全部残しておけば、元に戻せます

> 出ているエラーは直せそうなのですが、これを直しただけでは（python2.7の記法を3に直しただけでは）解決する問題ではなかったという記憶があります。結局　pynaoqi-python2.7-2.5.5.5-linux64 の中身をいじらないといけなくなって、心が折れた記憶があります。

`これを直す` -> https://github.com/ros-naoqi/naoqi_bridge/pull/9 で直ってるはずです。
`。結局　pynaoqi-python2.7-2.5.5.5-linux64 の中身をいじらないといけなくなって` -> そこをいじるのではなくて、`naoqi_bridge` を pi (NOT pynaoqi) に対応するというのが、https://github.com/ros-naoqi/naoqi_bridge/pull/9 で行ったことです。

> naoqi_appsなど

https://github.com/ros-naoqi/naoqi_driver/tree/master が動いたら、これベースで再度 kochigami-develop を作る
https://github.com/ros-naoqi/naoqi_bridge/pull/9 が動いたらマージする
上の2つが動いたらnaoqi_apps 等も確認する。

という予定です。

kochigami, 2024-06-25:
@k-okada　ありがとうございます。

Ubuntu 20.04, ROS noeticの環境で、

https://github.com/jsk-ros-pkg/jsk_robot/issues/1915#issuecomment-2182256494
を試しました。

```
pip install qi
Collecting qi
  Downloading qi-3.1.5-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (7.4 MB)
     |████████████████████████████████| 7.4 MB 92 kB/s 
Installing collected packages: qi
Successfully installed qi-3.1.5
```
```
ipython3
Python 3.8.10 (default, Nov 22 2023, 10:22:35) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.13.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import qi                                                               

In [3]: qi.__version__                                                          
Out[3]: '3.1.5'
```
ちなみに、naoqi_libqi, naoqi_libqicoreはソースでは入れていないがaptで入っている
```
aptitude search ros-noetic-naoqi-libqi
i   ros-noetic-naoqi-libqi          - Aldebaran's libqi: a core library for NAOq
p   ros-noetic-naoqi-libqi-dbgsym   - debug symbols for ros-noetic-naoqi-libqi  
i   ros-noetic-naoqi-libqicore      - Aldebaran's libqicore: a layer on top of l
p   ros-noetic-naoqi-libqicore-dbgs - debug symbols for ros-noetic-naoqi-libqico

aptitude search ros-noetic-naoqi-libqicore
i   ros-noetic-naoqi-libqicore      - Aldebaran's libqicore: a layer on top of l
p   ros-noetic-naoqi-libqicore-dbgs - debug symbols for ros-noetic-naoqi-libqico
```

`.bashrc`に`NAO_IP`と`ROS_IP`をセットした状態で、
`roslaunch pepper_bringup pepper_full.launch nao_port:=9503 network_interface:=enpXXXX password:=XXX`

を実行したところ、エラーが出ました。`NameError: name 'ALProxy' is not defined`

```
roslaunch pepper_bringup pepper_full.launch nao_port:=9503 network_interface:=XXX password:=XXX
```

```
... logging to /home/kochigami/.ros/log/41836716-32d1-11ef-8c78-3d4ef9921c6e/roslaunch-kochigami-ThinkPad-P16-Gen-1-26768.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://169.254.179.2:42437/

SUMMARY
========

PARAMETERS
 * /pepper_robot/camera/bottom_rectify_color/queue_size: 5
 * /pepper_robot/camera/camera_nodelet_manager/num_worker_threads: 4
 * /pepper_robot/camera/depth_metric/queue_size: 5
 * /pepper_robot/camera/depth_metric_rect/queue_size: 5
 * /pepper_robot/camera/depth_rectify_depth/interpolation: 0
 * /pepper_robot/camera/depth_rectify_depth/queue_size: 5
 * /pepper_robot/camera/depth_registered_hw_metric_rect/queue_size: 5
 * /pepper_robot/camera/depth_registered_metric/queue_size: 5
 * /pepper_robot/camera/depth_registered_rectify_depth/interpolation: 0
 * /pepper_robot/camera/depth_registered_rectify_depth/queue_size: 5
[... 96 lines cut ...]
[pepper_robot/camera/depth_metric-11] killing on exit
[pepper_robot/camera/depth_metric_rect-10] killing on exit
[pepper_robot/camera/depth_rectify_depth-9] killing on exit
[pepper_robot/camera/ir_rectify_ir-8] killing on exit
[pepper_robot/camera/bottom_rectify_color-7] killing on exit
[pepper_robot/camera/front_rectify_color-6] killing on exit
[pepper_robot/camera/camera_nodelet_manager-5] killing on exit
[pepper_robot-2] killing on exit
[pepper_robot/pose/pose_controller-3] killing on exit
[pepper_robot/pose/pose_manager-4] killing on exit
[FATAL] [1719305982.557752]: Could not connect to required "joint_trajectory" action server, is the nao_controller node running?
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

https://github.com/ros-naoqi/pepper_robot/pull/59
の、
https://github.com/ros-naoqi/pepper_robot/pull/59/files#diff-7fa4436ded590378f825aaf6ed0af1599635e046c9cc2a9526363f586bd01f72R10
のuserをusernameに変えたもので試しても、エラーは変わらずでした。

また、https://github.com/jsk-ros-pkg/jsk_robot/issues/1915#issuecomment-2180453437
については、現状 `roslaunch naoqi_driver naoqi_driver.launch nao_port:=9503 network_interface:=XXX password:=XXX`しか動かないため、実機Pepperとつながらず、実行できませんでした。

k-okada, 2024-06-25:
おかしいですね。
https://github.com/aldebaran/libqi-python/issues/22#issuecomment-1941618222
のサンプルコードで接続はできるでしょうか？

a-ichikura, 2024-06-25:
@k-okada 

岡田先生

pip install qi=1.8.3をした状態で
```
$ python2 
Python 2.7.17 (default, Mar  8 2023, 18:40:28) 
[GCC 7.5.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import qi
>>> 
```

以下を実行したところ
```
$ roslaunch pepper_bringup pepper_full.launch nao_ip:=$NAO_IP nao_port:=9503 roscore_ip:=$ROS_IP network_interface:=XXX user:=XXX passowrd:=XXX
```

次のようなエラーが出ました。
```
disconnected
[WARN] [1719353377.159321]: connectiong to 10.0.0.49503:9559 with nao/no_password
[W] 1719353377.182616 18483 qi.FutureSync: Error in future on destruction: 'disconnected' - continuing stack unwinding...
Traceback (most recent call last):
  File "/home/ichikura/pepper2.9_ws/src/naoqi_bridge/naoqi_pose/nodes/pose_controller.py", line 465, in <module>
    controller = PoseController()
  File "/home/ichikura/pepper2.9_ws/src/naoqi_bridge/naoqi_pose/nodes/pose_controller.py", line 57, in __init__
    NaoqiNode.__init__(self, 'pose_controller')
  File "/home/ichikura/pepper2.9_ws/src/naoqi_bridge/naoqi_driver_py/src/naoqi_driver/naoqi_node.py", line 138, in __init__
    self.app.start()
RuntimeError: disconnected
================================================================================REQUIRED process [pepper_robot-2] has died!
process has finished cleanly
log file: /home/ichikura/.ros/log/9b4ce344-333f-11ef-aaf7-287fcf9948c3/pepper_robot-2*.log
Initiating shutdown!
================================================================================
```

pepper_full.launchの中身を見てみたら、`naoqi_driver`,`naoqi_pose`, `perception.launch`のようでしたのでそれぞれ別々に実行したところ

naoqi_driver.launch ▶　エラーのようなものは見えないので立ち上がったっぽい
perception.launch ▶ 上に同じくエラーのようなものは見えないので立ち上がったっぽい

naoqi_poseに関して、
```
SUMMARY
========

PARAMETERS
 * /pose_manager/poses/init/joint_names: ['Body']
 * /pose_manager/poses/init/positions: [0.0, 0.0, 1.39, ...
 * /pose_manager/poses/init/time_from_start: 1.5
 * /pose_manager/poses/zero/joint_names: ['Body']
 * /pose_manager/poses/zero/positions: [0.0, 0.0, 0.0, 0...
 * /pose_manager/poses/zero/time_from_start: 1.5
 * /pose_manager/xap: /home/ichikura/pe...
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    pose_controller (naoqi_pose/pose_controller.py)
    pose_manager (naoqi_pose/pose_manager.py)

auto-starting new master
process[master]: started with pid [18267]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 65138bde-333f-11ef-aaf7-287fcf9948c3
process[rosout-1]: started with pid [18285]
started core service [/rosout]
process[pose_controller-2]: started with pid [18292]
process[pose_manager-3]: started with pid [18293]
[WARN] [1719353286.032312]: connectiong to 10.0.0.49503:9559 with nao/nao
[W] 1719353286.086774 18292 qi.FutureSync: Error in future on destruction: 'disconnected' - continuing stack unwinding...
Traceback (most recent call last):
  File "/home/ichikura/pepper2.9_ws/src/naoqi_bridge/naoqi_pose/nodes/pose_controller.py", line 465, in <module>
    controller = PoseController()
  File "/home/ichikura/pepper2.9_ws/src/naoqi_bridge/naoqi_pose/nodes/pose_controller.py", line 57, in __init__
    NaoqiNode.__init__(self, 'pose_controller')
  File "/home/ichikura/pepper2.9_ws/src/naoqi_bridge/naoqi_driver_py/src/naoqi_driver/naoqi_node.py", line 138, in __init__
    self.app.start()
RuntimeError: disconnected
[pose_controller-2] process has died [pid 18292, exit code 1, cmd /home/ichikura/pepper2.9_ws/src/naoqi_bridge/naoqi_pose/nodes/pose_controller.py --pip=10.0.0.4 --pport=9503 --user=nao --password=nao __name:=pose_controller __log:=/home/ichikura/.ros/log/65138bde-333f-11ef-aaf7-287fcf9948c3/pose_controller-2.log].
log file: /home/ichikura/.ros/log/65138bde-333f-11ef-aaf7-287fcf9948c3/pose_controller-2*.log
[FATAL] [1719353289.020441]: Could not connect to required "joint_trajectory" action server, is the nao_controller node running?
[pose_manager-3] process has finished cleanly
log file: /home/ichikura/.ros/log/65138bde-333f-11ef-aaf7-287fcf9948c3/pose_manager-3*.log
^C[rosout-1] killing on exit
[master] killing on exit
^Cshutting down processing monitor...
... shutting down processing monitor complete
done
```
となっていました。
nao_ipとportが
おかしなこと（10.0.0.49503:9559)になっているので、disconnectedなんだろうなということまでかんがえたのですが、その先むやみにいじるのはやめておこうと思ってここまでにしました。

k-okada, 2024-06-26:
@kochigami @a-ichikura ごめんあさい．間違いありました
https://github.com/k-okada/naoqi_bridge/tree/master
を最新にするのと，pepper_robotのリポジトリも https://github.com/k-okada/pepper_robot を使って下さい．

確認事項としては
> [WARN] [1719353377.159321]: connectiong to 10.0.0.49503:9559 with nao/no_password
> [WARN] [1719353286.032312]: connectiong to 10.0.0.49503:9559 with nao/nao

の部分を表示も，コードも直しているんですが，
no_password がでちゃっていると絶対動かない，というのと，表示がなおっていれば
`tcps://10.0.0.4:9503 with nao/nao`
になるはずです．ここが `tcp://`ではなく，`tcps://` である，というのと 9503というのが2.9の条件だと思っています．
https://github.com/k-okada/naoqi_bridge/commit/b95bb22fc6676a565aed3ff4f27299fc4dc18ab4

kochigami, 2024-06-26:
ありがとうございます。
Ubuntu 20.04, ROS noeticでは、
https://github.com/jsk-ros-pkg/jsk_robot/issues/1915#issuecomment-2182256494
と
> https://github.com/k-okada/naoqi_bridge/tree/master
> を最新にするのと，pepper_robotのリポジトリも https://github.com/k-okada/pepper_robot を使って下さい．

で
`roslaunch pepper_bringup pepper_full.launch nao_port:=9503 network_interface:=XXX password:=XXX`がエラーなく実行できました。

その上で、https://github.com/jsk-ros-pkg/jsk_robot/issues/1915#issuecomment-2180453437
を試し、関節角度列が送れることを確認しました。

```
roseus pepper-interface.l
pepper-init t
(setq *ri* (instance pepper-interface :init :type :naoqi-controller-disabled))
send *ri* :angle-vector (send *pepper* :reset-pose)
send *pepper* :head :neck-p :joint-angle -10
send *ri* :angle-vector (send *pepper* :angle-vector)
send *pepper* :larm :shoulder-p :joint-angle 10
send *ri* :angle-vector (send *pepper* :angle-vector)
send *ri* :angle-vector (send *pepper* :reset-pose)
```
`/joint_angles`のデフォルトのスピードが結構早く、ペッパーがガコガコするので、修正が必要そうです。
（今日はここで時間切れになりました。すみません・・・）

k-okada, 2024-06-26:
おお，良かったです．
で，新しい `naoqi_bridge (https://github.com/k-okada/naoqi_bridge/tree/master) ` を使うと，
```
(setq *ri* (instance pepper-interface :init :type :naoqi-controller-disabled))
```
は必要なくて，今までどおり
```
(setq *ri* (instance pepper-interface :init))
```
のままで，
```
/pepper_robot/pose/joint_trajectory/goal (naoqi_bridge_msgs/JointTrajectoryActionGoal) 
```
が送られるはずです．時間があるときに試してみて下さい．

a-ichikura, 2024-06-26:
@k-okada 

> @kochigami @a-ichikura ごめんなさい．間違いありました https://github.com/k-okada/naoqi_bridge/tree/master を最新にするのと，pepper_robotのリポジトリも https://github.com/k-okada/pepper_robot を使って下さい．
> 
> 確認事項としては
> 
> > [WARN] [1719353377.159321]: connectiong to 10.0.0.49503:9559 with nao/no_password
> > [WARN] [1719353286.032312]: connectiong to 10.0.0.49503:9559 with nao/nao
> 
> の部分を表示も，コードも直しているので，最新の k-okada/noqi_bridge:master にして，もう一度確認してほしいんですが， そのときに，no_password がでちゃっていると絶対動かない，というのと，表示がなおっていれば `tcps://10.0.0.4:9503 with nao/nao` になるはずです．ここが `tcp://`ではなく，`tcps://` である，というのと 9503というのが2.9で動く条件だと思っています． [k-okada/naoqi_bridge@b95bb22](https://github.com/k-okada/naoqi_bridge/commit/b95bb22fc6676a565aed3ff4f27299fc4dc18ab4)

こちらレポジトリを最新にしてbuildし直す等を行って、手元のプログラム的には問題ないと思う（仮）のですが…

pose_controllerについて

```
SUMMARY
========

PARAMETERS
 * /pose_manager/poses/init/joint_names: ['Body']
 * /pose_manager/poses/init/positions: [0.0, 0.0, 1.39, ...
 * /pose_manager/poses/init/time_from_start: 1.5
 * /pose_manager/poses/zero/joint_names: ['Body']
 * /pose_manager/poses/zero/positions: [0.0, 0.0, 0.0, 0...
 * /pose_manager/poses/zero/time_from_start: 1.5
 * /pose_manager/xap: /home/ichikura/pe...
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    pose_controller (naoqi_pose/pose_controller.py)
    pose_manager (naoqi_pose/pose_manager.py)

auto-starting new master
process[master]: started with pid [10504]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 555b1100-3414-11ef-aaf7-287fcf9948c3
process[rosout-1]: started with pid [10522]
started core service [/rosout]
process[pose_controller-2]: started with pid [10529]
process[pose_manager-3]: started with pid [10530]
[WARN] [1719444743.629978]: connectiong to tcps://10.0.0.4:9503 with nao/XXX
[INFO] [1719444743.964772]: Connecting to NaoQi at 10.0.0.4:9503
[WARN] [1719444744.243485]: Collection LLeg not found on your robot.
[WARN] [1719444744.274869]: Collection RLeg not found on your robot.
[INFO] [1719444744.421164]: nao_controller initialized
[INFO] [1719444744.423259]: nao pose_controller running...
[WARN] [1719444746.618149]: stop_walk_srv not available, pose_manager will not stop the walker before executing a trajectory. This is normal if there is no nao_walker running
```

このようなエラーでした。`pose_manage.py`の
```
class PoseManager():
    def __init__(self):
        # ROS initialization:
        rospy.init_node('pose_manager')

        self.poseLibrary = dict()
        self.readInPoses()
        self.poseServer = actionlib.SimpleActionServer("body_pose", BodyPoseAction,
                                                       execute_cb=self.executeBodyPose,
                                                       auto_start=False)
        self.trajectoryClient = actionlib.SimpleActionClient("joint_trajectory", JointTrajectoryAction)
        if self.trajectoryClient.wait_for_server(rospy.Duration(3.0)):
            try:
                rospy.wait_for_service("stop_walk_srv", timeout=2.0)
                self.stopWalkSrv = rospy.ServiceProxy("stop_walk_srv", Empty)
            except:
                rospy.logwarn("stop_walk_srv not available, pose_manager will not stop the walker before executing a trajectory. "
                          +"This is normal if there is no nao_walker running.")
            self.stopWalkSrv = None
            self.poseServer.start()

            rospy.loginfo("pose_manager running, offering poses: %s", list(self.poseLibrary.keys()));
```
ここのエラーだということまでわかりました。serverにアクセスできていないということでしょうか。

Pepper 2台あるので2台で試してみたのですが同じエラーで、
autonomous lifeをdiableやinteractiveにしてみたりしたのですが状況変わらず、プログラムの問題なのかも‥となっています。

k-okada, 2024-06-27:
@a-ichikura 
```
[WARN] [1719444746.618149]: stop_walk_srv not available, pose_manager will not stop the walker before executing a trajectory. This is normal if there is no nao_walker running
```
はnaoでない限り，ワーニングが出るのはOKで僕の方でも出ています．
なので，一回 roscore も含めて全部落ちている状態で `pepper_full.launch` を立ち上げたら，roseus から動かないかな？

k-okada, 2024-06-27:
@kochigami 
```
/joint_anglesのデフォルトのスピードが結構早く、ペッパーがガコガコするので、修正が必要そうです。
```
もうこれは使わなくていいので，必要性は低いですがspeedの計算方法が間違えていて
https://github.com/k-okada/jsk_robot/commit/d01044266e348195b8c177189f5e530b1656c076
で修正しました
```

a-ichikura, 2024-06-28:
@k-okada 

岡田先生
お騒がせしました。launchした状態でroseusからangle-vectorが送れることを確認できました。
ついでに`/pepper_robot/camera/front/image_raw`がimage_viewで見えることも確認できました。

[image]

a-ichikura, 2024-07-05:
@k-okada 

pepper 2.5でのhand touchの認識ができなくなっている気がします。

まず、
```
 roslaunch pepper_bringup pepper_full.launch nao_ip:=$NAO_IP nao_port:=9559  roscore_ip:=$ROS_IP network_interface:=wlp1s0 user:=nao passoword:=nao
```
をPepper 2.5で実行したのですが、手を触っても`/pepper_robot/hand_touch`に反応がありません。(他のトピックは値に変化があります）

同じlaunchファイルを2.9で実行すると手を触ると0か1で値が変化します。

なぜこの問題に気がついたかというと、Pepper2.9で作った[プログラム](https://gitlab.jsk.imi.i.u-tokyo.ac.jp/ichikura/rosielab/-/blob/master/pepper/painting_motion/middlevel_soak.py?ref_type=heads)を2.5でも実行できるようにしたくて、試していたのですが、2.9だとうまくいくものの、
2.5だとhand touchのところでエラーになってしまいます。

例えば、
```
python middlelevel_soak.py 2.5 10.0.0.3
```
と実行すると、
途中手の反応を待つところで、
```
Traceback (most recent call last):
  File "middlevel_soak.py", line 372, in <module>
    pepper.waiting_hand_touch()
  File "middlevel_soak.py", line 248, in waiting_hand_touch
    self.touch = self.memory_service.subscriber("TouchChanged")
RuntimeError: Invalid signature
[W] 1720144300.104955 13128 qitype.signal: disconnect: No subscription found for SignalLink 18446744073709551615.
[W] 1720144300.104958 13127 qitype.signal: disconnect: No subscription found for SignalLink 12.
[W] 1720144300.105018 13128 qitype.signal: disconnect: No subscription found for SignalLink 18446744073709551615.
```
となります。
```
python middlelevel_soak.py 2.9 10.0.0.4
```
だとうまくいくことを確認しています。
余談ですが`highlevel_soak.py`はtouchをsubscribeしていないので2.5でも実行できました。（JSKのものでも実行出来るはずです）

ちなみに参考はこちらを見つけました。
https://stackoverflow.com/questions/73801385/python-3-5-invalid-signature-pepper-robot

この記事の通りにしてもsession.listen()ができないとなってしまうのでうまくいかなかったのですが、libQiのversionで問題が起きているようでした。

そこでROS経由で調べたところなんの反応もないことを見つけたのですが、JSK(岡田先生)の環境ではrosからhand touchが見えているでしょうか？

kochigami, 2024-07-05:
割り込みすみません。

詳しい状況を思い出せていないのですが、
18.04で同じ現象が前も起きていた気がします。
https://github.com/k-okada/jsk_robot/issues/62#issuecomment-1207368175

こちら試します。ありがとうございます。
https://github.com/jsk-ros-pkg/jsk_robot/issues/1915#issuecomment-2191237644

k-okada, 2024-07-05:
https://stackoverflow.com/questions/73801385/python-3-5-invalid-signature-pepper-robot
のサンプルを入れたら動きました（`def onTouched`のなかでprintして確認）

ただ，ちょっとわからないのは，このstackoverflow で説明してくれているのは，`libqi<3`だと，MySubscriberを使う必要がある，ということですが，僕の環境(18.04 (qi==1.8.3) + Pepper 2.5)　(20.04 (qi==3.1.5) + Pepper 2.5) でもMySubscriberを使う作戦でないとうごかないので，ロボット側のバージョン依存のようです．

```
import sys
import qi
app = qi.Application(sys.argv, url="tcp://133.11.216.58:9559")
app.start()
memory_service = app.session.service("ALMemory")

# Listen to public addresses on a random port,                                                 
# so that services registered from here are accessible.                                        
app.session.listen("tcp://0.0.0.0:0")

# An object that will be registered as a service.                                              
touched = False
class MySubscriber:
    # It needs at least a callback to get ALMemory events.                                     
    def onTouched(key, value):
        global touched
        touched = True
        print("touched {}{}".format(key,value))
        pass

subscriber = MySubscriber()
# Register it with an arbitrary name.                                                          
service_name = "MySubscriber"
app.session.registerService(service_name, subscriber)

# Subscribe and tell exactly which service and method to call back.                            
memory_service.subscribeToEvent("TouchChanged", service_name, "onTouched")
```

a-ichikura, 2024-07-05:
@k-okada 

ありがとうございます。

若干違う問題のような根本は同じ問題のような感じなのですが、Trackingをためそうと思って、
http://doc.aldebaran.com/2-5/naoqi/trackers/trackers-sample.html#trackers-sample
公式tutorialを参照してpepper 2.9で試しているのですが、
ALTrackerというserviceがないとなってしまいます。

他にも使えないserviceがあるのかなと思い色々していたところ、SonarやPeoplePerceptionなども同じエラーになりました。

```
$ python highlevel_soak.py 2.9 10.0.0.4
started
[W] 1720222047.046677 7515 qi.path.sdklayout: No Application was created, trying to deduce paths
please enter your command:start
Life state is:disabled
start to Stand Init
end up Stand Init
Traceback (most recent call last):
  File "highlevel_soak.py", line 359, in <module>
    con_mng_service = pepper.app.session.service("ALSonar")
RuntimeError: Cannot find service 'ALSonar' in index
```
APIのCoreにALExtractorというものがあり、http://doc.aldebaran.com/2-8/naoqi/core/alextractor-api.html
これがそもそも同じエラーになるので認識系(?)がqi==3.1.5ではうまくいっていない気がしました。

2.9で動かすときにはこの記事のようにappを作って動かしています。
https://stackoverflow.com/questions/77987028/how-can-i-connect-to-pepper-naoqi-2-9-via-libqi-python

なんとなくこれもlibQiのバージョンが関係している気がして、
かつPerceptionだけ仕組みが違う？のかなと思いました。
https://qisdk.softbankrobotics.com/sdk/doc/pepper-sdk/ch4_api/perception/tuto/index_tuto_perception.html

a-ichikura, 2024-07-06:
Aldebaranの公式にissueを投げてみました。
https://github.com/aldebaran/libqi-python/issues/26#issue-2393265240

k-okada, 2024-07-06:
なるほど、https://github.com/aldebaran/libqi-python/issues/26#issuecomment-2211579093 みてもPepper2.5/Pepper2.9は中身だいぶ違うみたいですね。たぶん本体側の問題なので `qi==3.1.5` でもPepper 2.5の機体につなげれば、使えるんだと思います。

また、Pepper2.9でも、どういうサービスがあるか、とかは

```
>>> map(lambda x: x['name'], app.session.services()) 
['ServiceDirectory', 'LogManager', 'PackageManager', 'ALServiceManager', 'ALCloudToken', 'ALFileManager', 'ALMemory', 'ALLog
ger', 'ALPreferences', 'ALConnectionManager', 'ALPreferenceManager', 'ALFrameManager', 'ALDebug', 'ALNotificationManager', 'ALNotificationAdder', 'DCM', 'ALTabletService', 'ALTactileGesture', '_ALExpressiveness', 'ALResourceManager', 'ALRobotModel', 'ALDiagnosis', 'ALSonar', 'ALFsr', 'ALSensors', 'ALBodyTemperature', 'ALMotion', 'ALTouch', 'ALRobotPosture', 'ALMotionRecorder', 'ALLeds', 'ALWorldRepresentation', '
```
みたいに調べられて、
```
>>> a = app.session.service('ALRecharge') 
>>> dir(a)
・・・
>>> a.getMethodList()
['isStatsEnabled', 'enableStats', '・・・・
````
みたいにして、どういうメンバ関数があるかわかります。一回、`app.session.services()` を実行して全てのサービスを教えてほしいのと、そのなかでHumanとかTrackとかLookとかが付くサービスのそれぞれのメンバ関数一覧を知りたいです。

https://qisdk.softbankrobotics.com/sdk/doc/pepper-sdk/ch4_api/movement/reference/lookat.html?highlight=track

a-ichikura, 2024-07-08:
@k-okada 

岡田先生ありがとうございます。

こちらが結果になります。

```
>>> map(lambda x: x['name'], app.session.services())
['ServiceDirectory', 'LogManager', 'PackageManager', 'ALServiceManager', 'ALMemory', 'Mapping', 'ContextFactory', 'AccessControl', '_ALNotificationAdder', 'ALNotificationManager', '_ALExpressiveness', 'MainConversation', '_ALNotificationReader', 'ALSystem', 'AutonomousAbilities', 'ALConnectionManager', 'ALPreferences', 'ALKnowledge', 'ALUserInfo', 'HumanAwareness', 'ALAudioDevice', 'ALAudioPlayer', 'Knowledge', 'LoLA', 'ALRobotModel', 'ALPreferenceManager', '_ConditionChecker_pepper_3234_0', '_NaoqiInformationForSemantic', 'ALExpressionWatcher', 'ALTactileGesture', '_ALCloud', '_SemanticEngine', 'Semantics', 'ALStore', 'ALDiagnosis', 'ALBodyTemperature', 'ActuationPrivate', 'ALRobotPosture', 'ALMotion', 'Touch', 'Actuation', 'ALTextToSpeech', 'ALTouch', 'ALVoiceEmotionAnalysis', 'ALFrameManager', 'ALBattery', 'ALLeds', 'ALResourceManager', 'ALAutonomousBlinking', 'ALLauncher', '_AsrCompiler2', 'ALBehaviorManager', 'ALPythonBridge', '_AsrCompiler1', 'ALVideoDevice', 'ALRobotHealthMonitor', 'ALSpeechRecognition', 'Conversation', 'LanguageManager', 'ALModularity', '_ALBrightnessStatistics', 'Camera', 'ALNavigation', 'HumanPerception', 'ALMood', 'BasicAwareness', 'LifeStarter', '_ConditionChecker_pepper_3233_0', 'Focus', 'ALAutonomousLife', 'ALPodDetection', 'ALAnimatedSpeech', '_ALMovementScheduler', 'ALSpeakingMovement', 'ALDialog', 'ALRecharge', 'ALListeningMovement', 'ALBackgroundMovement', 'ALAnimationPlayer', 'ALSignsAndFeedback', 'ALRobotMood', 'ALTabletService']
```

```
Human Awareness:
['__bool__', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', 'async', 'call', 'engagedHuman', 'humansAround', 'isValid', 'makeEngageHuman', 'metaObject', 'recommendedHumanToEngage']
Human Perception:
['__bool__', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', 'async', 'blobs', 'bodies', 'call', 'dumpBlackBox', 'faces', 'humansAroundPrivate', 'isValid', 'metaObject', 'setDebugMode']
Basic Awareness:
['__bool__', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '_getEmptyFrame', 'async', 'call', 'isEnabled', 'isRunning', 'isValid', 'metaObject', 'setEnabled', 'setStrategies', 'state', 'trackedHuman', 'zoneOfInterest']
```

TrackingやLookのようなものは
Human Awareness, Human Perception, Basic Awarenessの3つのような感じでした。
とりあえずHuman Awarenessが一番近そうな気がするので使ってみようと思います。

a-ichikura, 2024-07-08:
追加です。

ALMotionの関数にLookatやTrackerを見つけました。
I found the `lookat` function and functions for tracking.

```
ALMotion:
__bool__
__class__
__delattr__
__dict__
__dir__
__doc__
__eq__
__format__
__ge__
__getattribute__
__gt__
__hash__
__init__
__init_subclass__
__le__
__lt__
__module__
__ne__
__new__
__pCall
__reduce__
__reduce_ex__
__repr__
__setattr__
__sizeof__
__str__
__subclasshook__
_addSensor
_animateLEDs
_blockedLegReflex
_changeSupportMode
_clearObstacleMap
_computeTransform
_computeTransformAtTime
_disableTouchType
_dumpBlackBox
_dumpBlackBoxUntil
_enableAutoBalance
_enablePhysicalInteractionForChain
_enableTouchType
_enforceTabletReachability
_fall
_followPath
_followPathInWorld
_freeze
_gazeFrame
_get3DMap
_getActiveSensorList
_getBlindZones
_getCartesianUnfeasible
_getChainIsMoving
_getCollisionStateForObstacleSummary
_getComWorld
_getCumulatedDisplacement
_getCurrent
_getDangerousRegion
_getEnabledTouchTypes
_getFullLimits
_getGroundCollision
_getGroundCollisionForFallManager
_getGroundCollisionForForceContact
_getGroundPlaneTf
_getInertia
_getJointIsMoving
_getMapPtr
_getMapperNames
_getMotionConfig
_getMotionCycleNumber
_getMotionPosture
_getMotionPostureList
_getMotionToDCM
_getNormalForceContact
_getNumJointCommandDiscontinuities
_getObstacleData
_getObstacleMap
_getOccupancyGrid
_getPreviousMapPtr
_getRealTorsoInWorld
_getRemainingPath
_getRemainingTrajectory
_getRobotGroundConvexHull
_getRobotGroundConvexHullDebug
_getSensorData
_getSensorFrame
_getSensorList
_getSensorTransformAtTime
_getSummary
_getSupportPolygonBipedDebug
_getSupportPolygonCenter
_getTorque
_getTouchSensorFrame
_getTrajectoryCompletion
_getWorldRotation
_lookAt
_lookAtWb
_lookAtWbPersistent
_makeFrame
_makeFreeFrame
_mapFrame
_moveToPod
_moveToTracker
_moveTracker
_naoqiIsReadyCallback
_onPeopleDetected
_onTouchChanged
_publishFrame
_relax
_relaxMotorsWhenSitting
_removeSensor
_resetCartesianUnfeasible
_resetIdleDefaultAnimation
_resetIdleDefaultPosture
_resetNumJointCommandDiscontinuities
_rest
_restReflex
_robotAtBootFrame
_robotFrame
_saveWholeBodyDump
_setAnimationModeEnabled
_setCartesianUnfeasible
_setFollowPathSpeedFactor
_setIdleAnimation
_setIdlePosture
_setMotionPosture
_setPushRecoveryEnabled
_setStiffnesses
_shutdown
_stopChain
_stopLookAt
_stopMoveTracker
_stopPointAt
_trackerLookAt
_trackerPointAt
_trackerWithSpeed
_wbAxisMaskEffector
_wbDefaultConfiguration
_wbEnableEffectorConstraint
_wbEnableJointOptimization
_wbGetBalanceState
_wbGetEffectorConstraint
_wbGetEffectorOptimization
_wbGetFootState
_wbIsActive
_wbSetArticularLimitPreview
_wbSetEffectorStiffness
_wbSetEffectorWeight
_wbSetJointStiffness
_wbSetJointWeighting
angleInterpolation
angleInterpolationBezier
angleInterpolationWithSpeed
areNotificationsEnabled
areResourcesAvailable
async
call
changeAngles
changePosition
changeTransform
closeHand
exit
getAngles
getBodyNames
getBreathConfig
getBreathEnabled
getBrokerName
getCOM
getChainClosestObstaclePosition
getCollisionProtectionEnabled
getDiagnosisEffectEnabled
getExternalCollisionProtectionEnabled
getFallManagerEnabled
getFootGaitConfig
getFootSteps
getIdlePostureEnabled
getJointNames
getLimits
getMass
getMethodHelp
getMethodList
getModuleHelp
getMotionCycleTime
getMoveArmsEnabled
getMoveConfig
getNextRobotPosition
getOrthogonalSecurityDistance
getPosition
getPushRecoveryEnabled
getRobotConfig
getRobotPosition
getRobotVelocity
getSensorNames
getSmartStiffnessEnabled
getStiffnesses
getSummary
getSupportPolygon
getTangentialSecurityDistance
getTaskList
getTransform
getUsage
getWalkArmsEnabled
isRunning
isValid
killAll
killMove
killTask
killTasksUsingResources
killWalk
metaObject
move
moveAlong
moveInit
moveIsActive
moveTo
moveToward
openHand
pCall
ping
positionInterpolation
positionInterpolations
rest
robotIsWakeUp
setAngles
setBreathConfig
setBreathEnabled
setCollisionProtectionEnabled
setDiagnosisEffectEnabled
setEnableNotifications
setExternalCollisionProtectionEnabled
setFallManagerEnabled
setFootSteps
setFootStepsWithSpeed
setIdlePostureEnabled
setMotionConfig
setMoveArmsEnabled
setOrthogonalSecurityDistance
setPosition
setPositions
setPushRecoveryEnabled
setSmartStiffnessEnabled
setStiffnesses
setTangentialSecurityDistance
setTransform
setTransforms
setWalkArmsEnabled
setWalkTargetVelocity
stiffnessInterpolation
stop
stopMove
stopWalk
transformInterpolation
transformInterpolations
version
wait
waitUntilMoveIsFinished
waitUntilWalkIsFinished
wakeUp
walkInit
walkIsActive
walkTo
wbEnable
wbEnableBalanceConstraint
wbEnableEffectorControl
wbEnableEffectorOptimization
wbFootState
wbGoToBalance
wbGoToBalanceWithSpeed
wbSetEffectorControl
```

a-ichikura, 2024-07-09:
BasicAwarenessをsetEnabled(True)にしてみると、PepperのheadだけでなくBodyも動くようになってしまい、自分で作った動きを再生しなくなってしまいました。
そこで、HumanAwareness→Lookatならうまくいくかと思い、まずHuman Frameを取ってくる方向にしました。
Look at:
https://qisdk.softbankrobotics.com/sdk/doc/pepper-sdk/ch4_api/movement/reference/lookat.html?highlight=track
HumanAwareness:
https://qisdk.softbankrobotics.com/sdk/doc/pepper-sdk/ch4_api/perception/reference/human.html

ただ、上のdocumentを見ると、getHumanAroundみたいな関数を使えば良いと思うのですが、
pythonだと
```
['__bool__', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', 'async', 'call', 'engagedHuman', 'humansAround', 'isValid', 'makeEngageHuman', 'metaObject', 'recommendedHumanToEngage']
```
が関数のすべてで、該当する関数がないように思いました…というところで挫折しました。

k-okada, 2024-07-09:
おお、だいぶ進みましたね。

> https://qisdk.softbankrobotics.com/sdk/doc/pepper-sdk/ch4_api/movement/tuto/lookat_tutorial.html#lookat-tutorial
で、普通にLookAt関数自体は使えているでしょうか？

> ただ、上のdocumentを見ると、getHumanAroundみたいな関数を使えば良いと思うのですが、
これって、ドキュメントのJava側を見ているんだと思うけど、このKotlin とかいうやつを見ると
```
val humansAround: List<Human> = humanAwareness.humansAround
```
とあるので、`humansAround` でいいのでは？あと、lookatできれば最後の策としてはOpenCVで顔認識してlookatもありそうだけど、周りに人がいっぱいいると破綻するかな。

ちなみに、HumanAwareness->Lookat作戦は、https://qisdk.softbankrobotics.com/sdk/doc/pepper-sdk/ch4_api/perception/reference/engage.html?highlight=head_only#human-fully-engaged とか見たのかな。

> BasicAwarenessをsetEnabled(True)にしてみると、PepperのheadだけでなくBodyも動くようになってしまい、自分で作った動きを再生しなくなってしまいました

ちなみにこれはどうやって発見しました？https://qisdk.softbankrobotics.com/sdk/doc/pepper-sdk/ch4_api/abilities/reference/BasicAwareness.html?highlight=basicawareness# みても、

2.5だと、http://doc.aldebaran.com/2-5/naoqi/interaction/autonomousabilities/albasicawareness.html#tracking-modes のように、どこを動かすか選択できていそう。

a-ichikura, 2024-07-10:
色々途中経過なのですが‥
こちらのコメントを参考にして、
https://github.com/aldebaran/libqi-python/issues/26#issuecomment-2218316353

1. HumanAwarenessを使って、humanAroundで良さそうでした。↓
https://gitlab.jsk.imi.i.u-tokyo.ac.jp/ichikura/rosielab/-/blob/main/pepper/painting_motion/highlevel_soak.py?ref_type=heads#L35

2. なんとなくFrameを得る方法がわかった気がします。↓
https://gitlab.jsk.imi.i.u-tokyo.ac.jp/ichikura/rosielab/-/blob/main/pepper/painting_motion/highlevel_soak.py?ref_type=heads#L22

中身の値をPythonで直接見る方法までは行っていないのですが、このFrameをLookatに渡せばできるのでは？と思っています。

以下返信です。
>ちなみに、HumanAwareness->Lookat作戦は、https://qisdk.softbankrobotics.com/sdk/doc/pepper-sdk/ch4_api/perception/reference/engage.html?highlight=head_only#human-fully-engaged とか見たのかな。

そのrefを見てLookatにFrame渡せばいけるかもとおもいました。

>BasicAwarenessをsetEnabled(True)にしてみると、PepperのheadだけでなくBodyも動くようになってしまい、自分で作った動きを再生しなくなってしまいました

これはやって気がついたのと、
>Tracking detected human
Follows the engaged person with his head and body.

上のページにこの記述があったので体ごと動いてしまうんだと思っていましたが、他のコメントにもあるように調整はできそうですね。

Lookatはこれからやってみます。

a-ichikura, 2024-07-11:
```
#! /usr/bin/env python
# -*- coding: utf-8 -*-

import qi
import os
import sys 
import time
import json
import collections as cl
import ndjson
import random
import functools
import asyncio
import logging

class Authenticator:

    def __init__(self, username, password):
        self.username = username
        self.password = password

    # This method is expected by libqi and must return a dictionary containing
    # login information with the keys 'user' and 'token'.
    def initialAuthData(self):
        return {'user': self.username, 'token': self.password}

class AuthenticatorFactory:

    def __init__(self, username, password):
        self.username = username
        self.password = password

    # This method is expected by libqi and must return an object with at least
    # the `initialAuthData` method.
    def newAuthenticator(self):
        return Authenticator(self.username, self.password)

class Pepper:

    def __init__(self):
        #self.app = qi.Application(sys.argv, url="tcp://133.11.216.52:9559")
        #
        # naoqi 2.5 have following erros
        #
        # AttributeError: 'Object' object has no attribute 'setClientAuthenticatorFactory'
        version = sys.argv[1]
        ip = sys.argv[2]
        if version == "2.9":
            url = "tcps://" + ip + ":9503"
            self.app = qi.Application(sys.argv, url=url)
            logins = ("nao", "nao")
            factory = AuthenticatorFactory(*logins)
            self.app.session.setClientAuthenticatorFactory(factory)
            self.app.start()
            self.autonomous_life = self.app.session.service("ALAutonomousLife")
            self.motion_service = self.app.session.service("ALMotion")
            self.posture_service = self.app.session.service("ALRobotPosture")
            self.audio_service = self.app.session.service("ALAudioPlayer")
            self.led_service = self.app.session.service("ALLeds")
            self.memory_service = self.app.session.service("ALMemory")
            self.blinking_service = self.app.session.service("ALAutonomousBlinking")
            self.tts_service = self.app.session.service("ALTextToSpeech")
            self.human_awareness = self.app.session.service("HumanAwareness")
            self.basic_awareness = self.app.session.service("BasicAwareness")
            
        elif version == "2.5":
            url = "tcp://" + ip + ":9559"
            self.session = qi.Session()
            self.session.connect(url)
[... 80 lines cut ...]
  /home/ichikura/anaconda3/envs/pepper/lib/python3.8/inspect.py(2233): _signature_from_callable
  /home/ichikura/anaconda3/envs/pepper/lib/python3.8/inspect.py(2854): from_callable
  /home/ichikura/anaconda3/envs/pepper/lib/python3.8/inspect.py(3105): signature
  tracking_test.py(117): print_human
  tracking_test.py(130): <lambda>
```

の感じで出来なかったことをご報告します。

>2.5だと、http://doc.aldebaran.com/2-5/naoqi/interaction/autonomousabilities/albasicawareness.html#tracking-modes のように、どこを動かすか選択できていそう。

`BasicAwareness`の`setTrackingMode`は関数がないと言われてしまいました…
```
$ python tracking_test.py 2.9 10.0.0.4
started
[W] 1720741326.426357 13430 qi.path.sdklayout: No Application was created, trying to deduce paths
please enter your command:start
Life state is:disabled
Traceback (most recent call last):
  File "tracking_test.py", line 127, in <module>
    pepper.basic_awareness.setTrackingMode("Head")
AttributeError: 'qi.qi_python.Object' object has no attribute 'setTrackingMode'
```

今やろうとしてることではTrackingを使うのを諦めたのですが、今後のために出来るようにしたいと考えています。

k-okada, 2024-07-12:
ちなみに、
```
#これでframeの変化を求められそう
def print_human(pepper,humans):
    try:
        human = humans[0]
        print(dir(humans[0]))
        frame = human.headFrame.value()
        print(human.headFrame.value()) # これがエラーになるなら print(human.headFrame) 
        # pepper.motion_service._lookAt(frame)
    except IndexError as e:
        print("no human")
```
とすると何が返ってくるのかな？

あと、
```
s = [self.app.session.service("ALAutonomousLife"), self.app.session.service("BasicAwareness")]
for x in s:
    print(x)
    print(dir(x))
    print(x.getMethodList())
```
が知りたいです。

a-ichikura, 2024-07-12:
```
#これでframeの変化を求められそう
def print_human(pepper,humans):
    try:
        human = humans[0]
        print(dir(humans[0]))
        frame = human.headFrame.value()
        print(human.headFrame.value()) # これがエラーになるなら print(human.headFrame) 
        # pepper.motion_service._lookAt(frame)
    except IndexError as e:
        print("no human")
```
↓
```
:['__bool__', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', 'async', 'attention', 'call', 'emotion', 'engagementIntention', 'estimatedAge', 'estimatedGender', 'facePicture', 'facialExpressions', 'headFrame', 'isValid', 'metaObject']
<qi.qi_python.Object object at 0x7efde95189a0>
```
` print(human.headFrame.value())`ここの部分は`qi.qi_python.Object`のインスタンスとしか表示されなくて、qi_pythonの中身をみようとしたら`qi_python.so`ファイルで中身が見られなかったので諦めました‥

```
s = [self.app.session.service("ALAutonomousLife"), self.app.session.service("BasicAwareness")]
for x in s:
    print(x)
    print(dir(x))
    print(x.getMethodList())
```
↓
とりあえずこのままやろうとすると、
```
Traceback (most recent call last):
  File "tracking_test.py", line 127, in <module>
    pepper = Pepper()
  File "tracking_test.py", line 88, in __init__
    print(x.getMethodList())
AttributeError: 'qi.qi_python.Object' object has no attribute 'getMethodList'
```
ここで詰まるので、
getMethodList()の行を抜いて
```
line 1
<qi.qi_python.Object object at 0x7f8e251b6ae0>
['__bool__', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__pCall', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '_actionTransition', '_loadConfigFile', '_loadModule', '_preemptFocusForNonRobot', '_preemptFocusForRemote', '_reportToFileTimeOfSwitchFocus', '_setTimeSwitchFocusReporterEnabled', '_sleep', '_stopAll', '_stopFocus', '_suggestAction', '_switchFocus', '_wakeUp', 'activityTransition', 'async', 'call', 'exit', 'focusedActivity', 'getActivityNature', 'getActivityStatistics', 'getAutonomousAbilitiesStatus', 'getAutonomousAbilityEnabled', 'getAutonomousActivityStatistics', 'getBrokerName', 'getFocusContext', 'getFocusHistory', 'getLifeTime', 'getMethodHelp', 'getMethodList', 'getModuleHelp', 'getState', 'getStateHistory', 'getUsage', 'isMonitoringLaunchpadConditions', 'isRunning', 'isSafeguardEnabled', 'isValid', 'metaObject', 'pCall', 'ping', 'setAutonomousAbilityEnabled', 'setSafeguardEnabled', 'setState', 'startMonitoringLaunchpadConditions', 'stop', 'stopAll', 'stopFocus', 'stopMonitoringLaunchpadConditions', 'switchFocus', 'version', 'wait']
line 2
<qi.qi_python.Object object at 0x7f8e251bf5e0>
['__bool__', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '_getEmptyFrame', 'async', 'call', 'isEnabled', 'isRunning', 'isValid', 'metaObject', 'setEnabled', 'setStrategies', 'state', 'trackedHuman', 'zoneOfInterest']
```
こんな感じです。

a-ichikura, 2024-07-26:
情報更新できておらずすみません。
若干の余談なのですが、今回一台だけMacを使う必要があって、libqiをMacにインストールしたかった(`pip install qi`をしたかった)のですがうまくいかなかったので、Ubuntuの仮想マシンをMacの中に構築することによりことなきを得ました。

使ったのは[Virtual Box](https://ubuntu.com/tutorials/how-to-run-ubuntu-desktop-on-a-virtual-machine-using-virtualbox#1-overview)です。
MacのOSは
```
macOS Monterey Version 12.7.4
Processor 2.2 GHz Dual-Core Intel Core i7
Memory 8 GB 16000 MHz DDR3
Graphics Intel HD Graphics 6000 1536MB
```

UbuntuのVersionは24.04のisoを使うとうまくいかなかったので、22.04のisoイメージを使いました。
私のprogramを動かす分には特に問題なくできています。

kochigami, 2024-10-30:
NAO（V6）についても、
https://github.com/k-okada/pepper_robot/commit/ec038747474ddf8b592f8fd63e19de673dbe82f6
と同じようにnao_bringup/launch/nao_full.launchを修正し、
（https://github.com/kochigami/nao_robot/tree/support-user-password）

naoqi_driver: ros-naoqiのmasterブランチ
naoqi_bridge: k-okadaのmasterブランチ
(pepper_robot: k-okadaのmasterブランチ)

として、
`roslaunch nao_bringup nao_full.launch network_interface:=<> nao_port:=9503 password:=<>`
が動きました。
volumeのサービスなどは使えないですが（ros-naoqiのmasterブランチにマージされていないため）、nao-interface.lも使えます。

pepperでpepper-interface.lが使えるかをまだ試せていないので（大変時間が掛かっておりすみません・・・）、今週やろうと思います！

kochigami, 2024-10-31:
↑ pepperでも pepper-interface.l が使えることを確認いたしました。
次のステップとして、インストールの仕方をまとめたいと思います。