jsk-ros-pkg/jsk_robot #1916: [jsk_fetch_startup][smart_device_protocol] Suggest to suppress the error log

mqcmd196, 2024-06-14:
A large amount of logs `[ERROR] [WallTime: 1718326967.920978] [node:/smart_device_protocol] [func:SerialClient.__init__]: Error opening serial: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'` flows in `jsk_fetch_startup.log` . 

Please suppress the log using something like log_once in the node.

sktometometo, 2024-06-14:
M5AtomS3 is detached. So it has been connected again.

Let's check if it works. Please add additinal error if exists.

mqcmd196, 2024-06-14:
Thanks, confirmed

Yoshiki Obinata (大日方 慶樹)
JSK Robotics Lab, The University of Tokyo
________________________________
差出人: Koki Shinjo ***@***.***>
送信日時: 2024年6月14日 12:10
宛先: jsk-ros-pkg/jsk_robot ***@***.***>
CC: Yoshiki Obinata ***@***.***>; Author ***@***.***>
件名: Re: [jsk-ros-pkg/jsk_robot] [jsk_fetch_startup][smart_device_protocol] Suggest to suppress the error log (Issue #1916)

M5AtomS3 is detached. So it has been connected again.

Let's check if it works. Please add additinal error if exists.