jsk-ros-pkg/jsk_robot #1920: [PR2] PR1040 does not publish /base_scan after a while after startup

Kanazawanaoaki, 2024-08-13:
最近，PR1040を使おうと思うと，ほぼ毎回'/base_scan'のトピックが出ておらず，move_baseなどが利用できないため再起動してから利用するというようになってしまっています．  

[pythonのプログラム](https://gist.github.com/Kanazawanaoaki/d4b0bb8f0002b482a1459eaecb27b10f#file-base_scan_checker-py ) で数日様子をみていましたが，以下のように3~12時間くらいの経過で/base_scanがpublishされなくなっています．  

#### 2024/08/08の15時くらいに起動した．
```
$ python base_scan_checker.py 
[INFO] [1723113616.636096]: base scan check start
[INFO] [1723137411.033373]: I haven't seen the base scan topic for 5.0 seconds.
```
Fri Aug 09 2024 02:16:51 GMT+0900 (日本標準時)
11時間くらい

#### 20240809の19時半くらいに再起動した．
```
$ python base_scan_checker.py 
[INFO] [1723199876.899508]: base scan check start
[INFO] [1723214917.328692]: I haven't seen the base scan topic for 5.0 seconds.
```
Fri Aug 09 2024 23:48:37 GMT+0900 (日本標準時)
4時間くらい

#### 20240810の15時半くらいに再起動した
```
$ python base_scan_checker.py 
[INFO] [1723283233.504669]: base scan check start
[INFO] [1723283642.839506]: I haven't seen the base scan topic for 5.0 seconds.
```
Sat Aug 10 2024 18:54:02 GMT+0900 (日本標準時)
3時間半くらい

#### 2024/08/10の19時半くらいに再起動した
```
$ python base_scan_checker.py 
[INFO] [1723285426.805099]: base scan check start
[INFO] [1723313578.201548]: I haven't seen the base scan topic for 5.0 seconds.
```
Sun Aug 11 2024 03:12:58 GMT+0900 (日本標準時)
8時間くらい

### 原因究明
[jsk-pr2-startup](https://gist.github.com/Kanazawanaoaki/d4b0bb8f0002b482a1459eaecb27b10f#file-jsk-pr2-startup-20240809-log)と[robot.log](https://gist.github.com/Kanazawanaoaki/d4b0bb8f0002b482a1459eaecb27b10f#file-robot-20240809-log ) のpublishさらなくなったAug  9 02:16あたりを見てみても原因はよくわかりませんでした．
```
Aug  9 02:16:47 pr1040 jsk-pr2-startup[4899]: #033[33m[ WARN] [1723137312.574944898]: wait-for-service /message_store/query_messages timeout.#033[0m
```
となっているのが気になりましたが，台車背面物理ブレーカから再起動し`/base_scan`が出ている状態でも`/message_store/query_messages`のrosserviceは存在しませんでした．

### 解決策
https://github.com/jsk-ros-pkg/jsk_robot/issues/1406#issuecomment-966466652 のrqt_pr2_dashboardからのブレーカの再起動．
```
sudo pr2-shutdown --reboot
```
での再起動では復活しませんでした．
物理的に台車背面のブレーカから再起動をすると復活します．

iory, 2024-08-13:
base_scanはUSBで接続されていると思うんだけど、別のhokuyoを外において1日置いてみるとどうなるかな。

Kanazawanaoaki, 2024-08-14:
ありがとうございます．今日後で試してみようと思います．

https://github.com/jsk-ros-pkg/jsk_robot/blob/develop/pr2-noetic/jsk_pr2_robot/jsk_pr2_startup/pr2_bringup.launch#L79-L97
を見るとtiltもhokuyoでc2にささっていてほぼ同じなのだと思うのですが，今もtopicがちゃんと出ているようなのでbaseの方のhokuyoのハードウェア的な問題でなにかを取り替えると解決するのかもですね．

Kanazawanaoaki, 2024-08-18:
単体テストを行うと
```
[ERROR] [1723953698.565381456] [/base_hokuyo_node:ros.hokuyo_node]: Laser returned abnormal status message, aborting: Trouble!! Error No. =[00->33] You may be able to find further information at http://www.ros.org/wiki/hokuyo_node/Troubleshooting/
```
[0->33]
There is interference from another IR source (often another Hokuyo). Unplugging and replugging the power seems to fix the problem.
のエラーでした．

PR2で使っているHokuyoの[UTM-30LX](https://jp.robotshop.com/products/hokuyo-utm-30lx-scanning-laser-rangefinder )はUSBケーブルと電源ケーブルの2本があり，電源ケーブルの方の抜き差しが必要なので，PR2本体のブレーカから再起動が必要になるようです．

エラーコードは異なりますが，エラーの説明の内容が似ていたので，その[ros wiki](https://wiki.ros.org/hokuyo_node/Troubleshooting/#A.5B0-.3E75.5D_Code)に添付されているプログラムをダウンロードして以下のようにbase_hokuyoのリセットをしてみたところブレーカの再起動無しで`/base_scan`が出るようになりました．
```
$ ./upgrade /etc/ros/sensors/base_hokuyo reset.cmd
```

これで`/base_scan`が出なくなる問題が解決すれば良いのですが，今後も出なくなる問題が頻繁に発生する場合は，その対処法としてこのhokuyoのリセットが良いかどうかはまた考える必要がありそうです．

Kanazawanaoaki, 2024-08-19:
様子を見ていましたが先程
```bash
[INFO] [1724038811.365504]: I haven't seen the base scan topic for 5.0 seconds.
```
となったので確認すると`/base_scan`が出なくなっていました．タイムスタンプを確認すると，だいたい
Sun Aug 18 2024 13:44:04 GMT+0900 (日本標準時)にスタートで，
Mon Aug 19 2024 12:40:11 GMT+0900 (日本標準時)にでなくなった．
ので約23時間程で出なくなったようです．
試しでそのままの状態で
pr1040s:~/Downloads/hokuyo_debug$ ./upgrade /etc/ros/sensors/base_hokuyo reset.cmd
をすると，`/base_scan`が出るようにはなりました．
（[ros wiki](https://wiki.ros.org/hokuyo_node/Troubleshooting/#A.5B0-.3E75.5D_Code) にはBefore running this command, shut down the hokuyo_node. Run: と書いてあるので本当はnodeを切ってから単体のlaunchを立ち上げるのが正しかったです．）

hokuyoのresetコマンドを頻繁に送るのが正しいかどうか次第で，
https://github.com/jsk-ros-pkg/jsk_robot/blob/develop/pr2-noetic/jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_sensors/check_openni_node.py#L58
を参考にリスタートするノードを書くこと自体はできそうです．

Kanazawanaoaki, 2024-08-20:
先程も/base_scanが出ていなかったので
kinectの[check_openni_node.py](https://github.com/jsk-ros-pkg/jsk_robot/blob/develop/pr2-noetic/jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_sensors/check_openni_node.py#L58 )を参考に書いた[check_base_scan_node-py](https://gist.github.com/Kanazawanaoaki/d4b0bb8f0002b482a1459eaecb27b10f#file-check_base_scan_node-py )をc2のhokuyo resetプログラムが置いてあるところで実行したところ/base_scanで復活することは確認できました．
その時の[log](https://gist.github.com/Kanazawanaoaki/d4b0bb8f0002b482a1459eaecb27b10f#file-20240820_log_check_base_scan_node-log )

mqcmd196, 2024-08-20:
hokuyoのハードウェア側の問題であれば，定期resetプログラムを走らせておくのが丸いですかね

Kanazawanaoaki, 2024-08-21:
> hokuyoのハードウェア側の問題であれば，定期resetプログラムを走らせておくのが丸いですかね

まあそうかなと思ったので https://github.com/jsk-ros-pkg/jsk_robot/pull/1922 のPRを作ってみてPR1040の体内で走らせてみて様子を見ています．
ログは，c2の中で
```
$ cd ~/.ros/log/latest
$ ls | grep hokuyo
$  tail -f base_hokuyo_scan_checker-4.log
```
とかで見られます．

closed 2024-12-10 by Kanazawanaoaki