jsk-ros-pkg/jsk_robot #1921: [Fetch1075] L515 in Fetch stops publishing depth after a while after startup

mqcmd196, 2024-08-19:
# 症状
- L515のdepth topicを別のPCからsubscribeしていると，depthのtopicが出なくなる．このときcolorは出ている

mqcmd196, 2024-08-19:
自分の使っているデスクトップ (Intel(R) Xeon(R) Gold 6336Y CPU @ 2.40GHz, RAM:128GB)で
```
roslaunch realsense2_camera rs_camera.launch align_depth:=true
```
を起動し，12時間放置した．launchした時，12時間経過したときでdepthを `rqt_image_view` で表示してみたが，問題なくdepthが表示されていることを確認

mqcmd196, 2024-08-19:
FetchのPCでは直ちに落ちるということは発生しないが, compressedDepthのHzがゆるやかに落ちていくという現象が発生．検証する

mqcmd196, 2024-08-19:
`/l515_head/aligned_depth_to_color/image_raw` を体内でsubscribeする分にはHz落ちの現象は再現しない

mqcmd196, 2024-08-19:
ちなみにzdepthだとHz落ちしない．正解はこちらか？

pazeshun, 2024-08-19:
以下のようなものが必要ってことかな。
https://github.com/jsk-ros-pkg/jsk_robot/pull/1493

mqcmd196, 2024-08-20:
@pazeshun 情報ありがとうございます．png_levelの適切な値の設定が必要ということでしょうか．depthについては `zdepth`
で解決していますが，rgbについてはとりあえず動いているものの，調整が必要かもしれません．

pazeshun, 2024-08-20:
depthに関して、zdepthを使わずにpng_levelを調整するのもありなんじゃないかと思った、ということでした。
zdepthは不可逆圧縮で、元のデータには戻せなかった気がするので、後で困る可能性あるかなと思いました。
ただ、png_level下げるとデータ容量が増えてネットワークごしの通信がつらくなる/rosbagが重くなるかとは思うので、その辺が許容できるかどうか次第でしょうか。

最終的にzdepth使うにしても、僕だったら、デフォルトのcompressedDepthはpng_levelをチューンしたpng圧縮にしておいて、別ノードでzdepth使うものを立ち上げるような気がしました。

mqcmd196, 2024-08-21:
確かにそうですね．hzが落ちない程度のチューニングはしておこうと思います

(余談ですが，金沢君に一昨日jpegと勘違いしてpngは不可逆圧縮って言ってしまった気がする...)