jsk-ros-pkg/jsk_robot #1923: [BelKa] Python3.8 on melodic cannot import PyKDL (which is used in tf2_ros) due to `No module named 'sip'`

sktometometo, 2024-09-12:
I want to use Python3.8 on BelKa (which runs ubuntu 18.04 on its onboard computer) in order to use OpenAI API.
So I have tried to use it like below.

- Install `python3.8-*` packages with `sudo apt install`
- Install `catkin_virtualenv` package
- Make `venv` with Python3.8 and `catkin_install_python` for the script.

[This package](https://github.com/sktometometo/test_python3.8_on_melodic) is a minimum version of it.

But I got an error message while executing it like below.

```bash
spot@BelKa:~/test_ws$ catkin config
-------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/melodic
Workspace:                   /home/spot/test_ws
-------------------------------------------------------
Build Space:       [missing] /home/spot/test_ws/build
Devel Space:       [missing] /home/spot/test_ws/devel
Install Space:      [unused] /home/spot/test_ws/install
Log Space:         [missing] /home/spot/test_ws/logs
Source Space:       [exists] /home/spot/test_ws/src
DESTDIR:            [unused] None
-------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------
spot@BelKa:~/test_ws$ vcs export
repositories:
  src/test_python3.8_on_melodic:
    type: git
    url: git@github_with_sktometometo:sktometometo/test_python3.8_on_melodic.git
    version: master
spot@BelKa:~/test_ws$ vcs status
=== ./src/test_python3.8_on_melodic (git) ===
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
spot@BelKa:~/test_ws$ catkin build
-------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/melodic
Workspace:                   /home/spot/test_ws
-------------------------------------------------------
Build Space:        [exists] /home/spot/test_ws/build
Devel Space:        [exists] /home/spot/test_ws/devel
Install Space:      [unused] /home/spot/test_ws/install
Log Space:         [missing] /home/spot/test_ws/logs
Source Space:       [exists] /home/spot/test_ws/src
DESTDIR:            [unused] None
-------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-------------------------------------------------------
[build] Found '1' packages in 0.0 seconds.                                                                                                                  
[build] Updating package table.                                                                                                                             
Starting  >>> catkin_tools_prebuild                                                                                                                         
Finished  <<< catkin_tools_prebuild                   [ 1.5 seconds ]                                                                                       
Starting  >>> test_python38_on_melodic                                                                                                                      
Finished  <<< test_python38_on_melodic                [ 8.9 seconds ]                                                                                       
[build] Summary: All 2 packages succeeded!                                                                                                                  
[build]   Ignored:   None.                                                                                                                                  
[build]   Warnings:  None.                                                                                                                                  
[build]   Abandoned: None.                                                                                                                                  
[build]   Failed:    None.                                                                                                                                  
[build] Runtime: 10.3 seconds total.                                                                                                                        
[build] Note: Workspace packages have changed, please re-source setup files to use them.
spot@BelKa:~/test_ws$ srou^C
spot@BelKa:~/test_ws$ source devel/setup.bash 
spot@BelKa:~/test_ws$ rosrun test_python38_on_melodic hoge.py 
original PYTHONPATH:['/home/spot/test_ws/src/test_python3.8_on_melodic/scripts', '/opt/ros/melodic/lib/python2.7/dist-packages', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/spot/test_ws/devel/.private/test_python38_on_melodic/share/test_python38_on_melodic/venv/lib/python3.8/site-packages', '/usr/local/lib/python3.8/dist-packages', '/usr/lib/python3/dist-packages']
modified PYTHONPATH:['/home/spot/test_ws/src/test_python3.8_on_melodic/scripts', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/spot/test_ws/devel/.private/test_python38_on_melodic/share/test_python38_on_melodic/venv/lib/python3.8/site-packages', '/usr/local/lib/python3.8/dist-packages', '/usr/lib/python3/dist-packages', '/opt/ros/melodic/lib/python3/dist-packages']
Traceback (most recent call last):
  File "/home/spot/test_ws/src/test_python3.8_on_melodic/scripts/hoge.py", line 10, in <module>
    import PyKDL
ModuleNotFoundError: No module named 'sip'
```

This seems because [Python-sip](https://github.com/Python-SIP/sip) installed into system is for Python3.6 (which is original python3 version of ubuntu 18.04) and it cannot be imported from Python3.8.

```bash
spot@BelKa:~$ python3.8
Python 3.8.0 (default, Dec  9 2021, 17:53:27) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sip
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'sip'
>>> exit
Use exit() or Ctrl-D (i.e. EOF) to exit
>>> exit()
spot@BelKa:~$ python3.6
Python 3.6.9 (default, Mar 10 2023, 16:46:00) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sip
>>> sip.__file__
'/usr/lib/python3/dist-packages/[sip.cpython-36m-x86_64-linux-gnu.so](http://sip.cpython-36m-x86_64-linux-gnu.so/)'
```

sktometometo, 2024-09-12:
@iory have found a workaround for this issue.

```bash
sudo apt update
sudo apt install ros-melodic-cv-bridge-python3 python3.8 python3-yaml ros-melodic-jsk-tf2-py-python3 ros-melodic-jsk-python-orocos-kdl-python3 python-catkin-tools vim python3.8-venv ros-melodic-catkin-virtualenv less mlocate python3-sip-dev python3-sip-dev libpython3.8-dev git flex bison
git clone https://github.com/Python-SIP/sip -b 4.19.14
cd sip
python3.8 build.py prepare
python3.8 configure.py
make
sudo make install

cd ~/
mkdir -p ~/ros/test_python3.8_on_melodic/src
cd ~/ros/test_python3.8_on_melodic/src
git clone https://github.com/sktometometo/test_python3.8_on_melodic

cd ~/ros/test_python3.8_on_melodic
rosdep update
rosdep install --from-paths -i -y -r .
catkin build
source devel/setup.bash
```

After this, `hoge.py` can be run with changes below.

```python
import os
import sys

# cv_bridge_python3 import
if os.environ['ROS_PYTHON_VERSION'] == '3':
    from cv_bridge import CvBridge
    import PyKDL
else:
    ws_python3_paths = [p for p in sys.path if 'devel/lib/python3' in p]
    if len(ws_python3_paths) == 0:
        # search cv_bridge in workspace and append
        ws_python2_paths = [
            p for p in sys.path if 'devel/lib/python2.7' in p]
        for ws_python2_path in ws_python2_paths:
            ws_python3_path = ws_python2_path.replace('python2.7', 'python3')
            if os.path.exists(os.path.join(ws_python3_path, 'cv_bridge')):
                ws_python3_paths.append(ws_python3_path)
        if len(ws_python3_paths) == 0:
            opt_python3_path = '/opt/ros/{}/lib/python3/dist-packages'.format(
                os.getenv('ROS_DISTRO'))
            sys.path = [opt_python3_path] + sys.path
            from cv_bridge import CvBridge
            import PyKDL
            sys.path.remove(opt_python3_path)
        else:
            sys.path = [ws_python3_paths[0]] + sys.path
            from cv_bridge import CvBridge
            import PyKDL
            sys.path.remove(ws_python3_paths[0])
    else:
        from cv_bridge import CvBridge
        import PyKDL
print(CvBridge)
print(PyKDL)
```

mqcmd196, 2024-09-12:
How aboutUSE_SYSTEM_PACKAGES FALSE ?

    url: ***@***.***_with_sktometometo:sktometometo/test_python3.8_on_melodic.git
    version: master

>>> import sip
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'sip'
>>> exit
Use exit() or Ctrl-D (i.e. EOF) to exit
>>> exit()
***@***.***:~$ python3.6
Python 3.6.9 (default, Mar 10 2023, 16:46:00) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sip
>>> sip.__file__
'/usr/lib/python3/dist-packages/[sip.cpython-36m-x86_64-linux-gnu.so](http://sip.cpython-36m-x86_64-linux-gnu.so/)'

sktometometo, 2024-09-12:
@iory Thank you for cooperation!

sktometometo, 2024-09-12:
@mqcmd196 

Same error happened with `USE_SYSTEM_PACKAGE FALSE`

```bash
spot@BelKa:~/test_ws/src/test_python3.8_on_melodic$ catkin clean

[clean] Warning: This will completely remove the following directories. (Use `--yes` to skip this check)
[clean] Log Space:     /home/spot/test_ws/logs
[clean] Build Space:   /home/spot/test_ws/build
[clean] Devel Space:   /home/spot/test_ws/devel

[clean] Are you sure you want to completely remove the directories listed above? [yN]: y
[clean] Removing develspace: /home/spot/test_ws/devel
[clean] Removing buildspace: /home/spot/test_ws/build
[clean] Removing log space: /home/spot/test_ws/logs
spot@BelKa:~/test_ws/src/test_python3.8_on_melodic$ catkin ^C
spot@BelKa:~/test_ws/src/test_python3.8_on_melodic$ cd ../..
spot@BelKa:~/test_ws$ catkin build
-------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/melodic
Workspace:                   /home/spot/test_ws
-------------------------------------------------------
Build Space:        [exists] /home/spot/test_ws/build
Devel Space:        [exists] /home/spot/test_ws/devel
Install Space:      [unused] /home/spot/test_ws/install
Log Space:         [missing] /home/spot/test_ws/logs
Source Space:       [exists] /home/spot/test_ws/src
DESTDIR:            [unused] None
-------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-------------------------------------------------------
[build] Found '1' packages in 0.0 seconds.
[build] Updating package table.
Starting  >>> catkin_tools_prebuild                  
Finished  <<< catkin_tools_prebuild                   [ 2.0 seconds ]
Starting  >>> test_python38_on_melodic               
Finished  <<< test_python38_on_melodic                [ 9.4 seconds ]
[build] Summary: All 2 packages succeeded!
[build]   Ignored:   None.
[build]   Warnings:  None.
[build]   Abandoned: None.
[build]   Failed:    None.
[build] Runtime: 11.4 seconds total.
[build] Note: Workspace packages have changed, please re-source setup files to use them.
(reverse-i-search)`catkin ': ^Ctkin build
spot@BelKa:~/test_ws$ ls
build  devel  logs  src
spot@BelKa:~/test_ws$ source devel/setup.bash
spot@BelKa:~/test_ws$ rosrun jsk_spot_autonomous_integration_demo demo.py -h
[rospack] Error: package 'jsk_spot_autonomous_integration_demo' not found
spot@BelKa:~/test_ws$ ^C
spot@BelKa:~/test_ws$ rosrun test_python38_on_melodic hoge.py
original PYTHONPATH:['/home/spot/test_ws/src/test_python3.8_on_melodic/scripts', '/opt/ros/melodic/lib/python2.7/dist-packages', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/spot/test_ws/devel/.private/test_python38_on_melodic/share/test_python38_on_melodic/venv/lib/python3.8/site-packages']
modified PYTHONPATH:['/home/spot/test_ws/src/test_python3.8_on_melodic/scripts', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/spot/test_ws/devel/.private/test_python38_on_melodic/share/test_python38_on_melodic/venv/lib/python3.8/site-packages', '/opt/ros/melodic/lib/python3/dist-packages']
Traceback (most recent call last):
  File "/home/spot/test_ws/src/test_python3.8_on_melodic/scripts/hoge.py", line 10, in <module>
    import PyKDL
ModuleNotFoundError: No module named 'sip'
spot@BelKa:~/test_ws$ vcs diff
=== ./src/test_python3.8_on_melodic (git) ===
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 14c6261..ec89a83 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -6,7 +6,7 @@ find_package(catkin REQUIRED COMPONENTS catkin_virtualenv)
 catkin_generate_virtualenv(
     PYTHON_INTERPRETER python3.8
     CHECK_VENV FALSE
-    USE_SYSTEM_PACKAGES TRUE
+    USE_SYSTEM_PACKAGES FALSE
 )

 catkin_package()
spot@BelKa:~/test_ws$ 
```

closed 2024-09-12 by sktometometo