jsk-ros-pkg/jsk_robot #1939: sound_playノードで日本語を話さない

sawada10, 2025-04-18:
@mqcmd196 @iory @k-okada 
(cc. @ayaha-n )

sound_playノードを使って、日本語を話そうとしているのですが、話すことができませんでした。（英語は話せました）
環境は Ubuntu20.04, noeticです。
`rosversion sound_play`の結果は`0.3.17`でした。
対処方法や確認すべきことをご存知でしたら教えていただけると嬉しいです。

**実行手順**
```
roscore
rosrun sound_play soundplay_node.py
cd path/to/script
python3 sample_say.py
```

sample_say.pyの中身は以下のとおりです
```
import rospy
from sound_play.msg import SoundRequest
from sound_play.libsoundplay import SoundClient

rospy.init_node('talker_node')
sound_client = SoundClient(sound_action="robotsound_jp")

rospy.sleep(1)

text = "こんにちは"
sound_client.say(text, voice='ja')
print("I said こんにちは")

text = "hello"
sound_client.say(text)
print("I said hello")

rospy.spin()
```

**実行結果**
「こんにちは」を飛ばして、「hello」はしゃべることは確認できた
sound_playノードの実行画面では以下のエラーがでている
```
$ rosrun sound_play soundplay_node.py 
/opt/ros/noetic/lib/sound_play/soundplay_node.py:330: PyGIDeprecationWarning: Since version 3.11, calling threads_init is no longer needed. See: https://wiki.gnome.org/PyGObject/Threading
  GObject.threads_init()
/opt/ros/noetic/lib/sound_play/soundplay_node.py:331: PyGIDeprecationWarning: GObject.MainLoop is deprecated; use GLib.MainLoop instead
  self.g_loop = threading.Thread(target=GObject.MainLoop().run)
[INFO] [1744966619.588938]: Loading from plugin definitions
[INFO] [1744966620.094725]: sound_play node is ready to play sound
SIOD ERROR: unbound variable : ja
[ERROR] [1744966625.140251]: Sound synthesis failed.Is festival installed?Is a festival voice installed?Try running "rosdep satisfy sound_play|sh".Refer to http://wiki.ros.org/sound_play/Troubleshooting
[ERROR] [1744966625.148869]: Failed to generate wavfile.
[ERROR] [1744966625.150012]: Exception in callback: 'こんにちは---ja'
[INFO] [1744966625.151224]: Traceback (most recent call last):
  File "/opt/ros/noetic/lib/sound_play/soundplay_node.py", line 192, in callback
    sound = self.select_sound(data)
  File "/opt/ros/noetic/lib/sound_play/soundplay_node.py", line 162, in select_sound
    sound = self.voicesounds[voice_key]
KeyError: 'こんにちは---ja'
```

iory, 2025-04-18:
日本語を喋るvoicevoxのようなプログラムは立ち上がっているかな？
sound_playデフォルトのものだと日本語は話さないようになっているので別途roslaunchを上げる必要があるんだけど、他に立ち上げているroslaunchがあれば教えてください。

sawada10, 2025-04-18:
すみません、voicevoxなしで日本語でも動くかと勘違いしていました。
ありがとうございます。

[こちら](https://github.com/iory/jsk_3rdparty/tree/voicevox)の@ioryのvoicevoxブランチをcatkin buildしたのですが、以下のようになりました。
`rosdep update; rosdep install --ignore-src --from-paths . -y -r`は実行しています。
これはcmakeのバージョンを上げるとよいのでしょうか？

```
$ catkin build voicevox 
------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/noetic
Workspace:                   /home/leus/kashiwagi_ws
------------------------------------------------------------
Build Space:        [exists] /home/leus/kashiwagi_ws/build
Devel Space:        [exists] /home/leus/kashiwagi_ws/devel
Install Space:      [unused] /home/leus/kashiwagi_ws/install
Log Space:          [exists] /home/leus/kashiwagi_ws/logs
Source Space:       [exists] /home/leus/kashiwagi_ws/src
DESTDIR:            [unused] None
------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------
[build] Found 62 packages in 0.0 seconds.                                                                                                                                                                         
[build] Package table is up to date.                                                                                                                                                                              
Starting  >>> voicevox                                                                                                                                                                                            
__________________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << voicevox:make /home/leus/kashiwagi_ws/logs/voicevox/build.make.003.log                                                                                                                              
make[3]: 警告: jobserver が利用不可: 今回は -j1 を使います. 親 make ルールに `+' を追加しましょう.
make[3]: 警告: jobserver が利用不可: 今回は -j1 を使います. 親 make ルールに `+' を追加しましょう.
make[3]: 警告: jobserver が利用不可: 今回は -j1 を使います. 親 make ルールに `+' を追加しましょう.
make[3]: 警告: jobserver が利用不可: 今回は -j1 を使います. 親 make ルールに `+' を追加しましょう.
    ERROR: Command errored out with exit status 1:
     command: /home/leus/kashiwagi_ws/build/voicevox/venv/bin/python /home/leus/kashiwagi_ws/build/voicevox/venv/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmph3641j55
         cwd: /tmp/pip-resolver-_iejkgpo/pyopenjtalk
    Complete output (28 lines):
    CMake Error at CMakeLists.txt:5 (cmake_minimum_required):
      Compatibility with CMake < 3.5 has been removed from CMake.
    
      Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
      to tell CMake that the project requires at least <min> but has been updated
      to work with policies introduced by <max> or earlier.
    
      Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.
    
    
    -- Configuring incomplete, errors occurred!
    Traceback (most recent call last):
      File "/home/leus/kashiwagi_ws/build/voicevox/venv/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py", line 280, in <module>
        main()
      File "/home/leus/kashiwagi_ws/build/voicevox/venv/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py", line 263, in main
        json_out['return_val'] = hook(**hook_input['kwargs'])
      File "/home/leus/kashiwagi_ws/build/voicevox/venv/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py", line 114, in get_requires_for_build_wheel
        return hook(config_settings)
      File "/tmp/pip-build-env-dzil2x8s/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 162, in get_requires_for_build_wheel
        return self._get_build_requires(
      File "/tmp/pip-build-env-dzil2x8s/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 143, in _get_build_requires
        self.run_setup()
[... 53 lines cut ...]
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/catkin_virtualenv/venv_lock", line 47, in <module>
    venv.lock(
  File "/opt/ros/noetic/lib/python3/dist-packages/catkin_virtualenv/venv.py", line 143, in lock
    run_command(command, check=True)
  File "/opt/ros/noetic/lib/python3/dist-packages/catkin_virtualenv/__init__.py", line 44, in run_command
    return subprocess.run(cmd, *args, **kwargs)
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/home/leus/kashiwagi_ws/build/voicevox/venv/bin/pip-compile', '--no-header', 'requirements_noetic.txt', '--pip-args', '-qq --retries 10 --timeout 30', '-o', '/home/leus/kashiwagi_ws/src/jsk_3rdparty/3rdparty/voicevox/requirements.txt']' return [... cut ...]
[build]   Abandoned: None.                                                                                                                                                                                        
[build]   Failed:    1 packages failed.                                                                                                                                                                           
[build] Runtime: 20.6 seconds total.                                                  
```

knorth55, 2025-04-18:
You can try japanese speech synthesis with aques_talk.
Inside the launch, `soundplay_node.py` is launching with Japanese speech synthesis configuration.

https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/master/3rdparty/aques_talk/launch/aques_talk.launch

```
roslaunch aques_talk aques_talk.launch
```

mqcmd196, 2025-04-19:
> sound_playノードを使って、日本語を話そうとしているのですが、話すことができませんでした。（英語は話せました）
> 環境は Ubuntu20.04, noeticです。

at least I know there are 2 options as you do and @knorth55 says, using aques_talk or voicevox.

> これはcmakeのバージョンを上げるとよいのでしょうか？

No.  CMake 4.0 dropped old CMake support and I think there is `cmake_minimum_required( <some version less than 3.5> )` line in some packages. I don't know where it is and when CMake=4.0 is used. Anyway, the upstream package maintainers noticed this issue and fixed it via 

- https://github.com/VOICEVOX/voicevox_engine/issues/1568
- https://github.com/VOICEVOX/pyopenjtalk/pull/22

mqcmd196, 2025-04-19:
Please try this 
- https://github.com/iory/jsk_3rdparty/pull/11

sawada10, 2025-04-19:
Thank you for your response.

I tried using voicevox in [this pull request](https://github.com/iory/jsk_3rdparty/pull/11), and confirmed that it builds successfully and works.

aques_talk also worked.
(Additional note: When I first tried aques_talk, I got the following error, but referring to [this page](https://stackoverflow.com/questions/66554561/modulenotfounderror-no-module-named-rospkg), I changed the shebang in `jsk_3rdparty/3rdparty/aques_talk/text2wave` from `#!/usr/bin/env python` to `#!/usr/bin/env python3`, and it started working.

```
$ roslaunch aques_talk aques_talk.launch 
... logging to /home/leus/.ros/log/150efb30-1d01-11f0-bcf4-4fc2871093a1/roslaunch-leus-ThinkPad-T480s-259492.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://133.11.216.103:41493/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    sound_play_jp (sound_play/soundplay_node.py)

auto-starting new master
process[master]: started with pid [259501]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 150efb30-1d01-11f0-bcf4-4fc2871093a1
process[rosout-1]: started with pid [259511]
started core service [/rosout]
process[sound_play_jp-2]: started with pid [259514]
/opt/ros/noetic/lib/sound_play/soundplay_node.py:330: PyGIDeprecationWarning: Since version 3.11, calling threads_init is no longer needed. See: https://wiki.gnome.org/PyGObject/Threading
  GObject.threads_init()
/opt/ros/noetic/lib/sound_play/soundplay_node.py:331: PyGIDeprecationWarning: GObject.MainLoop is deprecated; use GLib.MainLoop instead
  self.g_loop = threading.Thread(target=GObject.MainLoop().run)
Traceback (most recent call last):
  File "/home/leus/kashiwagi_ws/src/jsk_3rdparty/3rdparty/aques_talk//text2wave", line 6, in <module>
    import rospkg
ImportError: No module named rospkg
[ERROR] [1745055104.743333]: Sound synthesis failed.Is festival installed?Is a festival voice installed?Try running "rosdep satisfy sound_play|sh".Refer to http://wiki.ros.org/sound_play/Troubleshooting
[ERROR] [1745055104.744729]: Failed to generate wavfile.
[ERROR] [1745055104.745909]: Exception in actionlib callback: 'こんにちは---ja'
Traceback (most recent call last):
  File "/home/leus/kashiwagi_ws/src/jsk_3rdparty/3rdparty/aques_talk//text2wave", line 6, in <module>
    import rospkg
ImportError: No module named rospkg
[ERROR] [1745055104.899506]: Sound synthesis failed.Is festival installed?Is a festival voice installed?Try running "rosdep satisfy sound_play|sh".Refer to http://wiki.ros.org/sound_play/Troubleshooting
[ERROR] [1745055104.901235]: Failed to generate wavfile.
[ERROR] [1745055104.902704]: Exception in actionlib callback: 'hello---'
```

closed 2025-04-19 by sawada10

mqcmd196, 2025-04-19:
@sawada10 

Nice catch. Please don't hesitate to send a Pull Request if you fix something. 

At this time, the correct way is
  
- https://github.com/jsk-ros-pkg/jsk_3rdparty/pull/531

sawada10, 2025-04-19:
@mqcmd196 

Thank you for your review.
Okay, I will make Pull Request next time.