jsk-ros-pkg/jsk_robot #1943: PR2のグリッパや手首のみを再キャリブレーションしたい

Kanazawanaoaki, 2025-08-26:
PR2では，タスクに合わせてグリッパを付け替えることや，最近キャリブレーションで手首が90°ずれてしまうことがあるので，その時に全身の再キャリブレーションではなくて一部のみの再キャリブレーションができると嬉しい．

PR2の再キャリブレーション時に実行している以下のコマンドでは
```bash
roslaunch pr2_bringup pr2_recalibrate.launch
```
https://github.com/TAMS-Group/pr2_robot/blob/pr-various/pr2_bringup/pr2_recalibrate.launch のlaunchファイルが実行されていて，実行されいているプログラムは https://github.com/TAMS-Group/pr2_robot/blob/pr-various/pr2_bringup/scripts/calibrate_pr2.py ．

このプログラムではグリッパのみや手首のみといった指定はできないので元のキャリブレーションプログラムに少し変更を加えた[暫定プログラム](https://gist.github.com/Kanazawanaoaki/d33f24df4b222db4611cf7730d4ae687#file-partial_calibrate_pr2-py )を用意．`--only_gripper`でグリッパのみ，`--only_wrist`で手首のみのキャリブレーションが可能．（現状コントローラの切り替えで全身のサーボをオフにするようになってはいる．）
PR1040のC1の`/home/applications/Downloads`に置いてあり，以下のように実行可能．
```bash
python partial_calibrate_pr2.py --recalibrate --only_gripper /home/applications/ros/noetic/src/PR2/pr2_robot/pr2_controller_configuration/pr2_calibration_controllers.yaml /home/applications/ros/noetic/src/PR2/pr2_robot/pr2_controller_configuration/pr2_joint_position_controllers.yaml
```

mqcmd196, 2025-08-26:
https://github.com/PR2/pr2_robot ではなくて https://github.com/TAMS-Group/pr2_robot なのはなぜでしょうか？

ぱっと見ですが， https://github.com/TAMS-Group/pr2_robot の kinetic-devel-tamsの方は，改造されたPR2の変更が入っている気がしていて，例えば `forearm_r` が消去されている気がします (https://github.com/PR2/pr2_robot/compare/kinetic-devel...TAMS-Group:pr2_robot:kinetic-devel-tams#diff-7fec4d18727961df323da397f0dfe8749a15c3a3648ddbee6b8a71796f20754aL504) ．

> プログラムではグリッパのみや手首のみといった指定はできないので元のキャリブレーションプログラムに少し変更を加えた[暫定プログラム](https://gist.github.com/Kanazawanaoaki/d33f24df4b222db4611cf7730d4ae687#file-partial_calibrate_pr2-py)を用意．--only_gripperでグリッパのみ，--only_wristで手首のみのキャリブレーションが可能．（現状コントローラの切り替えで全身のコントローラを切り替えていて全身がサーボをオフにするようになってはいる

は需要ありそうなのと，散逸を防ぐためにPull Requestを出してみるのはどうでしょうか．

mqcmd196, 2025-08-26:
あ，申し訳ない．pr-various ブランチの方なので，kinetic-devel-tams とは関係ありませんでした．